stefanbode/Sonoff-Tasmota #81: HA/MQTT help needed

Sundown89, 2018-10-18:
Hello. I can not get the firmware to work in HA through MQTT. Tried both methods from the wiki, changing topics to my own configuration. Are these methods still relevant to the latest firmware version? Also, how do I check all MQTT topics used by the Sonoff?

In the firmware GUI the shutters work correctly. I control the shutters with Sonoff Dual R2 in normal operating mode.

On HA forum I find a couple threads on MQTT in HA but they work with the original Tasmota, so different payloads.

stefanbode, 2018-10-19:
Take a look at the wiki. There are some contributors configuring HA to work with shutters. I posted their examples in the Wiki. Hope this helps to understand the general method.

Sundown89, 2018-10-19:
I was hoping for a recent config from a user, possibly B1G1(?). He indeed is contributing on this matter. I can't get it to work though. It seems that he has the same hardware and shutters like I have. I'm struggling with the different functions in your firmware and the original Tasmota. That is why it's hard to troubleshoot for me.

B1G1, 2018-10-20:
Hi, yes i have 7 Sonoff Dual R2 running Stefan firmware, configured in HA like this

- platform: mqtt
  name: "Tapparella Salotto"
  availability_topic: "tele/tapparellaSalotto/LWT"
  state_topic: "stat/tapparellaSalotto/SHUTTER1"
  command_topic: "cmnd/tapparellaSalotto/backlog"
  set_position_topic: "cmnd/tapparellaSalotto/shutterposition"
  value_template: "{{ value }}"
  qos: 1
  retain: false
  payload_open: "SHUTTEROPEN"
  payload_close: "SHUTTERCLOSE"
  payload_stop: "SHUTTERSTOP"
  state_open: "ON"
  state_closed: "OFF"
  payload_available: "Online"
  payload_not_available: "Offline"
  optimistic: true

Add it in file covers.yaml, and in configuration.yaml add this line cover: !include covers.yaml

I use mosquitto addon for Hassio

Sundown89, 2018-10-22:
Thank you very much! I finallly got it working. 

Only thing is when I close the shutter, stop it at a random point and open it again it does not go all the way up. It's just a cm or so but I will troubleshoot this now. Something with shuttercloseduration/shutteropenduration I guess.

closed 2018-10-22 by Sundown89