stefanbode/Sonoff-Tasmota #82: Shutters close after reconnect to MQTT Broker

electrontube, 2018-10-20:
With MQTT is beeing configured there is an issue with Shutter Mode“:
If connection to the MQTT Broker is lost and re-established, all Stutter Modules are triggered and Shutters fully close (regardless of the previous position of the shutter).

This is really critical, imagine you are in your Garden... 
a glitch in your WiFi could lock you out of your house.

B1G1, 2018-10-20:
You have to set retain false on the mqtt side to prevent this kind of issue

electrontube, 2018-10-20:
Thanks for the hint, but retain is already set to false, that doesn't solve the problem.

B1G1, 2018-10-20:
May be there is already a retained message. In my case i had to reset the mqtt broker

electrontube, 2018-10-20:
There shouldn't be any retained messages, as everything was working fine before and there were no outstanding tasks at the point in time the Connection to the broker was reestablished.

electrontube, 2018-10-21:
Thanks @B1G1 , purging the persistence database helped for the moment, but I still feel there is more to be investigated.

stefanbode, 2018-10-21:
Please can you set loglevel 5 and send the log after restart. We should easily find out where a command comes from and what happen in detail. The log is your friend

electrontube, 2018-10-24:
@stefanbode 
The problem appears to be gone after purging the persistence database on the broker.
I will keep an eye on it and report if it occurs again.

stefanbode, 2018-10-24:
Yes, that sounds reasonable. We do not have this problem elsewhere and the MQTT retain stuff also creates some confusion in the Forum of the TASMOTA main forum. There you will find in the MQTT session also some commands how to delete values. I'm using MQTT-SPY for browsing my values and this works great. But there are also other tools available.

closed 2018-10-24 by stefanbode