stefanbode/Sonoff-Tasmota #84: Shutter mode not working

Sundown89, 2018-10-24:
Hi, I configured 3 Sonoff Dual R2's now and all is working well.
However, the fourth is showing strange behaviour. When enabling shutter mode by setoption80 1 the shutter commands are not recognized. Also, the slider does not appear. Followed all the same steps as before and triple checked if I had the correct firmware. Flashed multiple times. Also tried restarting after enabling shutter mode. 

Screenshot of the console:
[image]

stefanbode, 2018-10-24:
Ok, there is one option possible. Please send the command: "shutterrelay1 1"
The restart the device. Also having the full log file from the beginning is helpful. The shutter will only appear if the relay is >0 and less <33. Maybe some strange values already inside

Sundown89, 2018-10-25:
Hi Stefan, that did it. Thank you! Don't understand how the initial values can be different from other devices with exactly the same firmware and commands.

Anyhow, here's the log after putting "shutterrelay1 1" and restarting:

00:00:00 Shutter 0 (Relay:1): Init. Pos: -4037702 [255 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 510000, Opentime 255 [s], Closetime 255 [s], CoedffCalc: c0: -6152, c1 -15810, c2: -17040, c3: -1, c4: -1, binmask 3, is inverted 255, size 40
00:00:00 Project screenm ScreenMidden (Topic screenm, Fallback DVES_AC2580, GroupTopic sonoffs) Version 6.2.1.11 stb-1.6-2_4_2
00:00:00 WIF: Connecting to AP1 Smarthome in mode 11N as screenm-1408...
00:00:05 WIF: Connected
00:00:05 DNS: Initialized
00:00:05 HTP: Web server active on screenm-1408.local with IP address 10.0.0.103
00:00:05 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: tele/screenm/LWT = Online (retained)
00:00:06 MQT: cmnd/screenm/POWER = 
00:00:06 MQT: tele/screenm/INFO1 = {"Module":"Sonoff Dual R2","Version":"6.2.1.11 stb-1.6","FallbackTopic":"DVES_AC2580","GroupTopic":"sonoffs"}
00:00:06 MQT: tele/screenm/INFO2 = {"WebServerMode":"Admin","Hostname":"screenm-1408","IPAddress":"10.0.0.103"}
00:00:06 MQT: tele/screenm/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: stat/screenm/RESULT = {"POWER1":"OFF"}
00:00:06 MQT: stat/screenm/POWER1 = OFF
00:00:06 MQT: stat/screenm/RESULT = {"POWER2":"OFF"}
00:00:06 MQT: stat/screenm/POWER2 = OFF
10:02:18 MQT: tele/screenm/UPTIME = {"Time":"2018-10-25T10:02:18","Uptime":"0T00:00:09"}
10:02:23 MQT: tele/screenm/STATE = {"Time":"2018-10-25T10:02:23","Uptime":"0T00:00:14","Vcc":3.394,"POWER1":"OFF","POWER2":"OFF", "Wifi":{"AP ":1, "SSId ":"Smarthome", "BSSId ":"FE:EC:DA:8A:47:96", "Channel":6, "RSSI":78}, "DeepSleep":-1, "Heap":24808}
10:02:23 MQT: tele/screenm/SENSOR = {"Time":"2018-10-25T10:02:23","SHUTTER-1":-155}
10:02:51 CMD: shuttercloseduration 33
10:02:51 Shutter 0 (Relay:1): Init. Pos: -4037702 [255 %], Open Vel.: 100 Close Vel.: 772 , Max Way: 510000, Opentime 255 [s], Closetime 33 [s], CoedffCalc: c0: -6152, c1 -15810, c2: -17040, c3: -1, c4: -1, binmask 3, is inverted 255, size 40
10:02:51 MQT: stat/screenm/RESULT = {"shuttercloseduration1":33}
10:03:05 CMD: SHUTTEROPENDURATION 34
10:03:05 Shutter 0 (Relay:1): Init. Pos: -520884 [255 %], Open Vel.: 100 Close Vel.: 103 , Max Way: 68000, Opentime 34 [s], Closetime 33 [s], CoedffCalc: c0: 16656, c1 -2108, c2: 1223, c3: -1, c4: -1, binmask 3, is inverted 255, size 40
10:03:05 MQT: stat/screenm/RESULT = {"shutteropenduration1":34}

stefanbode, 2018-10-25:
Can you please correct some of the values. I see in the startlog there are definitely wrong values inside:
10:03:05 Shutter 0 (Relay:1): Init. Pos: -520884 [255 %], Open Vel.: 100 Close Vel.: 103 , Max Way: 68000, Opentime 34 [s], Closetime 33 [s], CoedffCalc: c0: 16656, c1 -2108, c2: 1223, c3: -1, c4: -1, binmask 3, **is inverted 255,** size 40.
Inverted should be set to 0, if you want 100 for open and 0 for close.

Sundown89, 2018-10-25:
I set inverted to 0. Then I used shutter50percent and shuttersetclose to start with. This got the Init. Pos to 0%. But still a shutter open or close command keeps going way longer than the set durations.

00:00:00 Shutter 0 (Relay:1): Init. Pos: 0 [0 %], Open Vel.: 100 Close Vel.: 103 , Max Way: 68000, Opentime 34 [s], Closetime 33 [s], CoedffCalc: c0: 16656, c1 -2108, c2: 1223, c3: -1, c4: -1, binmask 3, is inverted 0, size 40
00:00:00 Project screenm ScreenMidden (Topic screenm, Fallback DVES_AC2580, GroupTopic sonoffs) Version 6.2.1.11 stb-1.6-2_4_2
00:00:00 WIF: Connecting to AP1 Smarthome in mode 11N as screenm-1408...
00:00:07 WIF: Connected
00:00:07 DNS: Initialized
00:00:07 HTP: Web server active on screenm-1408.local with IP address 10.0.0.103
00:00:08 MQT: Attempting connection...
00:00:08 MQT: Connected
00:00:08 MQT: tele/screenm/LWT = Online (retained)
00:00:08 MQT: cmnd/screenm/POWER = 
00:00:08 MQT: tele/screenm/INFO1 = {"Module":"Sonoff Dual R2","Version":"6.2.1.11 stb-1.6","FallbackTopic":"DVES_AC2580","GroupTopic":"sonoffs"}
00:00:08 MQT: tele/screenm/INFO2 = {"WebServerMode":"Admin","Hostname":"screenm-1408","IPAddress":"10.0.0.103"}
00:00:08 MQT: tele/screenm/INFO3 = {"RestartReason":"Software/System restart"}
00:00:08 MQT: stat/screenm/RESULT = {"POWER1":"OFF"}
00:00:08 MQT: stat/screenm/POWER1 = OFF (retained)
00:00:08 MQT: stat/screenm/RESULT = {"POWER2":"OFF"}
00:00:08 MQT: stat/screenm/POWER2 = OFF (retained)
11:00:20 MQT: tele/screenm/STATE = {"Time":"2018-10-25T11:00:20","Uptime":"0T00:00:16","Vcc":3.394,"POWER1":"OFF","POWER2":"OFF", "Wifi":{"AP ":1, "SSId ":"Smarthome", "BSSId ":"FE:EC:DA:8A:47:96", "Channel":6, "RSSI":62}, "DeepSleep":-1, "Heap":24240}
11:00:20 MQT: tele/screenm/SENSOR = {"Time":"2018-10-25T11:00:20","SHUTTER-1":0}
.......

.......
11:03:13 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 566
11:03:14 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 586
11:03:15 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 606
11:03:16 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 626
11:03:17 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 646
11:03:18 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 666
11:03:19 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 686
11:03:20 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 706
11:03:21 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 726
11:03:22 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 746
11:03:23 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 766
11:03:24 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 786
11:03:25 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 806
11:03:26 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 826
11:03:27 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 846
11:03:28 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 866
11:03:29 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 886
11:03:30 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 907
11:03:31 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 926
11:03:32 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 946
11:03:33 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 966
11:03:34 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 986
11:03:35 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 1006
11:03:36 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 1026
11:03:37 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 1054
11:03:38 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 1071
11:03:39 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 1091
11:03:40 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 1111
11:03:41 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 1131
11:03:42 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 1151
11:03:43 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 1171
11:03:44 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 1191
11:03:45 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 1211
11:03:46 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 1231
11:03:47 Shutter 0: Real Pos: 0, Target -194144, source: Shutter, pos %: 0, direction: -1, rtcshutter: 1251
11:03:47 MQT: stat/screenm/RESULT = {"POWER2":"OFF"}
11:03:47 MQT: stat/screenm/POWER2 = OFF (retained)
11:03:47 MQT: stat/screenm/SHUTTER1 = 0 (retained)

stefanbode, 2018-10-25:
Yes I see the calculation of the target position is wrong. Let me check what I have broken last time

stefanbode, 2018-10-25:
in your case shutter50percent is not set to 50. this causes the behaviour.

Init should look like:
```
14:40:59 Shutter 0 (Relay:1): Init. Pos: 0 [0 %], Open Vel.: 100 Close Vel.: 103 , Max Way: 68000, Opentime 34.0 [s], Closetime 33.0 [s], CoedffCalc: c0: 0, c1 68
0, c2: 680, c3: 0, c4: 0, binmask 63, is inverted 0, size 40
```

stefanbode, 2018-10-27:
Defining SHUTTERRELAY2 is essential, because here I configure how many shutters are connected to one esp. it could be up to 4 in the current configuration.

closed 2018-10-27 by stefanbode