stefanbode/Sonoff-Tasmota #86: mqtt login error

madmicio, 2018-11-05:
Hi stefan,
with the v3 version of mosquitto the problems started. after several attempts the mqtt of the tasmota firmware must be configured with
user: DVES_USER
and password: DVES_PASS.

for the moment only with these parameters the clients are able to login.
this option unfortunately does not work on your version. honestly I can not tell you why.
can you kindly check?
I tried to recompile the bin by inserting only ip and wifi parameters and only then from the web interface enter the parameters of the mqtt, but fail.
I'm not using the user_config_override file

stefanbode, 2018-11-06:
Hi @madmicio. Could you explain the issue a bit more? I use the mosquitto for all my devices and they work like a charm without any issues. Therefore I do not understand the issue. Anyhow my fork is always very close to the MASTER and these kind of issues definitely must be visible in the master, too

stefanbode, 2018-11-06:
Please create the override file and put in your credentials as shown in the sample file. This makes many things easier and you have to type less.

B1G1, 2018-11-08:
@madmicio are you using HASSIO and mosquitto addon?

madmicio, 2018-11-08:
Oh, sorry... Yes! Hassio and morquitto

madmicio, 2018-11-09:
With the mosquitto V4 all ok!

closed 2018-11-12 by stefanbode