stefanbode/Sonoff-Tasmota #88: new cover management in home assistant

madmicio, 2018-11-12:
hassio 0.82 ntroduces a new management of the cover.
this is the suggested code

cover:
  - platform: mqtt
    name: "MQTT Cover"
    command_topic: "home-assistant/cover/set"
    state_topic: "home-assistant/cover/state"
    availability_topic: "home-assistant/cover/availability"
    qos: 0
    retain: true
    payload_open: "OPEN"
    payload_close: "CLOSE"
    payload_stop: "STOP"
    state_open: "open"
    state_closed: "closed"
    payload_available: "online"
    payload_not_available: "offline"
    optimistic: false
    value_template: '{{ value.x }}'

How can I implement this with your firmware?

stefanbode, 2018-11-12:
I think example no2 in the wiki should do the job.

https://github.com/stefanbode/Sonoff-Tasmota/wiki/Shutter-blinds-working-with-two-relays

stefanbode, 2018-11-12:
Can you be more specific, what is not working as a command and what works. Open close position, reporting an so on. For me the second example looks promising to match with the requirements

madmicio, 2018-11-12:
if I use this code:

cover:
  - platform: mqtt
    name: "Test"
    availability_topic: "tele/portafinestra/LWT"
    state_topic: "stat/portafinestra/RESULT"
    command_topic: "cmnd/portafinestra/backlog"
    value_template: '{{value_json.shutterposition1}}'
    qos: 1
    retain: true
    payload_open: "SHUTTEROPEN"
    payload_close: "SHUTTERCLOSE"
    payload_stop: "SHUTTERSTOP"
    state_open: "ON"
    state_closed: "OFF"
    payload_available: "Online"
    payload_not_available: "Offline"
    optimistic: true

I can move the shutter but I don't hane the slider. an in the home assistant log I got this error:

2018-11-12 17:18:22 ERROR (MainThread) [homeassistant.helpers.condition] Error during template condition: UndefinedError: 'None' has no attribute 'attributes'
2018-11-12 17:18:26 WARNING (MainThread) [homeassistant.components.cover.mqtt] Payload is not True or False: 
2018-11-12 17:18:28 WARNING (MainThread) [homeassistant.components.cover.mqtt] Payload is not True or False: 
2018-11-12 17:18:32 WARNING (MainThread) [homeassistant.components.cover.mqtt] Payload is not True or False: 
2018-11-12 17:18:34 WARNING (MainThread) [homeassistant.components.cover.mqtt] Payload is not True or False: 

if I add thi code:
    tilt_command_topic: 'cmnd/portafinestra/shutterposition'
    tilt_status_topic: 'cmnd/portafinestra/shutterposition'
    set_position_topic: 'cmnd/portafinestra/shutterposition'
    tilt_min: 0
    tilt_max: 100
    tilt_closed_value: 0
    tilt_opened_value: 100

The result is:
Testing configuration at /config
Failed config
  cover.mqtt: 
    - Invalid config for [cover.mqtt]: Set position topic must be set together with get position topic.. Got OrderedDict([('platform', 'mqtt'), ('name', 'Test'), ('availability_topic', 'tele/portafinestra/LWT'), ('state_topic', 'stat/portafinestra/RESULT'), ('command_topic', 'cmnd/portafinestra/backlog'), ('value_template', '{{value_json.shutterposition1}}'), ('qos', 1), ('retain', True), ('payload_open', 'SHUTTEROPEN'), ('payload_close', 'SHUTTERCLOSE'), ('payload_stop', 'SHUTTERSTOP'), ('state_open', 'ON'), ('state_closed', 'OFF'), ('payload_available', 'Online'), ('payload_not_available', 'Offlin.... (See ?, line ?). Please check the docs at https://home-assistant.io/components/cover.mqtt/
    - platform: mqtt
      availability_topic: tele/portafinestra/LWT
      command_topic: cmnd/portafinestra/backlog
      name: Test
      optimistic: True
      payload_available: Online
      payload_close: SHUTTERCLOSE
      payload_not_available: Offline
      payload_open: SHUTTEROPEN
      payload_stop: SHUTTERSTOP
      qos: 1
      retain: True
      set_position_topic: hm/cmnd/portafinestra/shutterposition
      state_closed: OFF
      state_open: ON
      state_topic: stat/portafinestra/RESULT
      tilt_closed_value: 0
      tilt_command_topic: hm/cmnd/portafinestra/shutterposition
      tilt_max: 100
      tilt_min: 0
      tilt_opened_value: 100
      tilt_status_topic: hm/cmnd/portafinestra/shutterposition
      value_template: {{value_json.shutterposition1}}
Successful config (partial)
  cover.mqtt:

stefanbode, 2018-11-12:
You should ask @b1g1 for help. Maybe there is an additional file missing for the template definition.

madmicio, 2018-11-12:
it could be a good idea. but how do I send a message ???

B1G1, 2018-11-12:
Hi, i didn’t update to 0.82 yet, too many changes and i still running 0.79.3

B1G1, 2018-11-12:
I already downgraded from 0.80 to 0.79.3 because that version broke my configuration

You have to use ssh to run this command 
hassio homeassistant update --options “version=x.xx.x”

madmicio, 2018-11-12:
ok just downgrade… all ok!

bastshoes, 2018-11-20:
For me new new mqtt.cover in HASS works fine. Here is my config:
```
- platform: mqtt
    name: "Balcony Blinds"
    availability_topic: "balconyblinds/tele/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"
    position_topic: balconyblinds/stat/SHUTTER1
    position_open: 100
    position_closed: 0
    set_position_topic: "balconyblinds/cmnd/shutterposition"
    value_template: "{{ value | int }}"
    command_topic: "balconyblinds/cmnd/backlog"
    payload_open: "SHUTTEROPEN"
    payload_close: "SHUTTERCLOSE"
    payload_stop: "SHUTTERSTOP"
    retain: false
    optimistic: false
    qos: 1
```

bastshoes, 2018-11-20:
According to new config in HASS state topic is not considered if position topic specified.
```
state_topic

    (string)(Optional)The MQTT topic subscribed to receive cover state messages. Use only if not using get_position_topic. State topic can only read open/close state. Cannot read position state.
```

Sundown89, 2018-11-20:
Take a look at your position_topic. 
Mine is: position_topic: screenr/stat/SHUTTER1
Yours starts with stat, maybe that's the problem.

madmicio, 2018-11-20:
whit this code all works fine, the only problem is that after the restart I do not have the position until I move the shutter. it's normal?

bastshoes, 2018-11-20:
Have you set 'poveretain 1' in console for you shutter? HASS do see previous messages if they do not have retain flag.

madmicio, 2018-11-20:
ok… with powerretain works fine!

bastshoes, 2018-11-21:
Added example to wiki. I guess issue can be closed.

titomontes, 2019-08-30:
@stefanbode, can be closed.

closed 2019-08-30 by stefanbode