stefanbode/Sonoff-Tasmota #90: strange movement of shutter2

madmicio, 2018-11-19:
Hi Stefan,
I have an abnormal behavior to tell you. after the restart, the shutter2 makes a small movement.
I am attaching the log. start condition shutter2 open and then give the restart command

00:00:00 SRC: Restart
00:00:00 Switched relay: 0 by Restart
00:00:00 Progetto sonoff Sonoff (Topic salone, Riserva salone, GroupTopic sonoffs) Versione 6.3.0.2 stb-1.4-2_4_2
00:00:00 CFG: Salvato nella flash in F4, Conteggio 48, Bytes 3824
00:00:00 WIF: Controllo connessione...
00:00:00 WIF: Tentativo di connessione...
00:00:00 WIF: Connessione ad AP1 Fast-Arci in modalità 11N come salone-3879...
00:00:01 WIF: Controllo connessione...
00:00:01 WIF: Tentativo di connessione...
00:00:02 WIF: Controllo connessione...
00:00:02 WIF: Tentativo di connessione...
00:00:03 WIF: Controllo connessione...
00:00:03 WIF: Tentativo di connessione...
00:00:04 WIF: Controllo connessione...
00:00:04 WIF: Connesso
00:00:04 DNS: Inizializzato
00:00:04 HTP: Web server attivo su salone-3879.local con indirizzo IP 192.168.1.40
00:00:04 APP: (UTC) Mon Nov 19 22:41:29 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
23:41:30 MQT: stat/salone/RESULT = {"POWER4":"OFF"}
23:41:30 MQT: stat/salone/POWER4 = OFF
23:41:30 SRC: MQTT
23:41:30 RSL: Topic Ricevuto cmnd/salone/backlog, Dimensione Dati 12, Dati SHUTTEROPEN2
23:41:30 RSL: Gruppo 0, Indice 1, Comando BACKLOG, Dati SHUTTEROPEN2
23:41:30 SRC: MQTT
23:41:30 RSL: Topic Ricevuto cmnd/salone/shutterposition2, Dimensione Dati 1, Dati 3
23:41:30 RSL: Gruppo 0, Indice 2, Comando SHUTTERPOSITION, Dati 3
23:41:30 shutterposition in: payload 3, index 2, source 2
23:41:30 lastsource 2:, realpos 110000, target 3300, payload 3
23:41:30 Start shutter in right direction -1
23:41:30 Delay6 5s
23:41:30 temp1 for mask is 3, temp2: 2, temp3 1. Bucketsize: 2, device: 4 to state 1, currentsate 0
23:41:30 imask1: 4
23:41:30 SRC: Shutter
23:41:30 Switched relay: 8 by Shutter
23:41:30 MQT: stat/salone/RESULT = {"POWER4":"ON"}
23:41:30 MQT: stat/salone/POWER4 = ON
23:41:30 MQT: stat/salone/RESULT = {"shutterposition2":21}
23:41:30 SRC: Backlog
23:41:30 RSL: Topic Ricevuto /SHUTTEROPEN2, Dimensione Dati 0, Dati 
23:41:30 RSL: Gruppo 0, Indice 2, Comando SHUTTEROPEN, Dati 
23:41:30 shutterposition in: payload 100, index 2, source 7
23:41:30 lastsource 7:, realpos 109000, target 110000, payload 100
23:41:30 Start shutter in right direction 1
23:41:30 Delay6 5s
23:41:31 temp1 for mask is 3, temp2: 1, temp3 2. Bucketsize: 2, device: 3 to state 1, currentsate 0
23:41:31 imask1: 8
23:41:31 Power off device: 4
23:41:31 SRC: Shutter
23:41:31 Switched relay: 8 by Shutter
23:41:31 MQT: stat/salone/RESULT = {"POWER4":"OFF"}
23:41:31 MQT: stat/salone/POWER4 = OFF
23:41:32 SRC: Shutter
23:41:32 Switched relay: 4 by Shutter
23:41:32 MQT: stat/salone/RESULT = {"POWER3":"ON"}
23:41:32 MQT: stat/salone/POWER3 = ON
23:41:32 MQT: stat/salone/RESULT = {"shutteropen2":21}
23:41:32 CFG: Salvato nella flash in FB, Conteggio 49, Bytes 3824
23:41:32 Shutter 1: Real Pos. 110000, Stoppos: 100, relay: 2, direction 1, pulsetimer: 0, rtcshutter: 1.1 [s], operationtime 0
23:41:32 MQT: stat/salone/SHUTTER2 = 100
23:41:32 SRC: Shutter
23:41:32 Switched relay: 4 by Shutter
23:41:32 MQT: stat/salone/RESULT = {"POWER3":"OFF"}
23:41:32 MQT: stat/salone/POWER3 = OFF
23:41:32 SRC: MQTT
23:41:32 RSL: Topic Ricevuto cmnd/salone/backlog, Dimensione Dati 12, Dati SHUTTEROPEN2
23:41:32 RSL: Gruppo 0, Indice 1, Comando BACKLOG, Dati SHUTTEROPEN2
23:41:32 SRC: MQTT
23:41:32 RSL: Topic Ricevuto cmnd/salone/shutterposition2, Dimensione Dati 1, Dati 3
23:41:32 RSL: Gruppo 0, Indice 2, Comando SHUTTERPOSITION, Dati 3
23:41:32 shutterposition in: payload 3, index 2, source 21
23:41:32 lastsource 21:, realpos 110000, target 3300, payload 3
23:41:32 Start shutter in right direction -1
23:41:32 Delay6 5s
23:41:32 temp1 for mask is 3, temp2: 2, temp3 1. Bucketsize: 2, device: 4 to state 1, currentsate 0
23:41:32 imask1: 4
23:41:32 SRC: Shutter
23:41:32 Switched relay: 8 by Shutter
23:41:32 MQT: stat/salone/RESULT = {"POWER4":"ON"}
23:41:32 MQT: stat/salone/POWER4 = ON
23:41:32 MQT: stat/salone/RESULT = {"shutterposition2":21}
23:41:32 SRC: Backlog
23:41:32 RSL: Topic Ricevuto /SHUTTEROPEN2, Dimensione Dati 0, Dati 
23:41:32 RSL: Gruppo 0, Indice 2, Comando SHUTTEROPEN, Dati 
23:41:32 shutterposition in: payload 100, index 2, source 7
23:41:32 lastsource 7:, realpos 108900, target 110000, payload 100
23:41:32 Start shutter in right direction 1
23:41:32 Delay6 5s
23:41:33 temp1 for mask is 3, temp2: 1, temp3 2. Bucketsize: 2, device: 3 to state 1, currentsate 0
23:41:33 imask1: 8
23:41:33 Power off device: 4
23:41:33 SRC: Shutter
23:41:33 Switched relay: 8 by Shutter
23:41:33 MQT: stat/salone/RESULT = {"POWER4":"OFF"}
23:41:33 MQT: stat/salone/POWER4 = OFF
23:41:33 SRC: Shutter
23:41:33 Switched relay: 4 by Shutter
23:41:33 MQT: stat/salone/RESULT = {"POWER3":"ON"}
23:41:33 MQT: stat/salone/POWER3 = ON
23:41:33 MQT: stat/salone/RESULT = {"shutteropen2":21}
23:41:33 Shutter 1: Real Pos. 110000, Stoppos: 100, relay: 2, direction 1, pulsetimer: 0, rtcshutter: 1.1 [s], operationtime 0
23:41:33 MQT: stat/salone/SHUTTER2 = 100
23:41:33 SRC: Shutter
23:41:33 Switched relay: 4 by Shutter
23:41:33 MQT: stat/salone/RESULT = {"POWER3":"OFF"}
23:41:33 MQT: stat/salone/POWER3 = OFF
23:41:34 CFG: Salvato nella flash in FA, Conteggio 50, Bytes 3824
23:41:41 MQT: tele/salone/STATE = {"Time":"2018-11-19T23:41:41","Uptime":"0T00:00:16","Vcc":3.457,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF", "Wifi":{"AP":1, "SSId":"Fast-Arci","BSSId":"10:13:31:F8:D3:43","Channel":6,"RSSI":98},"DeepSleep":0,"Heap":20144}
23:41:41 MQT: tele/salone/SENSOR = {"Time":"2018-11-19T23:41:41","SHUTTER-1":15,"SHUTTER-2":100}
23:41:52 WIF: Controllo connessione...
23:41:52 WIF: Connesso

madmicio, 2018-11-20:
no. but I did not understand if it should be set on or off?

madmicio, 2018-11-20:
I gave up. I changed mqtt topic name and now everything works correctly. you can close thanks

stefanbode, 2018-11-21:
This is again a sign, that in MQTT the message is stored. Never mind. Good to hear it work now.

closed 2018-11-21 by stefanbode