stefanbode/Sonoff-Tasmota #91: Anzeige beim DP "shutterposizion" springt immer auf 21

ihavenonick, 2018-11-22:
Im DP shutterposition kann man zwar wunderbar die gewünschte Position angeben, allerdings springt er danach immer auf 21 und somit kann man z.B. über VIS nicht die aktuelle Position des Rollladen abfragen.

80% über den DP "shutterposition" eingestellt:
[image]
Anzeige unter Objekte nach Ende des Kommando:
[image]

Hier sollte doch nun auch 80 stehen, oder nicht?

stefanbode, 2018-11-23:
Guten morgen, ja da sollte 80 stehen. Kannst Du mal dir das LOG anschauen. Sieht mal unter "KONSOLE". als befehl "weblog 5" erhöht auch noch mal die anzahl der Ausgaben. Mit dem Log kann ich Dir genauer sagen was passiert und warum und wie man das ggf korrigieren kann.

ihavenonick, 2018-11-23:
weblog5? Hab ich gemacht  und diese Ausgabe in der Konsole, wen nich den Regler auf 50% stelle.
`00:00:04 APP: (UTC) Fri Nov 23 09:29:04 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
10:29:05 WIF: Prüfe Verbindung...
10:29:05 WIF: verbunden
10:29:05 HTP: Web-Server aktiv bei Test-Rollladen mit IP-Adresse 192.168.100.25
10:29:05 HTP: Hauptmenü
10:29:05 MQT: Verbindungsversuch...
10:29:05 MQT: verbunden
10:29:05 MQT: tele/sonoff/LWT = online (beibehalten)
10:29:05 MQT: cmnd/sonoff/POWER = 
10:29:05 MQT: abonniere cmnd/sonoff/#
10:29:05 MQT: abonniere cmnd/sonoffs/#
10:29:05 MQT: abonniere cmnd/Test_Rollladen/#
10:29:05 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff Dual R2","Version":"6.3.0.2 stb-1.4","FallbackTopic":"Test_Rollladen","GroupTopic":"sonoffs"}
10:29:05 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"Test-Rollladen","IPAddress":"192.168.100.25"}
10:29:05 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
10:29:05 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
10:29:05 MQT: stat/sonoff/POWER1 = OFF
10:29:05 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
10:29:05 MQT: stat/sonoff/POWER2 = OFF
10:29:13 MQT: tele/sonoff/STATE = {"Time":"2018-11-23T10:29:13","Uptime":"0T00:00:13","Vcc":3.488,"POWER1":"OFF","POWER2":"OFF", "Wifi":{"AP":1, "SSId":"FRITZ!","BSSId":"5C:49:79:89:9F:0F","Channel":6,"RSSI":56},"DeepSleep":0,"Heap":28448}
10:29:13 MQT: tele/sonoff/SENSOR = {"Time":"2018-11-23T10:29:13","Switch1":"ON","Switch2":"ON","SHUTTER-1":0}
10:29:25 WIF: Prüfe Verbindung...
10:29:25 WIF: verbunden
10:29:30 HTP: Konsole
10:29:45 WIF: Prüfe Verbindung...
10:29:45 WIF: verbunden
10:29:46 HTP: Hauptmenü
10:29:49 HTP: Einstellungen
10:29:51 HTP: Logging konfigurieren
10:30:05 WIF: Prüfe Verbindung...
10:30:05 WIF: verbunden
10:30:15 HTP: Einstellungen
10:30:18 HTP: Hauptmenü
10:30:25 WIF: Prüfe Verbindung...
10:30:25 WIF: verbunden
10:30:45 WIF: Prüfe Verbindung...
10:30:45 WIF: verbunden
10:30:51 SRC: WebGui from 192.168.100.2
10:30:51 RSL: empfangenes topic /shutterposition1, Datengröße 2, Daten 46
10:30:51 RSL: Gruppe 0, Index 1, Befehl SHUTTERPOSITION, Daten 46
10:30:51 shutterposition in: payload 46, index 1, source 7
10:30:51 lastsource 7:, realpos 0, target 14720, payload 46
10:30:51 Start shutter in right direction 1
10:30:51 Delay6 5s
10:30:52 temp1 for mask is 3, temp2: 1, temp3 2. Bucketsize: 2, device: 1 to state 1, currentsate 0
10:30:52 imask1: 2
10:30:52 SRC: Shutter
10:30:52 Switched relay: 1 by Shutter
10:30:52 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
10:30:52 MQT: stat/sonoff/POWER1 = ON
10:30:52 MQT: stat/sonoff/RESULT = {"shutterposition1":21}
10:30:52 CFG: in Flash gespeichert am FA, zählen 18, Bytes 3824
10:30:53 Shutter 0: Real Pos: 2600, Target 14720, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 1.3  [s]
10:30:54 Shutter 0: Real Pos: 4700, Target 14720, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 2.3  [s]
10:30:55 Shutter 0: Real Pos: 6700, Target 14720, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 3.3  [s]
10:30:56 Shutter 0: Real Pos: 8700, Target 14720, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 4.3  [s]
10:30:57 Shutter 0: Real Pos: 10600, Target 14720, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 5.3  [s]
10:30:58 Shutter 0: Real Pos: 12600, Target 14720, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 6.3  [s]
10:30:59 Shutter 0: Real Pos: 14600, Target 14720, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 7.3  [s]
10:30:59 Shutter 0: Real Pos. 14800, Stoppos: 46, relay: 0, direction 1, pulsetimer: 0, rtcshutter: 7.4 [s], operationtime 2
10:30:59 MQT: stat/sonoff/SHUTTER1 = 46
10:30:59 SRC: Shutter
10:30:59 Switched relay: 1 by Shutter
10:30:59 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
10:30:59 MQT: stat/sonoff/POWER1 = OFF
10:30:59 CFG: in Flash gespeichert am F9, zählen 19, Bytes 3824
10:31:05 WIF: Prüfe Verbindung...
10:31:05 WIF: verbunden
10:31:10 HTP: Konsole
10:31:25 WIF: Prüfe Verbindung...
10:31:25 WIF: verbunden`

ihavenonick, 2018-11-23:
Das kommt, wenn ich 50 bei "shutterposition1" eingebe:

`10:36:48 RSL: empfangenes topic cmnd/sonoff/shutterposition1, Datengröße 2, Daten 50
10:36:48 RSL: Gruppe 0, Index 1, Befehl SHUTTERPOSITION, Daten 50
10:36:48 shutterposition in: payload 50, index 1, source 9
10:36:48 lastsource 9:, realpos 6700, target 16000, payload 50
10:36:48 Start shutter in right direction 1
10:36:48 Delay6 5s
10:36:48 temp1 for mask is 3, temp2: 1, temp3 2. Bucketsize: 2, device: 1 to state 1, currentsate 0
10:36:48 imask1: 2
10:36:48 SRC: Shutter
10:36:48 Switched relay: 1 by Shutter
10:36:49 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
10:36:49 MQT: stat/sonoff/POWER1 = ON
10:36:49 MQT: stat/sonoff/RESULT = {"shutterposition1":21}
10:36:49 CFG: in Flash gespeichert am F6, zählen 22, Bytes 3824
10:36:49 Shutter 0: Real Pos: 9000, Target 16000, source: Shutter, start-pos: 20 %, direction: 1, rtcshutter: 1.1  [s]
10:36:50 Shutter 0: Real Pos: 11000, Target 16000, source: Shutter, start-pos: 20 %, direction: 1, rtcshutter: 2.1  [s]
10:36:51 Shutter 0: Real Pos: 13100, Target 16000, source: Shutter, start-pos: 20 %, direction: 1, rtcshutter: 3.2  [s]
10:36:52 Shutter 0: Real Pos: 15100, Target 16000, source: Shutter, start-pos: 20 %, direction: 1, rtcshutter: 4.2  [s]
10:36:53 Shutter 0: Real Pos. 16100, Stoppos: 50, relay: 0, direction 1, pulsetimer: 0, rtcshutter: 4.7 [s], operationtime 6
10:36:53 MQT: stat/sonoff/SHUTTER1 = 50
10:36:53 SRC: Shutter
10:36:53 Switched relay: 1 by Shutter
10:36:53 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
10:36:53 MQT: stat/sonoff/POWER1 = OFF
10:36:53 CFG: in Flash gespeichert am F5, zählen 23, Bytes 3824`

stefanbode, 2018-11-23:
Also der richtige Wert steht schon mal in MQT: stat/sonoff/SHUTTER1 = 50. Und er fährt auch in die entsprechende Position. Warum er im RESULT mit dem falschen/zu dem Zeitpunk aktuelen Wert antwortet wundert mich.

ihavenonick, 2018-11-23:
Ja, die Funktion ist wie gewünscht, nur springt der DP immer auf 21 zurück. Wenn ich manuell eins der beiden Relais schalte, dann bleibt auch einfach 21 stehen und es ändert sich nichts an dem DP.
Bei Shutterposition, soll doch eigentlich immer, egal wie gesteuert wird, die aktuelle Position des Rolladen stehen, richtig?

Dann übrigens mal ein großen Dank für die Fork Version. Damit erspare ich mir mein Script, das exakt dies selbe Funktion hat, wie Deine Tasmota Version

stefanbode, 2018-11-23:
Zwei Anmerkungen: Ja es war ein Fehler drin durch einen Seiteneffekt, der mir die Variable überschrieben hat. Das ist gefixed. Aber ich bin zusätzlich der Meinung, dass Du die Position aus einem anderen MQTT topic holen musst: MQT: stat/sonoff/SHUTTER1 = 50

Wenn das Rollo mit der Hand geschaltet wird, dann wird im RESULT nichts drin stehen, aber dieser Wert wird die Position immer richtig reporten.

closed 2018-11-23 by stefanbode

ihavenonick, 2018-11-23:
Und wo und wie stell ich das an? Ich hab das über den Sonoff Adapter eingebunden und greife hier lediglch auf die dadurch generierten DPs zu.

stefanbode, 2018-11-23:
Klär mich mal auf: DP? VIS?

ihavenonick, 2018-11-23:
DP=Daten Punkt im IOBroker mit Sonoff Adapter.
VIS verwende ich nicht.

ihavenonick, 2018-11-25:
> Aber ich bin zusätzlich der Meinung, dass Du die Position aus einem anderen MQTT topic holen musst: MQT: stat/sonoff/SHUTTER1 = 50

Kannst Du mir bitte noch erklären, was Du damit meinst und wie ich das anstellen soll?

stefanbode, 2018-11-25:
Ich kenne den ioBroker nicht. Ich kann Dir daher nicht weiter helfen. Ich reporte den Wert auf den MQTT Server. Dort kann man den sich abholen, wie alle anderen Werte auch. Das ist eine Aufgabe der ioBroker Spezialisten.

ihavenonick, 2018-11-25:
Ich hab's mittlerweile verstanden, was Du meinst und auch schon dran gebastelt, allerdings mit mäßigem Erfolg. Ich hab nun eine Anfrage direkt an den Entwickler des Sonoff/Tasmota Adapter geschrieben.

commit 315e9c77 (2018-11-23): fixed #91

Due to some side effects reporting in RSULT was wrong