stefanbode/Sonoff-Tasmota #95: Close Shutter doesn't work

calcipher49, 2018-12-03:
Hi,
After successful install of first Shutter using Sonoff dual R2, I have a big issue with second one(identical configuration).
Open shutter works correctly, but not close (ShutterPosition1 0 when it's open).
In details, If Shutter is closed and I launch ShutterPosition1 100, sequence is:
POWER1 = OFF (useless)
POWER2 = OFF (useless)
POWER1 = ON (open shutter)
Waiting during 23s position go from 0 to 46000 in console
POWER1 = OFF (when SHUTTER1=100)
=> all is fine

But if now I launch ShutterPosition1 0, sequence is:
POWER1 = OFF (useless)
POWER2 = ON (close shutter)
POWER2 = OFF (stop close: FAILURE)
POWER1 = ON (open shutter: FAILURE)
Waiting during 23s position go from 46000 to 0 in console
POWER1 = OFF (failure as should be POWER2)

Full trace in attachment, with config at the beguinning.
[trace.txt](https://github.com/stefanbode/Sonoff-Tasmota/files/2641468/trace.txt)

Thanks for your help.

stefanbode, 2018-12-04:
Hi, thanks for rising up the issue. Please, can you increase the loglevel to 5. The I can see WHO is switching the devices ON and OFF. You get information about the SOURCE for the operation. I agree that this behavior is total nonsense for a normal connected shutter. The shuttersetclose you execute, if the shutter is in the close position to reset all values to default. This is more for debugging and executing without a blind connected to test without going to start all the time.

calcipher49, 2018-12-04:
Hi Stefan,

Thanks for answer. Shutter50Percent is already 50.

Inattchment new trace with both Syslog and Weblog to 4 (maximum).
[trace.txt](https://github.com/stefanbode/Sonoff-Tasmota/files/2643484/trace.txt)

Hope it can help to debug.

stefanbode, 2018-12-06:
hi @calcipher49 . I'm trying to reproduce the issue but still not successful. My device operates exactly as it should be. the shutter50percent 51 or 49 you should see initialistion parameters like this:
```
11:52:26 RSL: Received Topic /shutter50percent, Data Size 2, Data 49
11:52:26 RSL: Group 0, Index 1, Command SHUTTER50PERCENT, Data 49
11:52:26 Shutter accuracy digits: 1
11:52:26 Shutter 0 (Relay:1): Init. Pos: 10200 [50 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 20000, Opentime 10.0 [s], Closetime 10.0 [s], CoedffCalc: c0: 400, c1 196, c2: 276, c3: 0, c4: 0, binmask 15, is inverted 1, size 40
11:52:26 MQT: hm/status/sonoff/RESULT = {"shutter50percent1":49}
11:52:27 CFG: Saved to flash at 3F7, Count 1597, Bytes 3824
```

```
11:49:56 RSL: Received Topic /teleperiod, Data Size 3, Data 300
11:49:56 RSL: Group 0, Index 1, Command TELEPERIOD, Data 300
11:49:56 MQT: hm/status/sonoff/RESULT = {"TelePeriod":"300"}
11:49:57 MQT: hm/status/sonoff/STATE = {"Time":"2018-12-06T11:49:57","Uptime":"0T00:21:14","POWER1":"OFF","POWER2":"OFF", "Wifi":{"AP":1, "SSId":"BodeNet_Keller","BSSId":"7C:FF:4D:3D:A3:5B","Channel":11,"RSSI":86},"DeepSleep":0,"Heap":25088}
11:49:57 MQT: hm/status/sonoff/SENSOR = {"Time":"2018-12-06T11:49:57","SHUTTER-1":17}
11:49:57 CFG: Saved to flash at 3F4, Count 1592, Bytes 3824
11:50:11 CMD: shutterposition 0
11:50:11 SRC: WebConsole from 192.168.178.110
11:50:11 RSL: Received Topic /shutterposition, Data Size 1, Data 0
11:50:11 RSL: Group 0, Index 1, Command SHUTTERPOSITION, Data 0
11:50:11 shutterposition in: payload 0, index 1, source 9
11:50:11 lastsource 9:, realpos 16600, target 20000, payload 100
11:50:11 Start shutter in right direction 1
11:50:11 temp1 for mask is 3, temp2: 1, temp3 2. Bucketsize: 2, device: 1 to state 1, currentsate 0
11:50:11 imask1: 2
11:50:11 SRC: Shutter
11:50:11 Switched relay: 1 by Shutter
11:50:11 MQT: hm/status/sonoff/RESULT = {"POWER1":"ON"}
11:50:11 MQT: hm/status/sonoff/POWER1 = ON (retained)
11:50:11 Delay6 5s, xdrv 21
11:50:11 MQT: hm/status/sonoff/RESULT = {"shutterposition1":0}
11:50:12 Shutter 0: Real Pos: 18500, Target 20000, source: Shutter, start-pos: 83 %, direction: 1, rtcshutter: 1.0  [s]
11:50:12 CFG: Saved to flash at 3FB, Count 1593, Bytes 3824
11:50:13 Shutter 0: Real Pos. 20000, Stoppos: 100, relay: 0, direction 1, pulsetimer: 0, rtcshutter: 1.7 [s], operationtime 0
11:50:13 MQT: hm/status/sonoff/SHUTTER1 = 0 (retained)
11:50:13 SRC: Shutter
11:50:13 Switched relay: 1 by Shutter
11:50:13 MQT: hm/status/sonoff/RESULT = {"POWER1":"OFF"}
11:50:13 MQT: hm/status/sonoff/POWER1 = OFF (retained)
11:50:13 CFG: Saved to flash at 3FA, Count 1594, Bytes 3824
11:50:14 WIF: Checking connection...
11:50:14 WIF: Connected
11:50:23 CMD: shutterposition 50
11:50:23 SRC: WebConsole from 192.168.178.110
11:50:23 RSL: Received Topic /shutterposition, Data Size 2, Data 50
11:50:23 RSL: Group 0, Index 1, Command SHUTTERPOSITION, Data 50
11:50:23 shutterposition in: payload 50, index 1, source 9
11:50:23 lastsource 9:, realpos 20000, target 9800, payload 50
11:50:23 Start shutter in right direction -1
11:50:23 temp1 for mask is 3, temp2: 2, temp3 1. Bucketsize: 2, device: 2 to state 1, currentsate 0
11:50:23 imask1: 1
11:50:23 SRC: Shutter
11:50:23 Switched relay: 2 by Shutter
11:50:23 MQT: hm/status/sonoff/RESULT = {"POWER2":"ON"}
11:50:23 MQT: hm/status/sonoff/POWER2 = ON (retained)
11:50:23 Delay6 5s, xdrv 21
11:50:23 MQT: hm/status/sonoff/RESULT = {"shutterposition1":50}
11:50:23 CFG: Saved to flash at 3F9, Count 1595, Bytes 3824
11:50:24 Shutter 0: Real Pos: 17800, Target 9800, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 1.1  [s]
11:50:25 Shutter 0: Real Pos: 15800, Target 9800, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 2.1  [s]
11:50:26 Shutter 0: Real Pos: 13800, Target 9800, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 3.1  [s]
11:50:27 Shutter 0: Real Pos: 11800, Target 9800, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 4.2  [s]
11:50:28 Shutter 0: Real Pos. 9800, Stoppos: 50, relay: 1, direction -1, pulsetimer: 0, rtcshutter: 5.1 [s], operationtime 2
11:50:28 MQT: hm/status/sonoff/SHUTTER1 = 50 (retained)
11:50:28 SRC: Shutter
11:50:28 Switched relay: 2 by Shutter
11:50:28 MQT: hm/status/sonoff/RESULT = {"POWER2":"OFF"}
11:50:28 MQT: hm/status/sonoff/POWER2 = OFF (retained)
11:50:28 CFG: Saved to flash at 3F8, Count 1596, Bytes 3824
11:50:35 WIF: Checking connection...
11:50:35 WIF: Connected
```

stefanbode, 2018-12-06:
I still try to understand the log. My assumption is still a hick up in the calibration parameters. shutter50percent to 10 and then to 50 can fix this. But this is only a guess.

stefanbode, 2018-12-06:
I got it!!! You shutter works in the operation more ON/OFF OPEN/CLOSE. Relay 1 switches Power and Relay 2 the direction. For this case the Log is right. To switch to the other mode: setoption14 1
I assume setoption14 is still 0. Line 100ff declare how the mode will be determed

```
      // Determine shutter types
      if (Settings.flag3.paired_interlock) {
        if (Settings.pulse_timer[i] > 0) {
          shutterMode = PULSE_OPEN__PULSE_CLOSE;
        } else {
          shutterMode = OFF_OPEN__OFF_CLOSE;
        }
      } else {
        shutterMode = OFF_ON__OPEN_CLOSE;
      }
```

calcipher49, 2018-12-06:
Hi Stefan,
Thanks to investigate.

However SetOption14 is 1 on my config (checked and rechecked).

Based on your clue about code part, I can solve this issue replacing line 105 by:
`shutterMode = OFF_OPEN__OFF_CLOSE;`

So error is linked with Settings.flag3.paired_interlock at 0 in my case. But this value is not SetOption14

stefanbode, 2018-12-06:
ah, yes this is setoption81 I guess

stefanbode, 2018-12-06:
Please do not a code change. we have to investigate why this is wrong

calcipher49, 2018-12-06:
Yes of course, change code was only for test and help... I will wait for updated code

stefanbode, 2018-12-06:
Please can you check: setoption81 1
This should enable the paired interlock and then the shutter should be in the right mode.

calcipher49, 2018-12-06:
SetOption81 1 do not solve the issue.
I had already tried and I just checked again to confirm: No difference between SetOption81 0 and 1

I guess Settings.flag3.paired_interlock is linked with other parameter...

stefanbode, 2018-12-07:
the setoption81 requires a reboot to be active because the decision what kind of shutter it is will be made quite early in the code. I can double confirm, that this option is driving the paired interlock. I will make some code changes that will output additional logging that inform the user what shutter is used.

stefanbode, 2018-12-07:
I uploaded a new code fragment. please follow the instructions:
setoption81 1
upload new firmware
shutter50percent 49
shutter50percent 50
please write down the output and post it here in the issue Line should be something like:
`11:52:26 Shutter 0 (Relay:1): Init. Pos: 10200 [50 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 20000, Opentime 10.0 [s], Closetime 10.0 [s], CoedffCalc: c0: 400, c1 196, c2: 276, c3: 0, c4: 0, binmask 15, is inverted 1, shuttermode 0`

calcipher49, 2018-12-07:
Hi,

Thanks Stefan, new firmware solve the issue!!!

Below trace requested for commands sutter50percent:
`18:13:04 CMD: shutter50percent 49
18:13:04 Shutter accuracy digits: 1
18:13:04 Shutter 0 (Relay:1): Init. Pos: 0 [0 %], Open Vel.: 100 Close Vel.: 109 , Max Way: 46000, Opentime 23.0 [s], Closetime 21.0 [s], CoedffCalc: c0: -900, c1 469, c2: 289, c3: 0, c4: 0, binmask 3, is inverted 0, shuttermode 0
18:13:04 MQT: stat/sonos7/RESULT = {"shutter50percent1":49}
18:13:04 Shutter accuracy digits: 1
18:13:04 Shutter 0 (Relay:1): Init. Pos: 0 [0 %], Open Vel.: 100 Close Vel.: 109 , Max Way: 46000, Opentime 23.0 [s], Closetime 21.0 [s], CoedffCalc: c0: -900, c1 469, c2: 289, c3: 0, c4: 0, binmask 3, is inverted 0, shuttermode 0
18:13:05 MQT: stat/sonos7/RESULT = {"shutter50percent1":49}
18:13:16 CMD: shutter50percent 50
18:13:16 Shutter accuracy digits: 1
18:13:16 Shutter 0 (Relay:1): Init. Pos: 0 [0 %], Open Vel.: 100 Close Vel.: 109 , Max Way: 46000, Opentime 23.0 [s], Closetime 21.0 [s], CoedffCalc: c0: 0, c1 460, c2: 460, c3: 0, c4: 0, binmask 3, is inverted 0, shuttermode 0
18:13:16 MQT: stat/sonos7/RESULT = {"shutter50percent1":50}
18:13:16 Shutter accuracy digits: 1
18:13:16 Shutter 0 (Relay:1): Init. Pos: 0 [0 %], Open Vel.: 100 Close Vel.: 109 , Max Way: 46000, Opentime 23.0 [s], Closetime 21.0 [s], CoedffCalc: c0: 0, c1 460, c2: 460, c3: 0, c4: 0, binmask 3, is inverted 0, shuttermode 0
18:13:16 MQT: stat/sonos7/RESULT = {"shutter50percent1":50}`

closed 2018-12-08 by stefanbode