stefanbode/Sonoff-Tasmota #96: Shutter Feature - PR to TASMOTA Main?

meingraham, 2018-12-05:
@stefanbode - has there been any dialog with Theo to merge your roller shutter mods into the main development of TASMOTA?

Regards.

Mike

stefanbode, 2018-12-05:
Hi Mike, yes there is and I'm pretty sure we will get it back. We agreed that as long as the code is volatile and has bugs and need improvements this will be done in this fork. as soon as it gets more stable and fewer changes we move over. The main problem is, we need some code changes in the "master code" to get it work. It not just a simple extension in a file. e.g. the enhanced Interlock mode is essential for the shutter. Please be patient. My expectation is somewhere in spring 2019. Changes are definitely going down.

meingraham, 2018-12-05:
Thanks Stefan. Patience is my middle name (not!) ;-) It's actually not a rush for me. I'm still trying to find a more economical way to deploy shades with wireless control in my house. I have 16 shades I want to install and I'm looking for a mostly DIY approach. Most want a very nice premium for wireless controls... which are typically RF. I want Wi-Fi/MQTT. I've seen some DIY retrofits for the motorization too... but that may be beyond my abilities. But, nothing ventured, nothing gained.

I'll go ahead and close this thread and just monitor back in the spring.

Cheers!

Mike

closed 2018-12-05 by meingraham