stefanbode/Sonoff-Tasmota #97: using 4 wall switch with 8 relay board

jzamero, 2018-12-08:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**
Hi
Great work
can you explain please how do i connect 4 wall switch when using 8 relay board.
i use the wemos d1 mini 
and i want to control 4 shutters 
thank you

closed 2018-12-14 by jzamero

stefanbode, 2018-12-14:
@jzamero : sorry for not replying. 8 relays is fine with a pcf8574 board, but you get into trouble also to connect the 8 switches. This is no supported as fas I can see today. therefore I would not recommend to go with 4 shutter on ONE esp

jzamero, 2018-12-14:
Thank you
I saw in the wiki that you can connect with pcf8584 up to 64 io
If so why can’t I connect 8+8 relays and switches

stefanbode, 2018-12-17:
The problem is that my extension of the PCF still not support switches. I want to do this but delayed ti all the time

jzamero, 2018-12-17:
Thank you
I Will wait you are doing excellent job 👏