stefanbode/Sonoff-Tasmota #99: Compile Error

helmar74, 2018-12-13:
I tried to compile  this version, and got the following error:
Arduino: 1.8.8 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, ck, 26 MHz, 40MHz, DOUT, 1M (no SPIFFS), 2, v2 Higher Bandwidth, Disabled, None, Only Sketch, 115200"

WARNUNG: Bibliothek NewPing-1.9.1 behauptet auf (avr, arm) Architektur(en) ausgeführt werden zu können und ist möglicherweise inkompatibel mit Ihrem derzeitigen Board, welches auf (esp8266) Architektur(en) ausgeführt wird.
U:\02_daten\27_Programmierung\04_Arduino_ESP8266\06_Projekte_8266_ESPxx\13_SONOFF_Tasmota\04_SonoffAkku\Sonoff-Tasmota-master\sonoffAkku\sonoffAkku.ino: In function 'void MqttDataHandler(char*, byte*, unsigned int)':

sonoffAkku:983:84: error: 'COUNTERDEVIDER' was not declared in this scope

         Settings.pulse_devider[index -1] = Settings.pulse_devider[index -1] == 0 ? COUNTERDEVIDER : Settings.pulse_devider[index -1];

                                                                                    ^

sonoffAkku:1014:84: error: 'COUNTERDEVIDER' was not declared in this scope

         Settings.pulse_devider[index -1] = Settings.pulse_devider[index -1] == 0 ? COUNTERDEVIDER : Settings.pulse_devider[index -1];

                                                                                    ^

U:\02_daten\27_Programmierung\04_Arduino_ESP8266\06_Projekte_8266_ESPxx\13_SONOFF_Tasmota\04_SonoffAkku\Sonoff-Tasmota-master\sonoffAkku\settings.ino: In function 'void SettingsDefaultSet2()':

settings:493:46: error: 'MQTT_BUTTON_SWITCH_FORCE_LOCAL' was not declared in this scope

   Settings.flag3.button_switch_force_local = MQTT_BUTTON_SWITCH_FORCE_LOCAL;

                                              ^

settings:576:39: error: 'PRESSURE_CONVERSION' was not declared in this scope

   Settings.flag.pressure_conversion = PRESSURE_CONVERSION;

                                       ^

settings:586:36: error: 'CALC_RESOLUTION' was not declared in this scope

   Settings.flag2.calc_resolution = CALC_RESOLUTION;

                                    ^

U:\02_daten\27_Programmierung\04_Arduino_ESP8266\06_Projekte_8266_ESPxx\13_SONOFF_Tasmota\04_SonoffAkku\Sonoff-Tasmota-master\sonoffAkku\settings.ino: In function 'void SettingsDelta()':

settings:871:40: error: 'CALC_RESOLUTION' was not declared in this scope

       Settings.flag2.calc_resolution = CALC_RESOLUTION;

                                        ^

U:\02_daten\27_Programmierung\04_Arduino_ESP8266\06_Projekte_8266_ESPxx\13_SONOFF_Tasmota\04_SonoffAkku\Sonoff-Tasmota-master\sonoffAkku\xdrv_05_irremote.ino: In function 'void IrReceiveInit()':

xdrv_05_irremote:99:41: error: 'IR_RCV_BUFFER_SIZE' was not declared in this scope

   irrecv = new IRrecv(pin[GPIO_IRRECV], IR_RCV_BUFFER_SIZE, IR_RCV_TIMEOUT, IR_RCV_SAVE_BUFFER);

                                         ^

xdrv_05_irremote:99:61: error: 'IR_RCV_TIMEOUT' was not declared in this scope

   irrecv = new IRrecv(pin[GPIO_IRRECV], IR_RCV_BUFFER_SIZE, IR_RCV_TIMEOUT, IR_RCV_SAVE_BUFFER);

                                                             ^

xdrv_05_irremote:100:11: error: 'class IRrecv' has no member named 'setUnknownThreshold'

   irrecv->setUnknownThreshold(IR_RCV_MIN_UNKNOWN_SIZE);

           ^

xdrv_05_irremote:100:31: error: 'IR_RCV_MIN_UNKNOWN_SIZE' was not declared in this scope

   irrecv->setUnknownThreshold(IR_RCV_MIN_UNKNOWN_SIZE);

[... 74 lines cut ...]
 Nicht benutzt: D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_Akku_242\libraries\Adafruit-GFX-Library-1.2.9
Mehrere Bibliotheken wurden für "Adafruit_ILI9341.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_242\sketchbook\libraries\Adafruit_ILI9341-1.2.0-Tasmota-1.0
 Nicht benutzt: D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_Akku_242\libraries\Adafruit_ILI9341-1.2.0-Tasmota-1.0
Mehrere Bibliotheken wurden für "IRremoteESP8266.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_242\sketchbook\libraries\IRremoteESP8266-2.2.1.02
 Nicht benutzt: D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_Akku_242\libraries\IRremoteESP8266-2.5.2.03
Mehrere Bibliotheken wurden für "LiquidCrystal_I2C.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_242\sketchbook\libraries\LiquidCrystal_I2C-1.1.3
 Nicht benutzt: D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_Akku_242\libraries\LiquidCrystal_I2C-1.1.3
Mehrere Bibliotheken wurden für "Adafruit_SSD1306.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_242\sketchbook\libraries\Adafruit_SSD1306-1.1.2
 Nicht benutzt: D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_Akku_242\libraries\Adafruit_SSD1306-1.1.2
Mehrere Bibliotheken wurden für "Adafruit_SGP30.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_242\sketchbook\libraries\Adafruit_SGP30-1.0.0.13
 Nicht benutzt: D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_Akku_242\libraries\Adafruit_SGP30-1.0.0.13
exit status 1
'COUNTERDEVIDER' was not declared in this scope

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.

Compiling of "normal" Tasmota Version worked well.
Anything I can do?

stefanbode, 2018-12-13:
this is defined in the file my_user_config.h. seems to be that you have deleted or excluded this file.

helmar74, 2018-12-14:
Hi, thanks a lot. It was my error. I "took over" an existing my_user_config.h from normal Tasmota version.
Now i unzipped File and done configuration manually. 
Now it works.

Thanks a lot for your help and your great work!

(Im trying to setup a battery based ultra sonic sensor (HC-SR04).
The Device should sleep 3600 sec, wake up and should send the measured distance and after that go to sleep for again 3600 sec.)

Is it possible to do this with you version?
Is there something I should keep in mind, for this function?

stefanbode, 2018-12-14:
Yes have this up and running. TELELERIOD=10, DEEPSLEEP=3600 should do the job. The HC_SR04 needs 5V. Just keep in mind. with 3.7 V it might not wor correctly

helmar74, 2018-12-14:
Thanks for the information. Is the device battery buffered?
If so, which battery are you using?
(Yes I know, that the SR04 needs 5V)

stefanbode, 2018-12-19:
For UltraLowPower I use a bare ESP8266 with a 3.3V voltage regulator with very small current like the HT7333-A. Then power up this with one Li-Ion battery supporting between 4.2V - 3.7V. Any devices like sensors must be disconnected from power two. Best using one GPIO as relay and disconnect ground with a small transistor like the BC337-40. You cannot do 5V with this.

helmar74, 2018-12-19:
Sorry for disrupting you again.
Now I have an other strange behaviour.
I set up the Wemos D Mini with a HC-SR04 connected. Teleperiod is set to 10 seconds.
I receive the distance every 10 seconds.
After using "deepsleep 15" command, the Device wakesup every 15 seconds.
... but it doesn't send the distance any more.
After setting deepsleep to 0, I recognized, that the configuration I've done before is lost.
(Type of the device is set default, sensor configuration is gone.

Here's status 0 after losing configuration
`14:41:25 MQT: stat/sonoffWemosMini_8/RESULT = {"DeepSleep":"0 (0)"}
14:41:27 MQT: tele/sonoffWemosMini_8/STATE = {"Time":"2018-12-19T14:41:27","Uptime":"0T00:00:14","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":2,"SSId":"MYSSID_EG","BSSId":"9C:C7:A6:F7:72:A2","Channel":6,"RSSI":90},"DeepSleep":0,"Heap":19648}
14:41:27 CMD: status 0
14:41:27 MQT: stat/sonoffWemosMini_8/STATUS = {"Status":{"Module":1,"FriendlyName":["WemosMini-8-Akku"],"Topic":"sonoffWemosMini_8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:41:27 MQT: stat/sonoffWemosMini_8/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffAkkuGruppe","OtaUrl":"http://xxx.xxx.xxx.50:8080/static/firmware/sonoffAkku.ino.bin","RestartReason":"Deep-Sleep Wake","Uptime":"0T00:00:14","StartupUTC":"2018-12-19T13:41:13","Sleep":50,"BootCount":453,"SaveCount":520,"SaveAddress":"F5000"}}
14:41:27 MQT: stat/sonoffWemosMini_8/STATUS2 = {"StatusFWR":{"Version":"6.3.0.16 stb-1.4(sonoff)","BuildDateTime":"2018.12.19 14:28:39","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
14:41:27 MQT: stat/sonoffWemosMini_8/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MYSSIDMYSSID2","MYSSID_EG"],"TelePeriod":10,"SetOption":["00008009","558180C0","00000000"]}}
14:41:27 MQT: stat/sonoffWemosMini_8/STATUS4 = {"StatusMEM":{"ProgramSize":549,"Free":452,"Heap":17840,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":1458415,"Features":["00000003","00000407","0FCAE794","000383A0","23BE1FCC"]}}
14:41:27 MQT: stat/sonoffWemosMini_8/STATUS5 = {"StatusNET":{"Hostname":"sonoffWemosMini-8-Akku","IPAddress":"xxx.xxx.xxx.124","Gateway":"xxx.xxx.xxx.1","Subnetmask":"255.255.255.0","DNSServer":"xxx.xxx.xxx.2","Mac":"84:F3:EB:3B:79:93","Webserver":2,"WifiConfig":4}}
14:41:27 MQT: stat/sonoffWemosMini_8/STATUS6 = {"StatusMQT":{"MqttHost":"xxx.xxx.xxx.50","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3B7993","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
14:41:27 MQT: stat/sonoffWemosMini_8/STATUS7 = {"StatusTIM":{"UTC":"Wed Dec 19 13:41:27 2018","Local":"Wed Dec 19 14:41:27 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"07:57","Sunset":"16:14"}}
14:41:27 MQT: stat/sonoffWemosMini_8/STATUS10 = {"StatusSNS":{"Time":"2018-12-19T14:41:27"}}
14:41:28 MQT: stat/sonoffWemosMini_8/STATUS11 = {"StatusSTS":{"Time":"2018-12-19T14:41:27","Uptime":"0T00:00:14","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":2,"SSId":"MYSSID_EG","BSSId":"9C:C7:A6:F7:72:A2","Channel":6,"RSSI":88},"DeepSleep":0,"Heap":16200}}`

And here status 0 before sending to deepsleep 15:
`14:44:31 MQT: stat/sonoffWemosMini_8/STATUS = {"Status":{"Module":18,"FriendlyName":["WemosMini-8-Akku"],"Topic":"sonoffWemosMini_8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:44:34 MQT: tele/sonoffWemosMini_8/STATE = {"Time":"2018-12-19T14:44:34","Uptime":"0T00:00:17","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":2,"SSId":"xxxxxxxxxx_EG","BSSId":"9C:C7:A6:F7:72:A2","Channel":6,"RSSI":92},"DeepSleep":0,"Heap":19592}
14:44:34 MQT: tele/sonoffWemosMini_8/SENSOR = {"Time":"2018-12-19T14:44:34","ANALOG":{"A0":8},"SR04":{"Distanz":26.193}}
14:44:35 CMD: status 0
14:44:35 MQT: stat/sonoffWemosMini_8/STATUS = {"Status":{"Module":18,"FriendlyName":["WemosMini-8-Akku"],"Topic":"sonoffWemosMini_8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:44:35 MQT: stat/sonoffWemosMini_8/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffAkkuGruppe","OtaUrl":"http://xxx.xxx.xxx.50:8080/static/firmware/sonoffAkku.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:18","StartupUTC":"2018-12-19T13:44:17","Sleep":50,"BootCount":455,"SaveCount":524,"SaveAddress":"F9000"}}
14:44:35 MQT: stat/sonoffWemosMini_8/STATUS2 = {"StatusFWR":{"Version":"6.3.0.16 stb-1.4(sonoff)","BuildDateTime":"2018.12.19 14:28:39","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
14:44:35 MQT: stat/sonoffWemosMini_8/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxxxxxx","xxxxxxxxxx_EG"],"TelePeriod":10,"SetOption":["00008009","558180C0","00000000"]}}
14:44:35 MQT: stat/sonoffWemosMini_8/STATUS4 = {"StatusMEM":{"ProgramSize":549,"Free":452,"Heap":17712,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":1458415,"Features":["00000003","00000407","0FCAE794","000383A0","23BE1FCC"]}}
[... cut ...]
14:44:44 MQT: tele/sonoffWemosMini_8/SENSOR = {"Time":"2018-12-19T14:44:44","ANALOG":{"A0":8},"SR04":{"Distanz":26.193}}
14:44:54 MQT: tele/sonoffWemosMini_8/STATE = {"Time":"2018-12-19T14:44:54","Uptime":"0T00:00:37","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":2,"SSId":"xxxxxxxxxx_EG","BSSId":"9C:C7:A6:F7:72:A2","Channel":6,"RSSI":90},"DeepSleep":0,"Heap":19592}
14:44:54 MQT: tele/sonoffWemosMini_8/SENSOR = {"Time":"2018-12-19T14:44:54","ANALOG":{"A0":8},"SR04":{"Distanz":26.193}}`

Anything I can do?

stefanbode, 2018-12-19:
Please connect the - (minus) of the HR04 to the collector of a transistor. e.g. BC337-40 and define a Relay1 to turn on the transistor. This should initialize the HR04 correctly so you get a clear startup and measurement.

helmar74, 2018-12-22:
I connected a BC548 (with Basis Resistor 1KOhm) to a Pin, defined as Relay1, turned "Power1 ON".
Here's the configuration of the Device:
[image]

Here's the console log before entering deepsleep:
`10:52:13 MQT: tele/sonoffWemosMini_8/STATE = {"Time":"2018-12-22T10:52:13","Uptime":"0T00:03:08","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":2,"SSId":"hs.net_EG","BSSId":"E8:DF:70:67:B9:00","Channel":6,"RSSI":72},"DeepSleep":0,"Heap":19424}
10:52:13 MQT: tele/sonoffWemosMini_8/SENSOR = {"Time":"2018-12-22T10:52:13","ANALOG":{"A0":8},"SR04":{"Distanz":16.246}}
10:52:23 MQT: tele/sonoffWemosMini_8/STATE = {"Time":"2018-12-22T10:52:23","Uptime":"0T00:03:18","SleepMode":"Dynamic","Sleep":50,"LoadAvg":35,"POWER":"ON","Wifi":{"AP":2,"SSId":"hs.net_EG","BSSId":"E8:DF:70:67:B9:00","Channel":6,"RSSI":70},"DeepSleep":0,"Heap":19568}
10:52:23 MQT: tele/sonoffWemosMini_8/SENSOR = {"Time":"2018-12-22T10:52:23","ANALOG":{"A0":8},"SR04":{"Distanz":16.228}}
10:52:33 MQT: tele/sonoffWemosMini_8/STATE = {"Time":"2018-12-22T10:52:33","Uptime":"0T00:03:28","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":2,"SSId":"hs.net_EG","BSSId":"E8:DF:70:67:B9:00","Channel":6,"RSSI":70},"DeepSleep":0,"Heap":19424}
10:52:33 MQT: tele/sonoffWemosMini_8/SENSOR = {"Time":"2018-12-22T10:52:33","ANALOG":{"A0":8},"SR04":{"Distanz":16.228}}
10:52:43 MQT: tele/sonoffWemosMini_8/STATE = {"Time":"2018-12-22T10:52:43","Uptime":"0T00:03:38","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":2,"SSId":"hs.net_EG","BSSId":"E8:DF:70:67:B9:00","Channel":6,"RSSI":70},"DeepSleep":0,"Heap":19496}
10:52:43 MQT: tele/sonoffWemosMini_8/SENSOR = {"Time":"2018-12-22T10:52:43","ANALOG":{"A0":8},"SR04":{"Distanz":16.211}}
10:52:53 MQT: tele/sonoffWemosMini_8/STATE = {"Time":"2018-12-22T10:52:53","Uptime":"0T00:03:48","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":2,"SSId":"hs.net_EG","BSSId":"E8:DF:70:67:B9:00","Channel":6,"RSSI":62},"DeepSleep":0,"Heap":16000}
10:52:53 MQT: tele/sonoffWemosMini_8/SENSOR = {"Time":"2018-12-22T10:52:53","ANALOG":{"A0":8},"SR04":{"Distanz":16.246}}
10:53:03 MQT: tele/sonoffWemosMini_8/STATE = {"Time":"2018-12-22T10:53:03","Uptime":"0T00:03:58","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":2,"SSId":"hs.net_EG","BSSId":"E8:DF:70:67:B9:00","Channel":6,"RSSI":68},"DeepSleep":0,"Heap":19496}
10:53:03 MQT: tele/sonoffWemosMini_8/SENSOR = {"Time":"2018-12-22T10:53:03","ANALOG":{"A0":8},"SR04":{"Distanz":16.246}}
10:53:13 MQT: tele/sonoffWemosMini_8/STATE = {"Time":"2018-12-22T10:53:13","Uptime":"0T00:04:08","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":2,"SSId":"hs.net_EG","BSSId":"E8:DF:70:67:B9:00","Channel":6,"RSSI":68},"DeepSleep":0,"Heap":19424}
10:53:13 MQT: tele/sonoffWemosMini_8/SENSOR = {"Time":"2018-12-22T10:53:13","ANALOG":{"A0":8},"SR04":{"Distanz":16.246}}
10:53:23 MQT: tele/sonoffWemosMini_8/STATE = {"Time":"2018-12-22T10:53:23","Uptime":"0T00:04:18","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":2,"SSId":"hs.net_EG","BSSId":"E8:DF:70:67:B9:00","Channel":6,"RSSI":62},"DeepSleep":0,"Heap":16072}
10:53:23 MQT: tele/sonoffWemosMini_8/SENSOR = {"Time":"2018-12-22T10:53:23","ANALOG":{"A0":8},"SR04":{"Distanz":16.228}}
10:53:33 MQT: tele/sonoffWemosMini_8/STATE = {"Time":"2018-12-22T10:53:33","Uptime":"0T00:04:28","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":2,"SSId":"hs.net_EG","BSSId":"E8:DF:70:67:B9:00","Channel":6,"RSSI":72},"DeepSleep":0,"Heap":19424}
10:53:33 MQT: tele/sonoffWemosMini_8/SENSOR = {"Time":"2018-12-22T10:53:33","ANALOG":{"A0":8},"SR04":{"Distanz":16.228}}
10:53:43 MQT: tele/sonoffWemosMini_8/STATE = {"Time":"2018-12-22T10:53:43","Uptime":"0T00:04:38","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":2,"SSId":"hs.net_EG","BSSId":"E8:DF:70:67:B9:00","Channel":6,"RSSI":66},"DeepSleep":0,"Heap":19496}
10:53:43 MQT: tele/sonoffWemosMini_8/SENSOR = {"Time":"2018-12-22T10:53:43","ANALOG":{"A0":8},"SR04":{"Distanz":16.246}}`

I entered `deepsleep 15`:
`deepsleep 15`

Here's the console log after wakeup:
`00:00:00 Projekt sonoffAkku WemosMini-8-Akku (Topic sonoffWemosMini_8, Fallback DVES_3B7993, GroupTopic sonoffAkkuGruppe) Version 6.3.0.16 stb-1.4(sonoff)-2_4_2
00:00:00 WIF: verbinden mit AP2 xxxxxxxx in Modus 11N wie sonoffWemosMini-8-Akku...
00:00:08 WIF: verbunden
00:00:08 DNS: initialisiert
00:00:08 HTP: Web-Server aktiv bei sonoffWemosMini-8-Akku.local mit IP-Adresse 192.168.1.124
10:56:42 MQT: Verbindungsversuch...
10:56:42 MQT: verbunden
10:56:42 MQT: tele/sonoffWemosMini_8/LWT = Online (beibehalten)
10:56:42 MQT: cmnd/sonoffWemosMini_8/POWER = 
10:56:42 MQT: tele/sonoffWemosMini_8/INFO1 = {"Module":"Sonoff Basic","Version":"6.3.0.16 stb-1.4(sonoff)","FallbackTopic":"DVES_3B7993","GroupTopic":"sonoffAkkuGruppe"}
10:56:42 MQT: tele/sonoffWemosMini_8/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoffWemosMini-8-Akku","IPAddress":"192.168.1.124"}
10:56:42 MQT: tele/sonoffWemosMini_8/INFO3 = {"RestartReason":"Deep-Sleep Wake"}
10:56:42 MQT: stat/sonoffWemosMini_8/RESULT = {"POWER":"ON"}
10:56:42 MQT: stat/sonoffWemosMini_8/POWER = ON
.
.
.
00:00:00 Projekt sonoffAkku WemosMini-8-Akku (Topic sonoffWemosMini_8, Fallback DVES_3B7993, GroupTopic sonoffAkkuGruppe) Version 6.3.0.16 stb-1.4(sonoff)-2_4_2
00:00:00 WIF: verbinden mit AP2 hs.net_EG in Modus 11N wie sonoffWemosMini-8-Akku...
[... cut ...]
10:58:40 MQT: tele/sonoffWemosMini_8/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoffWemosMini-8-Akku","IPAddress":"192.168.1.124"}
10:58:40 MQT: tele/sonoffWemosMini_8/INFO3 = {"RestartReason":"Deep-Sleep Wake"}
10:58:40 MQT: stat/sonoffWemosMini_8/RESULT = {"POWER":"ON"}
10:58:40 MQT: stat/sonoffWemosMini_8/POWER = ON`

As you can see, the distance isn't shown anymore in console log. The complete configuration is also "gone" (Picture shows configuration after entering deepsleep for first time):
[image]

stefanbode, 2019-01-16:
Have you found a solution for this. As mentioned it worked on my wemos Mini devices without any problems. Also the generic esp8266 work. Therefore current assumption is that something in your hardware wire is strange. Will not be easy to find

helmar74, 2019-01-24:
Hi again,

I compiled you new version, the Wemos Mini D and also a bare ESP8266 are now waking up properly and sending the distance. -> GREAT JOB. Many Thanks for that!!!

My next problem ist the power draining.
With connected HC-SR04 the ESP and the Sensor consumes 3mA of current, which is too much for a battery powered device.
When I disconnect HC-SR04 Sensor, the current went down to a few µA.
So my question is: How can I reduce the current usage during deepsleep, with connected HC-SR04?

Thank's a lot

Helmar

stefanbode, 2019-01-24:
Take a look at the end of: https://github.com/stefanbode/Sonoff-Tasmota/wiki/Deepsleep
This should shutdown the HC-SR04. Please connect o a pin, that goes down on deepsleep (D8)

helmar74, 2019-01-30:
Hi,
finally i got it working, as expected. I've set teleperiod to 15 seconds and deepsleep to 300.
But after a certain time the Wemos Mini doesn't wake up anymore:
```
tele/WemosMiniAkku_1/SENSOR {"Time":"2019-01-29T18:59:10","ANALOG":{"A0":980},"DS18B20":{"Temperature":17.5},"SR04":{"Distanz":5.316},"TempU                                 nit":"C"}
tele/WemosMiniAkku_1/SENSOR {"Time":"2019-01-29T19:04:12","ANALOG":{"A0":980},"DS18B20":{"Temperature":17.3},"SR04":{"Distanz":5.316},"TempU                                 nit":"C"}
tele/WemosMiniAkku_1/SENSOR {"Time":"2019-01-29T19:09:14","ANALOG":{"A0":980},"DS18B20":{"Temperature":17.6},"SR04":{"Distanz":5.316},"TempU                                 nit":"C"}
tele/WemosMiniAkku_1/SENSOR {"Time":"2019-01-29T19:14:14","ANALOG":{"A0":979},"DS18B20":{"Temperature":17.4},"SR04":{"Distanz":5.316},"TempU                                 nit":"C"}
tele/WemosMiniAkku_1/SENSOR {"Time":"2019-01-29T19:19:16","ANALOG":{"A0":979},"DS18B20":{"Temperature":17.4},"SR04":{"Distanz":5.316},"TempU                                 nit":"C"}
tele/WemosMiniAkku_1/SENSOR {"Time":"2019-01-29T19:24:20","ANALOG":{"A0":978},"DS18B20":{"Temperature":17.4},"SR04":{"Distanz":5.316},"TempU                                 nit":"C"}
tele/WemosMiniAkku_1/SENSOR {"Time":"2019-01-29T19:29:24","ANALOG":{"A0":976},"DS18B20":{"Temperature":16.8},"SR04":{"Distanz":5.316},"TempU                                 nit":"C"}
tele/WemosMiniAkku_1/SENSOR {"Time":"2019-01-29T19:34:26","ANALOG":{"A0":974},"DS18B20":{"Temperature":16.9},"SR04":{"Distanz":5.316},"TempU                                 nit":"C"}
tele/WemosMiniAkku_1/SENSOR {"Time":"2019-01-29T19:39:28","ANALOG":{"A0":975},"DS18B20":{"Temperature":17.4},"SR04":{"Distanz":5.316},"TempU                                 nit":"C"}
tele/WemosMiniAkku_1/SENSOR {"Time":"2019-01-29T19:44:31","ANALOG":{"A0":975},"DS18B20":{"Temperature":17.4},"SR04":{"Distanz":5.316},"TempU                                 nit":"C"}
tele/WemosMiniAkku_1/SENSOR {"Time":"2019-01-29T19:49:33","ANALOG":{"A0":975},"DS18B20":{"Temperature":17.4},"SR04":{"Distanz":5.316},"TempU                                 nit":"C"}
tele/WemosMiniAkku_1/SENSOR {"Time":"2019-01-29T19:54:39","ANALOG":{"A0":974},"DS18B20":{"Temperature":16.8},"SR04":{"Distanz":5.316},"TempU                                 nit":"C"}
tele/WemosMiniAkku_1/SENSOR {"Time":"2019-01-29T19:59:40","ANALOG":{"A0":975},"DS18B20":{"Temperature":17.4},"SR04":{"Distanz":5.333},"TempU                                 nit":"C"}
tele/WemosMiniAkku_1/SENSOR {"Time":"2019-01-29T20:04:44","ANALOG":{"A0":974},"DS18B20":{"Temperature":17.5},"SR04":{"Distanz":5.316},"TempU                                 nit":"C"}
tele/WemosMiniAkku_1/SENSOR {"Time":"2019-01-29T20:09:46","ANALOG":{"A0":974},"DS18B20":{"Temperature":17.6},"SR04":{"Distanz":5.316},"TempU                                 nit":"C"}
tele/WemosMiniAkku_1/SENSOR {"Time":"2019-01-29T20:14:47","ANALOG":{"A0":974},"DS18B20":{"Temperature":17.4},"SR04":{"Distanz":5.316},"TempU                                 nit":"C"}
tele/WemosMiniAkku_1/SENSOR {"Time":"2019-01-29T20:19:48","ANALOG":{"A0":974},"DS18B20":{"Temperature":17.5},"SR04":{"Distanz":5.316},"TempU                                 nit":"C"}
tele/WemosMiniAkku_1/SENSOR {"Time":"2019-01-29T20:24:50","ANALOG":{"A0":974},"DS18B20":{"Temperature":17.5},"SR04":{"Distanz":5.333},"TempU                                 nit":"C"}
tele/WemosMiniAkku_1/SENSOR {"Time":"2019-01-29T20:29:50","ANALOG":{"A0":974},"DS18B20":{"Temperature":17.4},"SR04":{"Distanz":5.316},"TempUnit":"C"}
tele/WemosMiniAkku_1/SENSOR {"Time":"2019-01-29T20:34:51","ANALOG":{"A0":974},"DS18B20":{"Temperature":17.4},"SR04":{"Distanz":5.316},"TempUnit":"C"}
---> Now more messages sent <---
```
I powered off an on, than it was again working for about 1,5h. Than the same error occured.
Anything what I can do?
Here're the circuit and the settings:
[image]

[image]

stefanbode, 2019-01-30:
For me it looks like an electric problem. I would increase C4 to something about 470Micro Farad from 1. or even higher. I always use my 1000 to get the spikes of the ESP under control.

helmar74, 2019-01-30:
Are you using Elkos?

stefanbode, 2019-01-30:
Yes for sure. They only cost cent.

helmar74, 2019-01-30:
OK, will try that. Thanks a lot!
PS: 
Current usage in Deepsleep 70µA, when up 70-80mA.
Could you confirm these current usage?

stefanbode, 2019-01-30:
Yes this is ok. Anyhow if you connect an Oscilloscope during startup you will see spikes with 200mA, maybe more. I assume the voltage regulator cause the issue and the voltage drop below 3V for a mS or so. But this will go away directly with the elko.

helmar74, 2019-01-30:
Hi again,

I'm still running with 1µF condensator, ordered 1000µF/6,3V ELKOS.... waiting for shipping.

I had a deeper look on the behavior. Here's a summary of the send data:
[image]

What you can see is, that sometimes the data isn't send after about 75 seconds to the broker (as it should while deepsleep is 60 and teleperiod is 15).
Sometimes it takes double time ~150 seconds.

I will replace condensator, log data again and give you feedback.

stefanbode, 2019-01-30:
If you are german and write me to stefan_bode at Web.de i will send you 10 per post. So get it up and running at weekend

stefanbode, 2019-01-31:
Ok, but did you still havevthe problem that it will die forever?

stefanbode, 2019-01-31:
Please can can you also gather the uptime that is reported. Maybe it is a wlan issue. Also seen before. In the case it wakes up but could not connect an goto sleep again. You can detect this by much higher uptime_s reported to MQTT.

stefanbode, 2019-01-31:
The uptime will only be resettled to 0. if the value was successfully transmitted to MQTT. Also reboots do not set it to 0.

helmar74, 2019-01-31:
Maybe I found a solution. I increased teleperiod to 30. Deepsleep is 60.
Now I don't have any topics which aren't received by broker.
I timed with stopwatch:
20 seconds up  (~75mA)
60 seconds deepsleep (~70µA)

Will monitor a time longer.... and give feedback.

helmar74, 2019-01-31:
As a result of setting teleperiod 10 and deepsleep 60 , the wemos doesn't awake anymore...
Trying higher values for teleperiod...

helmar74, 2019-01-31:
After reflashing (no other way to get it run again), set teleperiod to 30, deepsleep to 60 it works again...

helmar74, 2019-02-03:
I still have the issue, the sometimes the MQTT Topic isn't send to Broker.
The Wemos is located 1 Meter neraby Accesspoint.. RSSI is 90 or 100:
[image]

helmar74, 2019-02-03:
A new Wemos (with no sensor connected and USB Powered, has the same issue):
[image]

helmar74, 2019-02-03:
Error is present again after restarting broker. I think it's an Software bug of you Tasmota fork.
Any possiblity to figure out?

helmar74, 2019-02-03:
I activated RemoteSyslog Here's the Log.
The Broker received the "State" Topic at 15:11:18 and at 15:13:47.
But the topic wasn't  received at 15:12:30 :

`3>1 2019-02-03T15:11:08.392969+01:00 sonoffAkku-5572 ESP-WIF - - - ESP-WIF: verbunden
<13>1 2019-02-03T15:11:08.392969+01:00 sonoffAkku-5572 ESP-HTP - - - ESP-HTP: Web-Server aktiv bei sonoffAkku-5572 mit IP-Adresse 192.168.1.155
<13>1 2019-02-03T15:11:08.628200+01:00 sonoffAkku-5572 ESP-APP - - - ESP-APP: (UTC) Sun Feb 03 14:11:08 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
<13>1 2019-02-03T15:11:09.591034+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: Verbindungsversuch...
<13>1 2019-02-03T15:11:09.617723+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: verbunden
<13>1 2019-02-03T15:11:09.623697+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/LWT = Online (beibehalten)
<13>1 2019-02-03T15:11:09.626226+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: cmnd/sonoffAkku/POWER = 
<13>1 2019-02-03T15:11:09.634062+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: abonniere cmnd/sonoffAkku/#
<13>1 2019-02-03T15:11:09.644560+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: abonniere cmnd/sonoffAkkuGruppe/#
<13>1 2019-02-03T15:11:09.693038+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: abonniere cmnd/DVES_0C35C4_fb/#
<13>1 2019-02-03T15:11:09.694439+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/INFO1 = {"Module":"Generic","Version":"6.4.1.9 stb-1.1(sonoff)","FallbackTopic":"cmnd/DVES_0C35C4_fb/","GroupTopic":"sonoffAkkuGruppe"}
<13>1 2019-02-03T15:11:09.698532+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoffAkku-5572","IPAddress":"192.168.1.155"}
<13>1 2019-02-03T15:11:09.704268+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/INFO3 = {"RestartReason":"Deep-Sleep Wake"}
<13>1 2019-02-03T15:11:09.887413+01:00 sonoffAkku-5572 ESP-APP - - - ESP-APP: Anzahl Startvorgänge 92
<13>1 2019-02-03T15:11:17.926738+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/STATE = {"Time":"2019-02-03T15:11:18","Uptime":"0T00:00:18","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"hs.net_EG","BSSId":"9C:C7:A6:F7:72:A2","Channel":6,"RSSI":94},"DeepSleep":60,"Heap":20936}
<13>1 2019-02-03T15:11:17.927031+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/SENSOR = {"Time":"2019-02-03T15:11:18","ANALOG":{"A0":0}}
<13>1 2019-02-03T15:11:17.927896+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/LWT = Offline
<13>1 2019-02-03T15:11:17.928982+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/UPTIME_S = {"Time":"2019-02-03T15:11:18", "Uptime_s":16}
<13>1 2019-02-03T15:12:21.005499+01:00 sonoffAkku-5572 ESP-WIF - - - ESP-WIF: verbunden
<13>1 2019-02-03T15:12:21.033223+01:00 sonoffAkku-5572 ESP-HTP - - - ESP-HTP: Web-Server aktiv bei sonoffAkku-5572 mit IP-Adresse 192.168.1.155
<13>1 2019-02-03T15:12:21.730709+01:00 sonoffAkku-5572 ESP-APP - - - ESP-APP: (UTC) Sun Feb 03 14:12:20 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
<13>1 2019-02-03T15:12:22.199821+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: Verbindungsversuch...
<13>1 2019-02-03T15:12:22.222216+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: verbunden
<13>1 2019-02-03T15:12:22.226847+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/LWT = Online (beibehalten)
<13>1 2019-02-03T15:12:22.228332+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: cmnd/sonoffAkku/POWER = 
<13>1 2019-02-03T15:12:22.229923+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: abonniere cmnd/sonoffAkku/#
<13>1 2019-02-03T15:12:22.232076+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: abonniere cmnd/sonoffAkkuGruppe/#
<13>1 2019-02-03T15:12:22.248717+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: abonniere cmnd/DVES_0C35C4_fb/#
<13>1 2019-02-03T15:12:22.268181+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/INFO1 = {"Module":"Generic","Version":"6.4.1.9 stb-1.1(sonoff)","FallbackTopic":"cmnd/DVES_0C35C4_fb/","GroupTopic":"sonoffAkkuGruppe"}
<13>1 2019-02-03T15:12:22.269050+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoffAkku-5572","IPAddress":"192.168.1.155"}
<13>1 2019-02-03T15:12:22.286876+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/INFO3 = {"RestartReason":"Deep-Sleep Wake"}
<13>1 2019-02-03T15:12:26.449225+01:00 sonoffAkku-5572 ESP-APP - - - ESP-APP: Anzahl Startvorgänge 93
<13>1 2019-02-03T15:12:26.758103+01:00 sonoffAkku-5572 ESP-CFG - - - ESP-CFG: in Flash gespeichert am F7, zählen 101, Bytes 3824
<13>1 2019-02-03T15:12:30.493491+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/STATE = {"Time":"2019-02-03T15:12:29","Uptime":"0T00:00:14","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"hs.net_EG","BSSId":"9C:C7:A6:F7:72:A2","Channel":6,"RSSI":72},"DeepSleep":60,"Heap":20936}
<13>1 2019-02-03T15:12:30.505250+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/SENSOR = {"Time":"2019-02-03T15:12:29","ANALOG":{"A0":0}}
<13>1 2019-02-03T15:12:30.509169+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/LWT = Offline
<13>1 2019-02-03T15:12:30.524130+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/UPTIME_S = {"Time":"2019-02-03T15:12:29", "Uptime_s":12}
<13>1 2019-02-03T15:13:37.280928+01:00 sonoffAkku-5572 ESP-WIF - - - ESP-WIF: verbunden
[... cut ...] "BSSId":"9C:C7:A6:F7:72:A2","Channel":6,"RSSI":90},"DeepSleep":60,"Heap":20936}
<13>1 2019-02-03T15:13:46.815063+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/SENSOR = {"Time":"2019-02-03T15:13:47","ANALOG":{"A0":0}}
<13>1 2019-02-03T15:13:46.816813+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/LWT = Offline
<13>1 2019-02-03T15:13:46.823657+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/UPTIME_S = {"Time":"2019-02-03T15:13:47", "Uptime_s":15}`

[image]

helmar74, 2019-02-03:
Here the relevant lines:
15:11:17:
<13>1 2019-02-03T15:11:17.926738+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/STATE = {"Time":"2019-02-03T15:11:18","Uptime":"0T00:00:18","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"hs.net_EG","BSSId":"9C:C7:A6:F7:72:A2","Channel":6,"RSSI":94},"DeepSleep":60,"Heap":20936}

15:12:30:
<13>1 2019-02-03T15:12:30.493491+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/STATE = {"Time":"2019-02-03T15:12:29","Uptime":"0T00:00:14","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"hs.net_EG","BSSId":"9C:C7:A6:F7:72:A2","Channel":6,"RSSI":72},"DeepSleep":60,"Heap":20936}

15:13:46:
<13>1 2019-02-03T15:13:46.779600+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/STATE = {"Time":"2019-02-03T15:13:47","Uptime":"0T00:00:18","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"hs.net_EG","BSSId":"9C:C7:A6:F7:72:A2","Channel":6,"RSSI":90},"DeepSleep":60,"Heap":20936}

helmar74, 2019-02-03:
Here's the Log of the wakeup, when the topic is missing on Broker:
<13>1 2019-02-03T15:12:21.005499+01:00 sonoffAkku-5572 ESP-WIF - - - ESP-WIF: verbunden
<13>1 2019-02-03T15:12:21.033223+01:00 sonoffAkku-5572 ESP-HTP - - - ESP-HTP: Web-Server aktiv bei sonoffAkku-5572 mit IP-Adresse 192.168.1.155
<13>1 2019-02-03T15:12:21.730709+01:00 sonoffAkku-5572 ESP-APP - - - ESP-APP: (UTC) Sun Feb 03 14:12:20 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
<13>1 2019-02-03T15:12:22.199821+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: Verbindungsversuch...
<13>1 2019-02-03T15:12:22.222216+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: verbunden
<13>1 2019-02-03T15:12:22.226847+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/LWT = Online (beibehalten)
<13>1 2019-02-03T15:12:22.228332+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: cmnd/sonoffAkku/POWER = 
<13>1 2019-02-03T15:12:22.229923+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: abonniere cmnd/sonoffAkku/#
<13>1 2019-02-03T15:12:22.232076+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: abonniere cmnd/sonoffAkkuGruppe/#
<13>1 2019-02-03T15:12:22.248717+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: abonniere cmnd/DVES_0C35C4_fb/#
<13>1 2019-02-03T15:12:22.268181+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/INFO1 = {"Module":"Generic","Version":"6.4.1.9 stb-1.1(sonoff)","FallbackTopic":"cmnd/DVES_0C35C4_fb/","GroupTopic":"sonoffAkkuGruppe"}
<13>1 2019-02-03T15:12:22.269050+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoffAkku-5572","IPAddress":"192.168.1.155"}
<13>1 2019-02-03T15:12:22.286876+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/INFO3 = {"RestartReason":"Deep-Sleep Wake"}
<13>1 2019-02-03T15:12:26.449225+01:00 sonoffAkku-5572 ESP-APP - - - ESP-APP: Anzahl Startvorgänge 93
<13>1 2019-02-03T15:12:26.758103+01:00 sonoffAkku-5572 ESP-CFG - - - ESP-CFG: in Flash gespeichert am F7, zählen 101, Bytes 3824
<13>1 2019-02-03T15:12:30.493491+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/STATE = {"Time":"2019-02-03T15:12:29","Uptime":"0T00:00:14","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"hs.net_EG","BSSId":"9C:C7:A6:F7:72:A2","Channel":6,"RSSI":72},"DeepSleep":60,"Heap":20936}
<13>1 2019-02-03T15:12:30.505250+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/SENSOR = {"Time":"2019-02-03T15:12:29","ANALOG":{"A0":0}}
<13>1 2019-02-03T15:12:30.509169+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/LWT = Offline
<13>1 2019-02-03T15:12:30.524130+01:00 sonoffAkku-5572 ESP-MQT - - - ESP-MQT: tele/sonoffAkku/UPTIME_S = {"Time":"2019-02-03T15:12:29", "Uptime_s":12}

helmar74, 2019-02-18:
I've setup a rsyslog server and activated it with syslog 4 on Tasmota device.
The device wakes up every 1800s reliable. It connects to mosquitto broker and start sending topics after teleperiod expires.
BUT: The broker doesn't receive  every topic. Here's the syslog from an "working" example (The SENSOR topic is received by broker) and a "non working" example, where the broker doesn't receive the topic:

Mosquitto receives Topic:
>Feb 18 05:23:31 WemosMiniAkku-1 ESP-HTP: Web-Server aktiv bei WemosMiniAkku-1 mit IP-Adresse 192.168.1.137
>Feb 18 05:23:31 WemosMiniAkku-1 ESP-APP: (UTC) Mon Feb 18 04:23:31 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
>Feb 18 05:23:32 WemosMiniAkku-1 ESP-MQT: Verbindungsversuch...
>Feb 18 05:23:32 WemosMiniAkku-1 ESP-MQT: verbunden
>Feb 18 05:23:32 WemosMiniAkku-1 ESP-MQT: tele/WemosMiniAkku_1/LWT = Online (beibehalten)
>Feb 18 05:23:32 WemosMiniAkku-1 ESP-MQT: cmnd/WemosMiniAkku_1/POWER =
>Feb 18 05:23:32 WemosMiniAkku-1 ESP-MQT: abonniere cmnd/WemosMiniAkku_1/#
>Feb 18 05:23:32 WemosMiniAkku-1 ESP-MQT: abonniere cmnd/sonoffAkkuGruppe/#
>Feb 18 05:23:32 WemosMiniAkku-1 ESP-MQT: abonniere cmnd/WemosMiniAkku-1_fb/#
>Feb 18 05:23:32 WemosMiniAkku-1 ESP-MQT: tele/WemosMiniAkku_1/INFO2 = {"WebServerMode":"Admin","Hostname":"WemosMiniAkku-1","IPAddress":"192.168.1.137"}
>Feb 18 05:23:32 WemosMiniAkku-1 ESP-MQT: tele/WemosMiniAkku_1/INFO3 = {"RestartReason":"Deep-Sleep Wake"}
>Feb 18 05:23:32 WemosMiniAkku-1 ESP-MQT: stat/WemosMiniAkku_1/RESULT = {"POWER":"ON"}
>Feb 18 05:23:32 WemosMiniAkku-1 ESP-MQT: stat/WemosMiniAkku_1/POWER = ON
>Feb 18 05:23:32 WemosMiniAkku-1 ESP-SRC: MQTT
>Feb 18 05:23:32 WemosMiniAkku-1 ESP-RSL: empfangenes topic cmnd/WemosMiniAkku_1/DeepSleep, Datengröße 4, Daten 1800
>Feb 18 05:23:32 WemosMiniAkku-1 ESP-RSL: Gruppe 0, Index 1, Befehl DEEPSLEEP, Daten 1800
>Feb 18 05:23:32 WemosMiniAkku-1 ESP-MQT: stat/WemosMiniAkku_1/RESULT = {"DeepSleep":"1800 (1800)"}
>Feb 18 05:23:40 WemosMiniAkku-1 ESP-MQT: tele/WemosMiniAkku_1/SENSOR = {"Time":"2019-02-18T05:23:41","ANALOG":{"A0":1024},"DS18B20":{"Temperature":5.0},"SR04":{"Distanz":52.772},"TempUnit":"C"} (beibehalten)
>Feb 18 05:23:40 WemosMiniAkku-1 ESP-MQT: tele/WemosMiniAkku_1/LWT = Offline
>Feb 18 05:23:40 WemosMiniAkku-1 ESP-MQT: tele/WemosMiniAkku_1/UPTIME_S = {"Time":"2019-02-18T05:23:41", "Uptime_s":15}

Mosquitto doesn't receive Topic:
>Feb 18 06:20:49 WemosMiniAkku-1 ESP-HTP: Web-Server aktiv bei WemosMiniAkku-1 mit IP-Adresse 192.168.1.137
>Feb 18 06:20:49 WemosMiniAkku-1 ESP-APP: (UTC) Mon Feb 18 05:20:48 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
>Feb 18 06:20:50 WemosMiniAkku-1 ESP-MQT: Verbindungsversuch...
>Feb 18 06:20:50 WemosMiniAkku-1 ESP-MQT: verbunden
>Feb 18 06:20:50 WemosMiniAkku-1 ESP-MQT: tele/WemosMiniAkku_1/LWT = Online (beibehalten)
>Feb 18 06:20:50 WemosMiniAkku-1 ESP-MQT: abonniere cmnd/WemosMiniAkku_1/#
>Feb 18 06:20:50 WemosMiniAkku-1 ESP-MQT: abonniere cmnd/sonoffAkkuGruppe/#
>Feb 18 06:20:50 WemosMiniAkku-1 ESP-MQT: abonniere cmnd/WemosMiniAkku-1_fb/#
>Feb 18 06:20:50 WemosMiniAkku-1 ESP-MQT: tele/WemosMiniAkku_1/INFO1 = {"Module":"Generic","Version":"6.4.1.9 stb-1.1(sonoff)","FallbackTopic":"cmnd/WemosMiniAkku-1_fb/","GroupTopic":"sonoffAkkuGruppe"}
>Feb 18 06:20:50 WemosMiniAkku-1 ESP-MQT: tele/WemosMiniAkku_1/INFO2 = {"WebServerMode":"Admin","Hostname":"WemosMiniAkku-1","IPAddress":"192.168.1.137"}
>Feb 18 06:20:50 WemosMiniAkku-1 ESP-MQT: tele/WemosMiniAkku_1/INFO3 = {"RestartReason":"Deep-Sleep Wake"}
>Feb 18 06:20:50 WemosMiniAkku-1 ESP-MQT: stat/WemosMiniAkku_1/RESULT = {"POWER":"ON"}
>Feb 18 06:20:51 WemosMiniAkku-1 ESP-SRC: MQTT
>Feb 18 06:20:51 WemosMiniAkku-1 ESP-RSL: empfangenes topic cmnd/WemosMiniAkku_1/DeepSleep, Datengröße 4, Daten 1800
>Feb 18 06:20:51 WemosMiniAkku-1 ESP-RSL: Gruppe 0, Index 1, Befehl DEEPSLEEP, Daten 1800
>Feb 18 06:20:51 WemosMiniAkku-1 ESP-MQT: stat/WemosMiniAkku_1/RESULT = {"DeepSleep":"1800 (1800)"}
>Feb 18 06:20:59 WemosMiniAkku-1 ESP-MQT: tele/WemosMiniAkku_1/STATE = {"Time":"2019-02-18T06:20:58","Uptime":"0T00:00:21","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"SSID","BSSId":"34:31:C4:76:5C:90","Channel":11,"RSSI":44},"DeepSleep":1800,"Heap":23048}
>Feb 18 06:20:59 WemosMiniAkku-1 ESP-MQT: tele/WemosMiniAkku_1/SENSOR = {"Time":"2019-02-18T06:20:58","ANALOG":{"A0":1024},"DS18B20":{"Temperature":5.1},"SR04":{"Distanz":52.281},"TempUnit":"C"} (beibehalten)
>Feb 18 06:20:59 WemosMiniAkku-1 ESP-MQT: tele/WemosMiniAkku_1/LWT = Offline
>Feb 18 06:20:59 WemosMiniAkku-1 ESP-MQT: tele/WemosMiniAkku_1/UPTIME_S = {"Time":"2019-02-18T06:20:58", "Uptime_s":16}

Here's the according Log of Broker:
[image]

@stefanbode : What I assume:
Could it be, that the Tasmota devices trys to send the topic to broker (and it also reports it via syslog), but goes to sleep to quickly? Maybe it would be helpful to wait for 1-2 seconds after teleperiod expires before to go to sleep?

Thanks a lot 

helmar

stefanbode, 2019-06-26:
Hi, very unlikely because this is an open port and the communication is synchronous. Also 3 messages are send. The uptime_s, lwt and the Sensor. And the sensor is the first one. The error in mosquito is correct. The disconnect does not work anyway. Therefore normal behavior on closing the socket.

stefanbode, 2019-08-30:
Ok, many additions and changes during the last 48h. My device was now reporting 500 wakeups without one missing message. Kernel 2.5.2, mosquitto broker. I fixed the shutdown process. Many new options now in, to execute commands before deepsleep. e.g. to shutdown something.

closed 2019-08-30 by stefanbode