stefanbode/Sonoff-Tasmota #100: deepsleep problems

bsculley, 2018-12-14:
I am trying to implement the deepsleep feature on ESP-12E and having a few problems.  I have included the results of STATUS 0 below.  I have set the teleperiod to 10 and deepsleep to 60.  GPIO16 and RST are jumpered.  Power from 2 AA cells.

The MQTT log shows the module rebooting every 8-10 seconds.  I don't think the module is sleeping at all, but then why the reboot?  The LED is always on when the board is powered (sleep mode or not).  How do you get the LED to turn off?  

The reboot looks like this:

```
00:00:00 Project sonoff Moisture Sensor 0 (Topic msens0, Fallback DVES_0DC860, GroupTopic sonoffs) Version 6.3.0.16 stb-1.4(sonoff)-2_4_2
00:00:00 WIF: Connecting to AP1 Ceriolo in mode 11N as msens0-2144...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on msens0-2144.local with IP address 192.168.1.101
09:58:59 MQT: Attempting connection...
09:58:59 MQT: Connected
09:58:59 MQT: tele/msens0/LWT = Online (retained)
09:58:59 MQT: cmnd/msens0/POWER = 
09:58:59 MQT: tele/msens0/INFO1 = {"Module":"Generic","Version":"6.3.0.16 stb-1.4(sonoff)","FallbackTopic":"DVES_0DC860","GroupTopic":"sonoffs"}
09:58:59 MQT: tele/msens0/INFO2 = {"WebServerMode":"Admin","Hostname":"msens0-2144","IPAddress":"192.168.1.101"}
09:58:59 MQT: tele/msens0/INFO3 = {"RestartReason":"Power on"}

```
```
09:48:55 CMD: status 0
09:48:55 MQT: stat/msens0/STATUS = {"Status":{"Module":18,"FriendlyName":["Moisture Sensor 0"],"Topic":"msens0","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:48:55 MQT: stat/msens0/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:00:05","StartupUTC":"2018-12-14T08:48:50","Sleep":50,"BootCount":53,"SaveCount":73,"SaveAddress":"3FB000"}}
09:48:55 MQT: stat/msens0/STATUS2 = {"StatusFWR":{"Version":"6.3.0.16 stb-1.4(sonoff)","BuildDateTime":"2018-12-13T21:53:00","Boot":5,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
09:48:55 MQT: stat/msens0/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Ceriolo","Ceriolo"],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
09:48:55 MQT: stat/msens0/STATUS4 = {"StatusMEM":{"ProgramSize":469,"Free":2600,"Heap":22296,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":1458400,"Features":["00000002","00000809","0F802794","000000A0","00000000"]}}
09:48:55 MQT: stat/msens0/STATUS5 = {"StatusNET":{"Hostname":"msens0-2144","IPAddress":"192.168.1.101","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"A0:20:A6:0D:C8:60","Webserver":2,"WifiConfig":5}}
09:48:55 MQT: stat/msens0/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.126","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0DC860","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:48:55 MQT: stat/msens0/STATUS7 = {"StatusTIM":{"UTC":"Fri Dec 14 08:48:55 2018","Local":"Fri Dec 14 09:48:55 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:36","Sunset":"16:53"}}
09:48:56 MQT: stat/msens0/STATUS10 = {"StatusSNS":{"Time":"2018-12-14T09:48:55","CHIRP":{"Voltage":   40,"Moisture":    0,"Temperature":   22},"TempUnit":"C"}}
09:48:56 MQT: stat/msens0/STATUS11 = {"StatusSTS":{"Time":"2018-12-14T09:48:56","Uptime":"0T00:00:06","SleepMode":"Dynamic","Sleep":50,"LoadAvg":7,"Wifi":{"AP":1,"SSId":"Ceriolo","BSSId":"E4:8D:8C:C9:D6:20","Channel":1,"RSSI":100},"DeepSleep":60,"Heap":22928}}

```

stefanbode, 2018-12-19:
The status 1 give the restart reason. Should be wake up or something similar. Please see info1 to 3 to get an idea what is happening. Maybe also a log file will help after automatic reboot to get an idea.

stefanbode, 2018-12-19:
I see you powered by 2AA cells that have 2.4V-3.0V depending on the type. It could be that there is an undervoltage during start causing a reset. Please try to go with USB power first and check the configuration

bsculley, 2018-12-19:
Thanks for the reply.  I have successfully put the module to sleep with a simpler piece of software using both USB and battery power, so I don't think power is the issue.  It seems as though there is some configuration of the chip or conflicting process in the Tasmota variant that is keeping it from sleeping.   What ESP core version are you using?

stefanbode, 2018-12-19:
Absolutely no issues with 2.3 and 2.4. There is a binary attached in the /bin file for sonoff. Maybe you can test this one. Deepsleep will work on all Arduino cores. important is that the device do NOT loose power. it must be 100% powered also during deepsleep. Please send a log of two cycles with the problem

bsculley, 2018-12-19:
In the log below you can see the module running (on USB power) with a teleperiod of 10.  When the deepsleep value is set to 60, the module logs entry to sleep (I added the "Sleeping for..." message) but then resets after only a few seconds with reason "Hardware Watchdog".  This behavior is consistent whether or not GPIO16 is connected to RST, so I don't think it has to do with sleep.

```
13:58:16 MQT: tele/msens0/SENSOR = {"Time":"2018-12-19T13:58:15","CHIRP":{"Voltage":  100,"Moisture":    2,"Temperature":   23},"TempUnit":"C"}
13:58:26 MQT: tele/msens0/STATE = {"Time":"2018-12-19T13:58:26","Uptime":"0T00:01:06","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Ceriolo","BSSId":"E4:8D:8C:C9:D6:20","Channel":1,"RSSI":100},"DeepSleep":0,"Heap":23224}
13:58:26 MQT: tele/msens0/SENSOR = {"Time":"2018-12-19T13:58:26","CHIRP":{"Voltage":  100,"Moisture":    2,"Temperature":   23},"TempUnit":"C"}
13:58:30 CMD: deepsleep 60
13:58:30 MQT: stat/msens0/RESULT = {"DeepSleep":"60 (60)"}
13:58:36 MQT: tele/msens0/STATE = {"Time":"2018-12-19T13:58:36","Uptime":"0T00:01:16","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Ceriolo","BSSId":"E4:8D:8C:C9:D6:20","Channel":1,"RSSI":98},"DeepSleep":60,"Heap":23256}
13:58:37 MQT: tele/msens0/SENSOR = {"Time":"2018-12-19T13:58:36","CHIRP":{"Voltage":  100,"Moisture":    2,"Temperature":   23},"TempUnit":"C"}
13:58:37 MQT: tele/msens0/LWT = Offline
13:58:37 MQT: tele/msens0/UPTIME_S = {"Time":"2018-12-19T13:58:37", "Uptime_s":67}
13:58:37 IFO: Sleeping for 60⸮
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
00:00:00 Project sonoff Sonoff (Topic msens0, Fallback DVES_0DC860, GroupTopic sonoffs) Version 6.3.0.16 stb-1.4(sonoff)-2_4_2
00:00:00 Max sleep 13194 13185843193
00:00:00 WIF: Connecting to AP1 Ceriolo in mode 11N as msens0-2144...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on msens0-2144.local with IP address 192.168.1.101
13:58:53 MQT: Attempting connection...
13:58:53 MQT: Connected
13:58:53 MQT: tele/msens0/LWT = Online (retained)
13:58:53 MQT: cmnd/msens0/POWER = 
13:58:53 MQT: tele/msens0/INFO1 = {"Module":"Generic","Version":"6.3.0.16 stb-1.4(sonoff)","FallbackTopic":"DVES_0DC860","GroupTopic":"sonoffs"}
13:58:53 MQT: tele/msens0/INFO2 = {"WebServerMode":"Admin","Hostname":"msens0-2144","IPAddress":"192.168.1.101"}
13:58:53 MQT: tele/msens0/INFO3 = {"RestartReason":"Hardware Watchdog"}
13:59:01 MQT: tele/msens0/STATE = {"Time":"2018-12-19T13:59:01","Uptime":"0T00:00:14","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Ceriolo","BSSId":"E4:8D:8C:C9:D6:20","Channel":1,"RSSI":100},"DeepSleep":60,"Heap":23520}
13:59:02 MQT: tele/msens0/SENSOR = {"Time":"2018-12-19T13:59:01","CHIRP":{"Voltage":  100,"Moisture":    0,"Temperature":   23},"TempUnit":"C"}
13:59:02 MQT: tele/msens0/LWT = Offline
13:59:02 MQT: tele/msens0/UPTIME_S = {"Time":"2018-12-19T13:59:02", "Uptime_s":13}
13:59:02 I⸮
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
00:00:00 Project sonoff Sonoff (Topic msens0, Fallback DVES_0DC860, GroupTopic sonoffs) Version 6.3.0.16 stb-1.4(sonoff)-2_4_2
00:00:00 Max sleep 13180 13174308857
00:00:00 WIF: Connecting to AP1 Ceriolo in mode 11N as msens0-2144...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on msens0-2144.local with IP address 192.168.1.101
13:59:18 MQT: Attempting connection...
13:59:18 MQT: Connected
13:59:18 MQT: tele/msens0/LWT = Online (retained)
13:59:18 MQT: cmnd/msens0/POWER = 
13:59:18 MQT: tele/msens0/INFO1 = {"Module":"Generic","Version":"6.3.0.16 stb-1.4(sonoff)","FallbackTopic":"DVES_0DC860","GroupTopic":"sonoffs"}
13:59:18 MQT: tele/msens0/INFO2 = {"WebServerMode":"Admin","Hostname":"msens0-2144","IPAddress":"192.168.1.101"}
13:59:18 MQT: tele/msens0/INFO3 = {"RestartReason":"Hardware Watchdog"}
13:59:26 MQT: tele/msens0/STATE = {"Time":"2018-12-19T13:59:26","Uptime":"0T00:00:14","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Ceriolo","BSSId":"E4:8D:8C:C9:D6:20","Channel":1,"RSSI":100},"DeepSleep":60,"Heap":23520}
13:59:26 MQT: tele/msens0/SENSOR = {"Time":"2018-12-19T13:59:26","CHIRP":{"Voltage":  100,"Moisture":    0,"Temperature":   23},"TempUnit":"C"}
13:59:26 MQT: tele/msens0/LWT = Offline
13:59:26 MQT: tele/msens0/UPTIME_S = {"Time":"2018-12-19T13:59:26", "Uptime_s":13}
13:59:27 I⸮
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
[... 26 lines cut ...]
wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
00:00:00 Project sonoff Sonoff (Topic msens0, Fallback DVES_0DC860, GroupTopic sonoffs) Version 6.3.0.16 stb-1.4(sonoff)-2_4_2
00:00:00 Max sleep 13171 13171687417
00:00:00 WIF: Connecting to AP1 Ceriolo in mode 11N as msens0-2144...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on msens0-2144.local with IP address 192.168.1.101
14:00:07 MQT: Attempting connection...
14:00:07 MQT: Connected
14:00:07 MQT: tele/msens0/LWT = Online (retained)
14:00:07 MQT: cmnd/msens0/POWER = 
14:00:07 MQT: tele/msens0/INFO1 = {"Module":"Generic","Version":"6.3.0.16 stb-1.4(sonoff)","FallbackTopic":"DVES_0DC860","GroupTopic":"sonoffs"}
[... cut ...]
14:00:15 MQT: tele/msens0/STATE = {"Time":"2018-12-19T14:00:15","Uptime":"0T00:00:14","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Ceriolo","BSSId":"E4:8D:8C:C9:D6:20","Channel":1,"RSSI":100},"DeepSleep":60,"Heap":23336}
14:00:15 MQT: tele/msens0/SENSOR = {"Time":"2018-12-19T14:00:15","CHIRP":{"Voltage":  100,"Moisture":    0,"Temperature":   23},"TempUnit":"C"}
14:00:15 MQT: tele/msens0/LWT = Offline
14:00:15 MQT: tele/msens0/UPTIME_S = {"Time":"2018-12-19T14:00:15", "Uptime_s":12}
14:00:16 I⸮```

stefanbode, 2018-12-19:
Thanks for sharing the log. This is always very helpful to find the issue as fast as possible. The reboot is executed by: RestartReason":"Hardware Watchdog". This is not as it should be. I cannot reproduce it on my side currently. after the last MQT message, there should be NO other line. The additional line with the timestamp causes the problem or is part of the problem. Let me check the code, if I can find something meaningful.

stefanbode, 2018-12-19:
Have you set the DEEP SLEEP GPIO pin? In the configuration you can set this, but he purpose of his switch is to temorary switch off the deepsleep mode with a switch. this is explained in the wiki. If you have defined this one please undefine and check if it work. If yes, then ready the wiki carefully how to wire the GPIO16 (D0) for the switching aspect.

bsculley, 2018-12-20:
I have removed any reference to the DEEP SLEEP pin.  Here is the module configuration:
[image]

The only difference this seems to have made is that the restart reason is now "RestartReason":"External System".  Everything else is the same as before.  I am trying to implement a module similar to the CHIRP sensor: [https://www.tindie.com/products/aprbrother/esp-soil-moisture-sensor-rev-21/]

The changes are confined to xsns_92_chirp_mois.ino and the sensor works fine, so I don't think that is contributing to the problem.

bsculley, 2018-12-21:
Further to this issue, I started over on my project and everything is working now.  I must have screwed something up before.  Wish I knew what, but …

Anyway thanks for you support.  FYI in the 2.4.2 ESP core the sleep time has been extended to 64 bits and there is a function available (ESP.deepSleepMax()) to specify the maximum time (about 3.5 hours I think).

closed 2018-12-21 by bsculley