stefanbode/Sonoff-Tasmota #103: Sonoff DUAL R2 should work ?

rt400, 2018-12-19:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Development/Compiler/Upload tools used : Arduino
- [X] Hardware used : Sonoff Dual R2
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```
hi..
i try to use your Shutter in sonoff dual r2 but i dont understand how to config.. and in the wiki i cant found any sonoff dual config..
i glad to get a explain how to i need to config it. 

best regard ,
yuval
**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2018-12-19:
The sonoff dual works as any ESP. just select the sonoff dual from the list of devices. Then set the different setoptions as described in the wiki. Done!
If you want to integrate the buttons for manual changing the shutter it will get tricky. GPIO0 and GPIO09 are defined for switch 0 and 1. Please look for youtibe videos how to connect to these with switches. e.g. https://www.youtube.com/watch?v=n1fNsVS30CA minute 5:50 and later

rt400, 2018-12-19:
so i need to select "Sonoff Dual R2" not a Generic ?
and in your wiki the setoption is for Generic.. so what its ok for dual also ?

stefanbode, 2018-12-19:
the setoptions always overlay with the device selected. Therefore it does not matter which device you have selected. The device selection is only to make it convenient for people to assign the right GPIO to the right function. 
Short answer: Yes

rt400, 2018-12-20:
@stefanbode its looks work great.. thanks.
i try to make an automation..if you can explain me how in automation it work that will be great..

jokerigno, 2018-12-29:
Him

this part of the wiki is still valid?

> **Special configuration for SONOFF DUAL required**
> In sonoff_template.h, in section Sonoff Dual R2, replace the line
> 
> GPIO_KEY1, // GPIO10 Button on casing with
> 
> 0, // GPIO10 Button on the casing
> 
> then in configure module section you're able to select button 1 for GPIO0 (before was not available), and button 2 for GPIO9.

Because if I change it I receive a compiling error.

Thank you

jokerigno, 2018-12-29:
That's the error:

```
sonoff\sonoff_template.h:44:3: error: expected '}' before numeric constant
sonoff\sonoff_template.h:44:3: error: expected unqualified-id before numeric constant
sonoff\sonoff_template.h:146:19: error: expected declaration before '}' token
```

rt400, 2018-12-30:
can you share the code that you change ?

stefanbode, 2018-12-31:
I see an error in line 44. there is no change required in this section. I assume you accidentally changed something. Yes the change should still make the job and the wiki is valid

rt400, 2018-12-31:
@stefanbode i want to use SONOFF T1 3CH but i want that relay3 will close and relay2 will stop any action..
how can i do this ?

rt400, 2018-12-31:
@jokerigno you need to create it with cover in HA 
```
cover:
  ```

jokerigno, 2018-12-31:
Yes. It's a mqtt cover

rt400, 2018-12-31:
so now you need to add it to UI-LOVELACE.YAML and when you click on it in the HA WEB you will see the slider

jokerigno, 2018-12-31:
I've done it and I can't see any slider.

rt400, 2018-12-31:
did you see the entity from HA WEB ?

jokerigno, 2018-12-31:
Yes. Both on old front end and in Lovelace .

I can see buttons. Control shutter up and down. But can't see any slider.

rt400, 2018-12-31:
[image]
do you have like this ?

rt400, 2018-12-31:
if do juck click on the name and you will see the slider .

[image]

jokerigno, 2018-12-31:
Yes. I do have it, but without the slider.

rt400, 2018-12-31:
try to follow this config : 
```
- platform: mqtt
  name: "Shutters Blinds - Salon"
  availability_topic: "tele/SalonShutter/LWT"
  payload_available: "Online"
  payload_not_available: "Offline"
  position_topic: "stat/SalonShutter/SHUTTER1"
  position_open: 100
  position_closed: 0
  set_position_topic: "cmnd/SalonShutter/shutterposition"
  value_template: "{{ value | int }}"
  command_topic: "cmnd/SalonShutter/backlog"
  payload_open: "SHUTTEROPEN"
  payload_close: "SHUTTERCLOSE"
  payload_stop: "SHUTTERSTOP"
  retain: false
  optimistic: false
  qos: 0
```
with HA 0.82 version and up
change the Topic on yours Topic

rt400, 2019-01-01:
which HA ver you have ?

jokerigno, 2019-01-01:
Last available version. 0.84.6

rt400, 2019-01-01:
this is so strange..
what mqtt broker you have?

jokerigno, 2019-01-01:
Toke mosquito running on docker.

rt400, 2019-01-01:
can you give the status 0 from switch web console?

jokerigno, 2019-01-01:
Sorry, I don't understand

rt400, 2019-01-01:
in the switch web go to the CONSOLE write a command "status 0" and write here the result .

jokerigno, 2019-01-01:
17:05:41 CMD: Status 0
17:05:42 MQT: sonoff/stat/STATUS = {"Status":{"Module":39,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:05:42 MQT: sonoff/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"1T00:20:24","StartupUTC":"2018-12-31T15:45:18","Sleep":50,"BootCount":7,"SaveCount":51,"SaveAddress":"F9000"}}
17:05:42 MQT: sonoff/stat/STATUS2 = {"StatusFWR":{"Version":"6.3.0.16 stb-1.4(sonoff)","BuildDateTime":"2018-12-29T22:34:53","Boot":7,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
17:05:42 MQT: sonoff/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Casa Concordia","Casa Concordia Domotica"],"TelePeriod":300,"SetOption":["000AC009","558180C0","C0000200"]}}
17:05:42 MQT: sonoff/stat/STATUS4 = {"StatusMEM":{"ProgramSize":514,"Free":488,"Heap":16928,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":1327185,"Features":["00000003","00000809","0FDAE794","000383A0","23B617CE"]}}
17:05:42 MQT: sonoff/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff-6889","IPAddress":"192.168.2.151","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"60:01:94:B4:BA:E9","Webserver":2,"WifiConfig":4}}
17:05:42 MQT: sonoff/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.100","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B4BAE9","MqttUser":"jarvis","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:05:42 MQT: sonoff/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Jan 01 16:05:42 2019","Local":"Tue Jan 01 17:05:42 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:03"}}
17:05:42 MQT: sonoff/stat/STATUS10 = {"StatusSNS":{"Time":"2019-01-01T17:05:42","SHUTTER-1":100}}
17:05:42 MQT: sonoff/stat/STATUS11 = {"StatusSTS":{"Time":"2019-01-01T17:05:42","Uptime":"1T00:20:24","Vcc":3.241,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Casa Concordia","BSSId":"00:70:C2:7A:44:85","Channel":11,"RSSI":68},"DeepSleep":0,"Heap":16888}}

jokerigno, 2019-01-02:
That's why topic are different:

> To keep MQTT discovery messages short, "base topic" feature is used.
> This feature is not compatible with the default Tasmota topic prefix/topic/COMMAND, e.g. cmnd/sonoff/POWER
> The topic will automatically be changed to topic/prefix/COMMAND, e.g. sonoff/cmnd/POWER
> If you want to use a custom fulltopic, it MUST end with %prefix%/ or it will be overwritten with topic/prefix/COMMAND.
> This is an example of an allowed fulltopic: tasmota/%topic%/%prefix%/.

closed 2019-01-16 by stefanbode