stefanbode/Sonoff-Tasmota #105: user_config_override_sample.h und platformio.ini kleine Fehler

ihavenonick, 2018-12-30:
Servus Stefan,
2 kleine Fehler sind mir aufgefallen.
1. "user_config_override_sample.h" muss in "user_config_override.h" umbenannte werden
2. In der platformio.ini muss unter [platformio], wenn man nur die Sonoff Version in deutsch will das "-de" bei "sonoff-de" entfernt werden. Sonst passt der Pfad nicht.

Dann mein altes Problem, wobei das wohl nicht an Deiner Fork liegt, Dir vielleicht aber was dazu einfällt.
Also die aktuelle Position (SHUTTER-1) wird mir mittlerweile angezeigt, allerdings erst, wenn ich den Sonoff neu starte. Irgend wie geht das Topic nicht über den MQTT bei einer Änderung, beim Neustart dann allerdings schon.

Da mir Deine Fork sehr gut gefällt, habe ich dazu in meinem Home Forum ein WiKi Eintrag angelegt, wo auch diese Dinge angesprochen werden. 
Kannst ja mal rein schauen [http://forum.creationx.de/lexicon/index.php?entry/34-rollladensteuerung-mit-der-stefanbode-tasmota-fork/](url)

stefanbode, 2018-12-31:
Vielen Dank,für das Feedback. Zu 1. das ist so richtig wie es ist, weil nicht jeder eine override Datei hat und Ich meine eigene beim commit nicht mitsenden möchte. Deswegen heist die auch sample.
Zu 2: Danke wird korrigiert, aber nicht so schnell, da ich aktuell gaaaaanz weit weg bin. Ohne Laptop

ihavenonick, 2019-01-03:
Danke Stefan,

ja zu 1. kann man lassen, da das zwar ne Fehlermeldung bei mir beim ersten Versuch bringt, aber bei beim zweiten dann auch so läuft.
eine Frage oder Anregung noch. Das läuft ja auch auf dem Shelly 2 mit 2MB Flash ganz gut. Nur habe ich da Probleme mit Deiner platformio.ini. Wenn ich das "eagle.flash.2m.ld" so anpasse, wird zwar der Flash mit 2MB erkannt, allerdings nicht genutzt. Ich hatte da eh zufällig in meinem Home Forum gerade ein Thread am Laufen und das angesprochen. Ein User hatte dann einfach die platformio.ini aus der original Tasmota 6.4.1.4 dazu in Deine Fork kopiert und siehe da es läuft (ich brauch eh nur die Sonoff-de Version).

Ich habe dann gleich noch einen drauf gesetzt und den Core 2.5.0 aktiviert. Das läuft bis jetzt auch ohne Auffälligkeiten. Könntest Du das (zumindest die Flashgröße) bei Dir anpassen, damit das direkt in Deiner Fork funktioniert ohne eine "fremd" platformio.ini da zu verwenden?

Es eilt nicht und hat Zeit :-)

stefanbode, 2019-01-28:
Ist korrigiert. Ich bin bei der Platform.ini zurück auf den standard

closed 2019-01-28 by stefanbode