stefanbode/Sonoff-Tasmota #107: Release generic binaries

jaimemrjm, 2019-01-06:
Related with #101 but it's closed now. I think binary compilation for newbies can fail due to many reasons. For example, I have the same [issue](https://github.com/arendst/Sonoff-Tasmota/issues/4155) for this branch. I would like to install a binary from this branch in my Sonoff T1 EU 2Ch to control roller shutter but it seems the only binary at `bin` folder is for Pro 4Ch.

stefanbode, 2019-01-16:
Please take a look at the binaries folder. Not the newest one but ok. You need the Sonoff one for correct memory allocation

closed 2019-05-23 by stefanbode

meingraham, 2019-05-23:
We recently became aware of [Gitpod online IDE](https://github.com/arendst/Sonoff-Tasmota/wiki/Compiling-Tasmota-on-Gitpod). I believe this might be helpful for compiling your Tasmota fork as well.