stefanbode/Sonoff-Tasmota #108: Use of mure than 4 relays on D1 Mini

mrmin24, 2019-01-07:
Hi,

I have successfully used the PCF-8574 to implement 10 relays(including onboard and serial) with the SV. 

Is there however no way to use more than 4 relays on the D1 mini, seeing that there are so many pins available. Or am I doing something wrong? It only allows me to map 4.

Thx,

stefanbode, 2019-01-08:
If you successfully attached the pcf on the i2c Bus there should be a new web interface where you can configure the relays. Oootb there are 32 relays supported. With some minor type changes up to 64. Please give the wiki a try and follow the instructions.

mrmin24, 2019-01-08:
Hi, sorry if I may have been unclear. I actually meant using the D1 mini without the PCF.

But I now realise that that the PCF probably just adds over and above the 4 x standard relays. So I suppose this is a stupid question. 

Sorry.

closed 2019-01-08 by mrmin24