stefanbode/Sonoff-Tasmota #109: T1 3 channels

Pako75, 2019-01-12:
Hi,
Do you think it would be possible to use the interlock option only on 2 channels of a t1 with 3 ch???
I think that changing something on this code this would be possible, can you help me???

stefanbode, 2019-01-13:
This is already implemented. See the the option to mask the relay. T is described in the wiki

Pako75, 2019-03-01:
Finally I've found to flash your mod on T3s.... Great, it works!!!

I've installed on 3 Sonoff T3 US, the first is working flawlessly, the other two are giving me problems.
Sometimes I hear a beep (3 beeps for accuracy, beep (pause) beep beep) and one of the three keys is activated, opening / closing or turning on the light at random.

What can I do?

Thank you

stefanbode, 2019-03-01:
I would send a reset 2 to ensure a reset to factory default and overwrite the rest with 0. Then configure again.

Pako75, 2019-03-01:
Please, can you explain what to do?? Reset 2? Reset 0?

Pako75, 2019-03-01:
Ok, i’ve Read about the reset 2, i’ve already tried to erase the whole flash and write agami the firmware but i had no changes

closed 2019-05-21 by stefanbode