stefanbode/Sonoff-Tasmota #110: Relay pulse at top or bottom

lsd87, 2019-01-13:
Setting up a  Sonoff T1 EU 2 CH and noticed an odd behavior - 
When the position is at say 100%, pressing button 1 or toggle 1 via web screen, the relay 1 pulses on/off for about a second. The web screen also changes state to on then off
Same happens when at 0% and button/toggle 2 is selected (Assume relay 2 pulses)
It looks like the relay is tuned on, then the firmware checks to see if its at say 100% already and then turns the relay off.
Maybe the firmware could check to see if the shutter is at the limits already before issuing a output  command

Noticed this when playing with single/multi/long button presses (Option1 and 11 and ButtonTopic) 

Anyway thanks for creating this fork - it should be implemented/you credited in the standard tasmota release.
Leigh

stefanbode, 2019-01-16:
Hi, yes this is a known limitation. Currently my algorithm is after tasmota already switches on the relay. The only thing I can do is to switch it off immediately. Most shutter have endstops. Therefore the issue is not really a problem.

lsd87, 2019-01-16:
Hi Stefan, I have some shutters that use micro switches (Great and work fine) and some that have electronic end stops (Problems) - the electronic ones are supposed to be programmed with top and bottom end points but after  a while they loose them and use motor over current to act as the end stop. I don't think it would be a problem.
Anyway thanks for creating this version really appreciated :-)

closed 2019-01-16 by lsd87