stefanbode/Sonoff-Tasmota #112: SHUTTERSETCLOSE needs some explanation

jaimemrjm, 2019-01-14:
The command SHUTTERSETCLOSE description [in the wiki ](https://github.com/stefanbode/Sonoff-Tasmota/wiki/Shutter-blinds-working-with-two-relays#calibrating) is very brief and needs some further explanation. Some example to apply it and check the calibration is needed.

stefanbode, 2019-01-14:
Thanks for pointing this out. The wiki is open. If you like you can improve

jaimemrjm, 2019-01-14:
I would like to update the wiki but I need to understand the command firstly :smiley:

stefanbode, 2019-01-14:
Ok shutter50percent is to declare the 50 percent opening position. With this calibration the percent matches quite ok. During testing it might be that your real position and the one in the firmware are out of sync regarding the position . In this case just close the blinds and execute the command. Now everything is in sync again. The command is never required after setup. Normally....

jaimemrjm, 2019-01-14:
So, if I have understood from #113, I can set the actual close position by calling SHUTTERSETCLOSE when the shutter is actually close. shutter50percent command doesn't work for me currently because I cannot move the shutter to the actual 50% position (I have reach the 100 SHUTTERPOSITION but the actual shutter position is more or less at 30%).

jaimemrjm, 2019-01-15:
Wiki updated. Thanks all!

closed 2019-01-15 by jaimemrjm

WolfgangFahl, 2024-08-24:
https://tasmota.github.io/docs/Blinds-and-Shutters/#calibration

has 
```
ShutterSetClose<x>. 
```
but no example for x

stefanbode, 2024-08-24:
Agree. And I hope you guess it right. It is the shutter number. Only relevant if you have more than one shutter defined.

WolfgangFahl, 2024-08-25:
I've been struggling to properly set up my shutters ever since I switched from the Nous L12T modules ([link](https://nous.technology/product/l12t.html)) to the Nous B3T modules ([link](https://nous.technology/product/b3t.html)).

I'm particularly stuck on linking the software settings, which feels like trying to decipher ancient Greek. The whole process has been trial and error, especially since my WiFi connections have been unreliable—a problem I had hoped to solve by moving to the B3T modules. Unfortunately, I've already bricked and discarded several L12T devices, and I'm finding it extremely frustrating to have to tinker with the settings without understanding what I'm doing.

The documentation for configuring blinds and shutters in Tasmota ([link](https://tasmota.github.io/docs/Blinds-and-Shutters/)) is confusing. It introduces new terms like "Interlock 1,2" without explaining what these numbers mean. Are these numbers referring to something on the physical devices? Which one corresponds to "up" and which one to "down"?

In my opinion, it would make sense to have a setup where `setUp=1`, `setDown=2`, and then an interlock for "Up" and "Down" or something similar. Since I can't change the interface, a more detailed explanation or tutorial would be extremely helpful.

For instance, the documentation for the B3T module ([link](https://nous.technology/product/pdf?slug=b3t&alias=eng)) mentions:
```
To work in "Blinds" mode, you need to activate the following commands:
• SetOption80 1
• ShutterRelay1 1
• Interlock 1,2
• Interlock ON
• SwitchMode1 4
• SwitchMode2 4
```

However, there's no accompanying explanation, making it as cryptic as ancient Greek or Mesopotamian. Any guidance or a step-by-step tutorial would be greatly appreciated.

[image]

[image]

stefanbode, 2024-08-25:
Hi. The reason I do not explain what interlock is doing is because you find this in the command section. Duplicate information here makes it more difficult to change. But I can insert links that make it easier to find. Agree. For any command you should search and take a look in the command section. Here you will get the most detailed information about what the command is doing, what the defaults are and so on. 

With the relay you referenced you should be fine exactly follow the steps. Even if you do not know what you do. The up and down problem I do not understand. You see the button in the web ui and see what relay is triggered. Should be straightforward.

stefanbode, 2024-08-27:
Hallo Wolfgang,

 

ich probiere es mal auf Deutsch. Vielleicht macht es das etwas klarer. 

Alle Kommandos mit der Beschreibung findest du hier:
https://tasmota.github.io/docs/Commands/#shutters

 

Was der Interlock macht steht gleioch auf der ersten Seite der Hilfe im Bereich "Shutter mode 1". Er macht am Ende nichts anderes als sicher zu stellen, dass Relay1 und Relay2 niemals unter welchen Umständen auch immer gleichzeitig an sind.

 

Welches Relay was macht kommt ca 10cm weiter unten in der Tabelle. Das hängt halt davon ab, ob am den Rollanden "invertiert" Liegt daran dass verschiedene Automatisierungslösungen mal die 100 erwarten wenn komplett auf und mal 100 wenn komplett zu. Das kann man dann mit dem "shutterinvert" anpassen.

 

Der Default ist: Relay1 = AUF und Relay2 = ZU

 

Der Switchmode (auch das findet man in der Doku) definiert, wie ein Schalter (Switch) funktioniert.

https://tasmota.github.io/docs/Buttons-and-Switches/#switchmode

 

Die von Nous haben das doch schon ganz gut gemacht mit der Liste der Kommandos. Dur auf dem einen Bild der Configuration fehlen rechts die NUMMERN. Das wäre echt noch wichtig gewesen

 

 

Gesendet: Sonntag, 25. August 2024 um 15:58 Uhr

 

grafik.png (view on web)

grafik.png (view on web)