stefanbode/Sonoff-Tasmota #113: How to manual control of the shutter

jaimemrjm, 2019-01-14:
Wiki describes:
> You can define SWITCH1 and SWITCH2 to manually control the shutter. Please be aware that it might require to set SWITCHMODE1 4 and SWITCHMODE2 4 (inverse follow) to get a proper behavior.

But how you can define that variables? Is it possible to manage from the console or are compiled variables? I have tried to move up/down my shutter with my T1 2C but I don't known how to to open more than SHUTTERPOSITION 100 or close more than SHUTTERPOSITION 0. Is there any way to set the SHUTTERPOSITION manually **without** move the shutter?

stefanbode, 2019-01-14:
Correct you set them in console or send as an MQTT command. Both works. They are persistent like all other variables. This is standard function from Tasmota and described in the master wiki

stefanbode, 2019-01-14:
You cannot set the shutterposition manually, but you can close the shutter and then use the init command to tell the firmware the shutter is closed. This command is shuttersetclose. See wiki

jaimemrjm, 2019-01-14:
OK, so I understand that by setting SWITCHMODE1 4 and SWITCHMODE2 4, the buttons for SWITCH1 and SWITCH2 will appear. And I shall move the shutter manually without firmware limitations. Would be required to enable SetOption14 (inter-locking) to avoid motor issues?

closed 2019-01-15 by jaimemrjm