stefanbode/Sonoff-Tasmota #114: Can I install stefanbode fork via OTA?

micdg96, 2019-01-15:
Hi, sorry for the stupid question but i'm new in the world of tasmota. In the Wiki I didn't find any instruction on how to install this firmware (I want to install the stefanbode fork in order to controll shutters) on a sonoff dual... Can you give me some explaination on how to do It?

jaimemrjm, 2019-01-15:
I tried to install this fork by installing Sonota firstly in my Sonoff Touch T1, compiling this fork from Arduino IDE and then WiFi upgrade from Sonota firmware but I had no luck to boot after upgrading. Maybe it works for you.

salcabo, 2019-01-15:
I'm interested too, can't find the way of uploading into sonoff t1 via ota.

jaimemrjm, 2019-01-16:
> I'm interested too, can't find the way of uploading into sonoff t1 via ota.

[Sonota allows to install Tasmonta into sonoff T1](https://github.com/mirko/SonOTA/wiki#known-working-configurations)

stefanbode, 2019-01-16:
OTA is working as in the master branch. Therefore you should not see any differences there

bjeram, 2019-01-16:
OTA works for me as well. What kind of problem do you have? 
It might e that you want to install Tasmota for the 1st time. 1st time I installed Tasmota using the serial connection because I read that there are problems/is not possible to install Tasmota for the 1st time using OTA (anymore).

lsd87, 2019-01-16:
I install tasmota  sonoff-mini (Full sonoff is over 500K in size) via serial to overwrite the original then use OTA to load stefans great version

micdg96, 2019-01-16:
So I have to buy a sonoff dual and then via OTA I can directly install this firmware downloading it from stefanbode GitHub?

lsd87, 2019-01-16:
If you buy an ITEAD sonoff it will come with EWelink firmware which will not allow OTA of tasmota.
So you have to the serial link upload first (See Tasmota GitHub/youtube for loads of info on uploading)

There is sonOTA which used to work directly with EWelink but last time I look it had been blocked.

BTW. I started off using a dual but I'm going to use the T1's (EU Dual) as it's will be a direct replacement for the current 230V switches in use now (just need to add neutrals to each switch). This gives manual/Auto control and if the home automation crashes (I'm using Hass.io) then the touch buttons can still be used to control the shutter.

bjeram, 2019-01-16:
I am using Dual R2 with the addition of RF (buttons), so I can also control shutter with RF (up/down).
There is RF support build into tasmota.

micdg96, 2019-01-16:
Ok i checked on mirko's wiki, for a sonoff dual r2 sonota should work with the firmware 2.0.1. For other firmware I have  to flash by serial. Now I didn't understand if stefanbode fork Is already included in the tasmota firmware that sonota installs via OTA or I have to change to firmware file.

jaimemrjm, 2019-01-16:
No, it doesn't include in Sonota. You should install stefanbode fork binary *after* sonota installation.

bjeram, 2019-01-16:
You have Firmware button where you can browse to the binary file

micdg96, 2019-01-16:
Ok thank you for the help!!

micdg96, 2019-01-16:
I would like to use a itead sonoff dual (putting only the pcb board into the wall) and connect my phisical switches to the pins button 0 - 1 and ground. Should It work? And also, moving manually the blinds should I see the percentage of opening changing on tasmota page (or on homeassistent if I integrate tasmota on it)?

lsd87, 2019-01-16:
The binary I got last week from Stefan's zip file does not work, see his wiki shutter section
Go to the bottom of the page to sonoff_template reference. You have to compile the binary yourself.
Both buttons will then work.
HA will see the position if you manually change the shutter using the buttons - use the sketch that Stephan has created

lsd87, 2019-01-16:
position_topic: stat/sonoff/SHUTTER1 
The is the line that HA uses to keep track of shutter position

micdg96, 2019-01-16:
Sorry but how to compile by my self the binary file?

bjeram, 2019-01-17:
Most information can be found on original Tasmota wiki: https://github.com/arendst/Sonoff-Tasmota/wiki
Here is for example how you set up Arduino IDE where you can build binaries yourselF: https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

micdg96, 2019-01-17:
Sorry but i'm finding difficulties... Does the file firmware.sonoff.bin in the bin direcory works for shutters if I upload it through the tasmota user interface? Or do I have to compile following the guide for setting up Arduino IDE downloading the zip file in the code section of stefanbode repo??

bjeram, 2019-01-17:
In a previous post, you asked about how to compile the binary file, so I've answered this.
But there is precompiled binary:
https://github.com/stefanbode/Sonoff-Tasmota/blob/master/bin/firmware_sonoff.bin
that can be loaded using OTA/ Firmware button in the web interface.

lsd87, 2019-01-18:
I found that Stefans precompiled one of the external contacts (not sure which one Button 0 or 1) did not work on the DUAL R2 that I had - as per Stefans notes. 
micdg96 If you want the version I did give me an email address and I'll send it to you (347K zipped)

micdg96, 2019-01-18:
Ok thank you. There is some way to send a private message?

lsd87, 2019-01-18:
Think I've loaded the binary into GitHub (never used github before !)
[https://github.com/lsd87/Sonoff-DUAL-Binary](url)

micdg96, 2019-01-18:
Lsd87 thank you really! So with this firmware i Will Be able to use and itead sonoff dual r2 with both phisical switches working (i have to solder the switches on the button pins on the board) if I config like you wrote in the readme?

lsd87, 2019-01-18:
Its the one I'm using on by dual :-)

closed 2019-01-25 by stefanbode