stefanbode/Sonoff-Tasmota #119: New Interlock TASMOTA Feature

meingraham, 2019-01-26:
@stefanbode - First, thank you for your `Interlock` feature TASMOTA PR.

**Have you looked for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
A relay cannot be part of two different interlocked relay groups.

The new `Interlock` feature ported from this fork over to TASMOTA main allows great flexibility in defining the groups of relays to be interlocked via the software configuration. One can define multiple groups and can define which relays are part of each group. However, a relay cannot be part of two different interlock groups.

**Describe the solution you'd like**
In my use case, I need to be able to interlock a relay individually with two other relays as separate interlocking groups. I have a ceiling fan speed setting scenario where:

- Low - 1 is on by itself
- Medium - 1&2 are on together
- High - 3 is on by itself

In this scenario, relay 1 can act independently from relay 2. However, relay 3 should never be latched when EITHER relay 1 or relay 2 are latched. Using the current `Interlock` syntax, the groups would be defined as:

`Interlock 1,3 2,3`

In other words, define the interlock groups as follows:

A. Relay 1 & 3
B. Relay 2 & 3
C. Do not interlock Relay 1 & 2 

`Interlock 1,2,3` won't allow 1&2 to be on at the same time, which the circuitry in the fan requires for selecting Medium speed.

**Describe alternatives you've considered**
Use Interlock for one of the relay pairs and use Rules for the other.

**Additional context**
https://github.com/arendst/Sonoff-Tasmota/pull/5014#issuecomment-457644164
https://github.com/arendst/Sonoff-Tasmota/pull/5014#issuecomment-457684473

Thank you for your consideration of this enhancement.

Mike

stefanbode, 2019-01-28:
Functionality will be replaced with the next sync.

closed 2019-01-28 by stefanbode

meingraham, 2019-01-28:
Stefan - replaced with your mods that Theo merged into TASMOTA or completely overhauled by you? If the latter, will the new functionality allow the type of interlocking I described above in detail?

Thanks.

Mike

stefanbode, 2019-01-28:
I have to take a look into the new implementation. It is definitely better than mine. From the concept your use case should work. Maybe some tweak is required