stefanbode/Sonoff-Tasmota #120: Servo Control

ezraholm50, 2019-01-27:
Hey, thanks for this fork stefan!

I was wondering if anyone managed to get a Continuous Servo setup with this fork?
The servo has GND 3.3/5V DATA, Pins

I hooked the GND and 5v up to a separate PSU. Now setting the board to Generic (nodemcu/wemos D1 mini)

How would one continue from here?

I'm using a custom lua script with mqtt/servo.h/wifi right now with: servo.write: 180; delay or servo.write: 0; delay

to rotate it one way or the other for a various amount of time.

stefanbode, 2019-01-28:
Imhave Never worked with servos. Isn’t there already something intake original firmware? How is the servo position or direction controlled? ILM totally new to this topic

ezraholm50, 2019-01-28:
Not that i know of... just a simple implement of servo.h lib would do. Just need to specify the data pin for the servo (1 in this case) and the delay

with: servo.write(180) (one way) 
servo.write(0) (other way)
then a delay to figure out how many rotations it needs

I just dont know how to get these settings to show up in the webgui, i know a lot of ppl are searching for this option!

ezraholm50, 2019-02-02:
it works out of the box with the original Tasmota, just compiled ith with pwm_min freq 50. 
setoption15 1
pwmfrequency 50

set the module to generic, select the pin you have the servo on and set it at PWM1

now with dimmer 1 / dimmer 0 / dimmer 10 i can let it go one way / stop / go the other way

I use backlog with mqtt commands to automate this to control my horizontal curtains 50/50 to close.

Thanks

temocastro, 2020-03-08:
@ezraholm50 is it too much to ask for your help here? Can I use Tasmota to control the servo  without additional changes? how?

I have the PWM pin in the ESP8266, tasmota shows 2 switches, I can move the servo small jumps only, maybe 5-10 degrees each time

ezraholm50, 2020-03-08:
You can, my previous posts describe it all. Cant recall how to exactly anymore.Setoption 15 onDimmer 1 is one way and dimmer 10+ the other waySet gpio as pmw1

stefanbode, 2020-03-08:
Can you post a link to a servo motor and maybe a description. For me it looks like the Pwm frequency is giving the position if the stuff with the dimmer works. Sounds strange to me

stefanbode, 2020-03-08:
As far as I can see right now the motor is a normal one with reverse polarity or direction pin and then powered for a period of time. Additionally to a normal shutter there is a rotary encoder delivering for example 360 ticks per round. So the only enhancement against the normal shutter should be to calculate position not by time but with the same method like the stepper. Correct?

ezraholm50, 2020-03-08:
I only use continous servos to keep turning one way or the other. So i have no idea

stefanbode, 2020-03-08:
I have ordered a continuous servo from amazon. As far as I understand pwm 0 turn left and Pwm 90 is stop and pwm 180 is Right. Additionally there seems to be a pin sending a counter signal. Will have time on Tuesday evening to get more insight

stefanbode, 2020-03-08:
Ahh I see there is no counter pin. Plus,minus and the pwm and the pulse length control the direction. Frequency must be 50 Hz to get the 20ms Intervall. Then pulselength 1.0ms is left (5%). 1.5ms is stop (7.5%). 2ms is right (10%). 11-100% is not used. Same with 0%-4%.

ezraholm50, 2020-03-08:
I used it with 100 and 50hz both with dimmer commands where there was a slight difference in speed and what way it turned. Never used the pwm command. I modified all my servos by removing 1 part so it became a continuous servo.

stefanbode, 2020-03-10:
OK, I have now one small FS90R here at my desk and checked a bit of how it can work. In general, it behaves more like a normal motor that is controlled by PWM and not Relay1 and Relay2. Under load, the movement gets slower without the ability to see this in the COUNTER. Therefore my suggestion is to implement the SERVO like a normal shutter with openduration and closeduration. The frequency will be fixed at 50Hz (20ms timeslice). There is no counter needed nor evaluated. Internal DIMMER:50 is stop and DIMMER:40 and DIMMER:60 the two directions. Mayby I have to think also about the NON-continous servos, where the PWM is the POSITION.

dmartella, 2020-07-09:
Hi @stefanbode , thanks for what you have done. I have a question: are you thinking to include in your fork the support for servo or  the only way is to use it as ezraholm50 is suggesting?
thank you
D

stefanbode, 2020-09-13:
Servo control is now possible with the main tasmota project. I committed a new driver including servos. Also soft start and stop is now included. The 360 degree continuous servos are prepared but not implemented yet. Now you are important. How should we measure opening positions? Keep in mind that we can control velocity at start and stop. Therefore just time might be wrong. I have one cheap continuous servo. Works ok, but it is not clear how the pwm duty cycle really behaves. We now already have Pwm min and max, so we can assume near 0 we have full speed in one direction and near 100 into the other direction.

mikep1998, 2020-09-17:
Hello, I would like to use these new features for servo control. What version is it available in? Is the documentation updated for these changes?

Thanks
Mike

mikep1998, 2020-09-18:
I have found your code in the latest release 8.5.0.1. I have been able to get the servo to move a bit not consistently. And the software is doing a "software reset" sometimes. 

I'm not sure if I have it configured correctly. Do you have an example? I have read the documentation online. But I'm not clear that I'm doing it correctly. 

I have done the following:

setoption80 true
ShutterMode 5
PwmFrequency 200 as in documentation and tried 100 and 50. I think 50 should be the correct value but not sure.
setoption15 0

then I run shutteropen and shutterclose some times the servo moves some times the servo does not move?

What am I doing wrong? 

Also the servo I have has 270 degree rotation. I only need it to move about 45 degrees. How do I set the limits?

Thanks

stefanbode, 2020-09-18:
hi @mikep1998 : The problem is the duty cycle. Every servo is different here. There is a new command called shutterpwmrange default is 90,500. This works for me at pwmfrequency 200 best. PWM 90 is 0° and PWM 500 is 180°. If you lower the pwmfrequency the range gets smaller. e.g. pwmfreq100 --> 45,250 (half the values). In this case, the range is only 205 from 0..max. Therefore higher frequency is better, but not all servos like higher frequencies. My servo 400Hz would be great because -> 180..1000 is 820 steps instead of 410, but my servo does not operate correctly anymore on higher frequencies than 200Hz. 
To get YOUR boundaries: start with frequency 50 and use PWM xxx to see what values move you servo from min to max. Then double frequency and check new boundaries. should be ~double the original number.

stefanbode, 2020-09-18:
The "software reset" is an overload of the USB power. The servo take to much power and the voltage drops below the limit. This causes ESP to restart. Had the same behavior here. More power helps and also shuttermotordelay 1 is a cool option to prevent overload. Anyhow: If the pwmrange is wrong the reset is very likely because the relay just flip from one position to another.

mikep1998, 2020-09-18:
Thanks for the answers. I don't quite understand how to my boundaries.
"To get YOUR boundaries: start with frequency 50 and use PWM xxx to see what values move you servo from min to max. Then double frequency and check new boundaries. should be ~double the original number."

Can you show and example how to get these values?

mikep1998, 2020-09-18:
What commands to send to the console.

mikep1998, 2020-09-19:
Hello, I figured out how to find the range settings I have a range of 97 and 120. 
When I do this with a the default time of 100 the servo is moving very very slow.  So I have changed the default times to 2 seconds and that works, but still a little slow. 
When I change the times to 1 then it will not move at all. Because this "abs(Shutter[index].target_position - Shutter[index].real_position ) / Shutter[index].close_velocity > 2)" results in 0 and the code think there is no change in position. also the difference is 2000 the divide by close_velocity results in 0.

I don't think this is the correct code for me to use. Because I am just using a servo to turn a water valve about 40 degrees which takes less than a second to move. I'm not trying to open a shutter that takes a minute.

Do you know of another way to move a servo? I have done it with just pwn1 command and that seems to work fine. But there are no controls on the GUI which the shutters have added.

Thanks
mike

stefanbode, 2021-12-08:
Implemented in main project

closed 2021-12-08 by stefanbode