stefanbode/Sonoff-Tasmota #121: 'bodemota'-minimal.bin

tmarquespt, 2019-01-28:
**Have you look for this feature in other issues and in the wiki?**
yes
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._
When it's not possible to OTA the full .bin, a minimal.bin like tasmota one would be great to avoid loosing the configs which happens if we use the tasmota-minimal.bin in order to OTA a new version of your tasmota. The alternative is using the tasmota-minimal, but then we have to reconnect to the sonoffAP and recofigure or upload the backup.
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

calcipher49, 2019-02-24:
Hi,
I use minimal firmware from standard Tasmota (version 6.x.x) with no loss on config. So specific minimal firmware is not needed.

stefanbode, 2019-02-24:
Yes arend did it very smart. Even in Not included in Compiler all Configs are defined. Therefore the minimal will always work without any loss as long as you don’t start to configure the minimal during runtime. This may cause side effects. Also e.g. the relay locking required for the shutter is not enabled. The use of this version is just to minimize flash usage to upgrade

closed 2019-03-29 by stefanbode