stefanbode/Sonoff-Tasmota #125: Sonoff dual R2 loose wifi connection continuosly

alfonsosj, 2019-02-15:
Sonoff dual R2 looses wifi connection continuosly and after restart it turns on one the switches...

ascillato, 2019-02-15:
Hi,

Please, address this in the Tasmota Support Chat. Thanks

Zoldex, 2019-03-28:
I can confirm the same behaviour with my two duals R2!
The switch turning on after reboot is prtetty annoying.
Any hint?

alfonsosj, 2019-03-29:
I have solved it...! I reflashed all the sonoff with the original Sonoff-Tasmota of arendst and then flash again with this fork of stefanbode and it has solved the problem since one month!! Try ...

Zoldex, 2019-03-29:
just flashed my first dual with tasmota minimal. 
Flashing went good but now i can't find my dual anymore...

stefanbode, 2019-03-29:
Wifi going ON/OFF is normally not an issue of the firmware. Maybe a problem with your WIFI and/or IP address. My branch has nothing specific in this way. Sometimes it might help to switch to a different core of the ESP. This is preconfigured and easy to change in the platform.ini

closed 2019-04-02 by stefanbode

junior013, 2019-05-06:
@stefanbode I think there may be some problem in the code modification (timings?), because I use the same version (6.5.0.3) of original Tasmota and yours on different devices (compiled with the same platform.ini), and originals info and status looks like this (few boot - looks real, and stable link since 27 days):
`Flash write Count 49 at 0xF5000
Boot count 7

ESP-MQT: tele/coffee/STATE = {"Time":"2019-05-06T12:10:40","Uptime":"27T16:15:32","Vcc":3.174,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"JrNet_IoT","BSSId":"F2:9F:C2:F7:E3:6B","Channel":1,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:04"}}`

your version looks like this (much more reboot and a lot of downtime):
`Flash write Count 3255 at 0xF4000
Boot count 305

ESP-MQT: tele/livingroom/STATE = {"Time":"2019-05-05T22:58:50","Uptime":"29T07:06:43","Vcc":3.427,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"JrNet_IoT","BSSId":"F2:9F:C2:F7:E3:6B","Channel":1,"RSSI":76,"LinkCount":9100,"Downtime":"0T21:03:16","DeepSleep":0,"Heap":24680"}}`