stefanbode/Sonoff-Tasmota #126: Wall momentary buttons

B1G1, 2019-02-18:
<<<<<<< HEAD
=======
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

>>>>>>> 8f585b7f5354b19aff4bc6d1f238df4588d5693f
**Describe the bug**
I still have the problem reported here https://github.com/stefanbode/Sonoff-Tasmota/issues/89
I push up or down button (momentary buttons) to move the shutter and when I want to stop it I push the same button a second time, then the shutter stops but after few seconds it start to move again until it's completely open if I was opening it or until it's completely closed if I was closing it.

here a log, in this case I was trying to calibrate the shutter but I could not do it.
`10:48:55 CMD: shutterclose
10:48:55 SRC: WebConsole from 192.168.1.50
10:48:55 RSL: Group 0, Index 1, Command SHUTTERCLOSE, Data 
10:48:55 shutterposition in: payload 0, index 1, source 7
10:48:55 lastsource 7:, realpos 47200, target 0, payload 0
10:48:55 Start shutter in right direction -1
10:48:56 SRC: Shutter
10:48:56 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"ON"}
10:48:56 MQT: stat/tapparellaStudio/POWER2 = ON (retained)
10:48:56 Delay6 5s, xdrv 21
10:48:56 MQT: stat/tapparellaStudio/RESULT = {"shutterclose1":0}
10:48:57 SRC: PulseTimer
10:48:57 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"OFF"}
10:48:57 MQT: stat/tapparellaStudio/POWER2 = OFF (retained)
10:48:57 Shutter 0: Real Pos: 44475, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 1.3  [s]
10:48:58 Shutter 0: Real Pos: 42295, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 2.2  [s]
10:48:59 Shutter 0: Real Pos: 40115, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 3.2  [s]
10:49:00 Shutter 0: Real Pos: 37935, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 4.2  [s]
10:49:01 Shutter 0: Real Pos: 35755, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 5.2  [s]
10:49:02 Shutter 0: Real Pos: 33575, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 6.2  [s]
10:49:03 Shutter 0: Real Pos: 31395, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 7.2  [s]
10:49:04 Shutter 0: Real Pos: 29215, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 8.3  [s]
10:49:05 Shutter 0: Real Pos: 27035, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 9.3  [s]
10:49:06 Shutter 0: Real Pos: 24855, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 10.3  [s]
10:49:07 Shutter 0: Real Pos: 22675, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 11.3  [s]
10:49:08 Shutter 0: Real Pos: 20495, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 12.2  [s]
10:49:09 Shutter 0: Real Pos: 18315, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 13.3  [s]
10:49:10 Shutter 0: Real Pos: 16135, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 14.3  [s]
10:49:11 Shutter 0: Real Pos: 14064, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 15.2  [s]
10:49:12 Shutter 0: Real Pos: 11884, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 16.2  [s]
10:49:13 Shutter 0: Real Pos: 9704, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 17.2  [s]
10:49:14 Shutter 0: Real Pos: 7524, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 18.2  [s]
10:49:15 Shutter 0: Real Pos: 5344, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 19.2  [s]
10:49:16 Shutter 0: Real Pos: 3164, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 20.2  [s]
10:49:17 Shutter 0: Real Pos: 984, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 21.2  [s]
10:49:17 Shutter 0: Real Pos. 0, Stoppos: 0, relay: 1, direction -1, pulsetimer: 5, rtcshutter: 21.8 [s], operationtime 0
10:49:17 MQT: stat/tapparellaStudio/SHUTTER1 = 0 (retained)
10:49:17 SRC: Shutter
10:49:17 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"ON"}
10:49:17 MQT: stat/tapparellaStudio/POWER2 = ON (retained)
10:49:17 MQT: tele/tapparellaStudio/RESULT = {"SHUTTER-1":{"position":0, "direction":0}}
10:49:18 SRC: PulseTimer
10:49:18 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"OFF"}
10:49:18 MQT: stat/tapparellaStudio/POWER2 = OFF (retained)
10:49:18 CFG: Saved to flash at FB, Count 46, Bytes 3824
10:49:22 CMD: SHUTTER50PERCENT 50
10:49:22 SRC: WebConsole from 192.168.1.50
10:49:22 RSL: Group 0, Index 1, Command SHUTTER50PERCENT, Data 50
10:49:22 Shutter accuracy digits: 1
10:49:22 Shutter 0 (Relay:1): Init. Pos: 0 [0 %], Open Vel.: 100 Close Vel.: 109 , Max Way: 47200, Opentime 23.6 [s], Closetime 21.5 [s], CoedffCalc: c0: 0, c1 472, c2: 472, c3: 0, c4: 0, binmask 3, is inverted 0, shuttermode 2
10:49:22 MQT: stat/tapparellaStudio/RESULT = {"shutter50percent1":50}
10:49:23 CFG: Saved to flash at FA, Count 47, Bytes 3824
10:49:47 APP: Button1 multi-press 1
10:49:48 SRC: Button
10:49:48 Delay2 5s
10:49:48 Shutter 0: Target: 47200, powerstatelocal 1
10:49:48 MQT: stat/tapparellaStudio/RESULT = {"POWER1":"ON"}
[... 58 lines cut ...]
>>>>>>> 8f585b7f5354b19aff4bc6d1f238df4588d5693f
- [ ] _Provide the output of command_``status 0`` :
```
11:09:11 CMD: status 0
11:09:11 SRC: WebConsole from 192.168.1.50
11:09:11 RSL: Group 0, Index 1, Command STATUS, Data 0
11:09:11 MQT: stat/tapparellaStudio/STATUS = {"Status":{"Module":39,"FriendlyName":["tapparellaStudio","tapparellaStudio2"],"Topic":"tapparellaStudio","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
11:09:11 MQT: stat/tapparellaStudio/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:26:24","StartupUTC":"2019-02-18T09:42:47","Sleep":50,"BootCount":7,"SaveCount":48,"SaveAddress":"F9000"}}
11:09:11 MQT: stat/tapparellaStudio/STATUS2 = {"StatusFWR":{"Version":"6.4.1.9 stb-1.1(sonoff)","BuildDateTime":"2019-02-10T21:58:53","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
11:09:11 MQT: stat/tapparellaStudio/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ2.4",""],"TelePeriod":300,"SetOption":["0000C029","558180C0","C0000000"]}}
11:09:11 MQT: stat/tapparellaStudio/STATUS4 = {"StatusMEM":{"ProgramSize":517,"Free":484,"Heap":18080,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":1327185,"Features":["00000003","00000809","0FDAE794","000383A1","23B617CE"]}}
11:09:11 MQT: stat/tapparellaStudio/STATUS5 = {"StatusNET":{"Hostname":"tapparellaStudio-6592","IPAddress":"192.168.1.24","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"DC:4F:22:86:D9:C0","Webserver":2,"WifiConfig":2}}
11:09:11 MQT: stat/tapparellaStudio/STATUS6 = {"StatusMQT":{"MqttHost":"Hassio","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_86D9C0","MqttUser":"B1G","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:09:11 MQT: stat/tapparellaStudio/STATUS7 = {"StatusTIM":{"UTC":"Mon Feb 18 10:09:11 2019","Local":"Mon Feb 18 11:09:11 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:53","Sunset":"18:15"}}
11:09:11 MQT: stat/tapparellaStudio/STATUS10 = {"StatusSNS":{"Time":"2019-02-18T11:09:11","SHUTTER-1":{"position":100, "direction":0}}}
11:09:11 MQT: stat/tapparellaStudio/STATUS11 = {"StatusSTS":{"Time":"2019-02-18T11:09:11","Uptime":"0T00:26:24","Vcc":3.156,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"FRITZ2.4","BSSId":"C8:0E:14:1E:C4:9C","Channel":1,"RSSI":100},"DeepSleep":0,"Heap":18040}}
```

**To Reproduce**
Push wall buttons one time to move te shutter and a second time to stop it. 

**Expected behavior**
The shutter should remain in the position where it was stopped with the second push of the button

**(Please, remember to close the issue when the problem has been addressed)**

B1G1, 2019-02-21:
@stefanbode any chanche to get this bug fixed?

stefanbode, 2019-04-02:
Hi Bigi, I'm now back and will take a look into it now. Maybe I can find a root cause in the log files you shared perfectly

B1G1, 2019-04-03:
thank you, it's ok now!

closed 2019-04-03 by B1G1