stefanbode/Sonoff-Tasmota #127: separate auth for power control

devZer0, 2019-02-18:
**Have you look for this feature in other issues and in the wiki?**
yes

**Is your feature request related to a problem? Please describe.**
you can only control power on/off  without auth or with admin-password  - but both also allows administration of the tasmota device. 

**Describe the solution you'd like**
i'd like a separate authentication/user for power control, i.e. administration of the device and usage of the device should be separate things

**Describe alternatives you've considered**
i have asked in the upstream project if this feature would be possible but the project denied that such feature woud be accepted/implemented

calcipher49, 2019-02-20:
Hi,
I fear it's not possible if arendst refused it.
As workaround, you can put webserver in user mode, this remove all administration options. Command is: WebServer 1
To come back to full webserver mode, you must launch command from mqtt: WebServer 2

stefanbode, 2019-02-21:
Yes, agree. From the current limitation of memory it does not make sense to have 2 users. My suggestion anyway is to use MQTT and shut down the webinterface completly. The. You can control access though your home automation software.

closed 2019-02-21 by stefanbode