stefanbode/Sonoff-Tasmota #128: "shutterposition1":100 stoped at 58%

gpoerzgen, 2019-02-19:
Hallo Markus,

ich habe deine angepasste Tasmota Software bei 8 Sonoff 4CH Pro R2 im Einsatz.
Davon funktionieren 7 tadellos, nur einer zickt herum.

Tasmota Version: Sonoff-Tasmota-6.3.2-Shutter-StefanBode

1. Schalter S6 auf 1
2. DIP K5 1111
3. DIP K6 0000
4. Über Konsole:  Backlog SETOPTION80 1; SETOPTION14 1; SETOPTION81 1; SHUTTERRELAY2 3; SHUTTEROPENDURATION1 30; SHUTTERCLOSEDURATION1 30; SHUTTEROPENDURATION2 30; SHUTTERCLOSEDURATION2 30; TelePeriod 10

Wenn ich die Rolladen vollständig rauf oder runter fahre, wird beim herauf fahren bei 58% (17s) und beim herunterfahren bei 41% (17s) gestoppt.
Die Rolladen läuft also jeweils nur 17s und nicht die 30s die ich konfiguriert habe.
Hast du eine Idee woran das liegen kann.

Schöne Grüße
Guido

Logfile mit Loglevel 5:

02:15:34 RSL: empfangenes topic /shutterposition1, Datengröße 3, Daten 100
02:15:34 RSL: Gruppe 0, Index 1, Befehl SHUTTERPOSITION, Daten 100
02:15:34 shutterposition in: payload 100, index 1, source 7
02:15:34 lastsource 7:, realpos 0, target 60000, payload 100
02:15:34 Start shutter in right direction 1
02:15:35 temp1 for mask is 3, temp2: 1, temp3 2. Bucketsize: 2, device: 1 to state 1, currentsate 0
02:15:35 imask1: 2
02:15:35 SRC: Shutter
02:15:35 Switched relay: 1 by Shutter
02:15:35 MQT: sonoff/stat/sonoff-4chpro-8/RESULT = {"POWER1":"ON"}
02:15:35 MQT: sonoff/stat/sonoff-4chpro-8/POWER1 = ON
02:15:35 Delay6 5s, xdrv 21
02:15:35 MQT: sonoff/stat/sonoff-4chpro-8/RESULT = {"shutterposition1":100}
02:15:35 Shutter 0: Real Pos: 1100, Target 60000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 0.6  [s]
02:15:35 CFG: in Flash gespeichert am F8, zählen 84, Bytes 3824
02:15:35 HTP: Konsole
02:15:35 Shutter 0: Real Pos: 2800, Target 60000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 1.4  [s]
02:15:36 Shutter 0: Real Pos: 4800, Target 60000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 2.4  [s]
02:15:37 Shutter 0: Real Pos: 6700, Target 60000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 3.3  [s]
02:15:38 Shutter 0: Real Pos: 8700, Target 60000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 4.3  [s]
02:15:39 Shutter 0: Real Pos: 10700, Target 60000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 5.3  [s]
02:15:40 Shutter 0: Real Pos: 12700, Target 60000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 6.3  [s]
02:15:41 Shutter 0: Real Pos: 14700, Target 60000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 7.3  [s]
02:15:42 Shutter 0: Real Pos: 16700, Target 60000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 8.4  [s]
02:15:43 Shutter 0: Real Pos: 18700, Target 60000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 9.4  [s]
02:15:44 Shutter 0: Real Pos: 20800, Target 60000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 10.4  [s]
02:15:45 Shutter 0: Real Pos: 22800, Target 60000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 11.4  [s]
02:15:46 Shutter 0: Real Pos: 24800, Target 60000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 12.4  [s]
02:15:47 Shutter 0: Real Pos: 26800, Target 60000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 13.4  [s]
02:15:48 Shutter 0: Real Pos: 28800, Target 60000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 14.4  [s]
02:15:49 Shutter 0: Real Pos: 30800, Target 60000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 15.4  [s]
02:15:50 WIF: Prüfe Verbindung...
02:15:50 WIF: verbunden
02:15:50 Shutter 0: Real Pos: 32800, Target 60000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 16.4  [s]
02:15:51 Shutter 0: Real Pos: 34800, Target 60000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 17.4  [s]
02:15:52 APP: Knopf1 level 1-0
02:15:52 SRC: Button
02:15:52 Switched relay: 1 by Button
02:15:52 Shutter 0: Switch OFF motor. Target: 35100, source: Button, powerstate_local 0, switchedRelay 1, manual change 1
02:15:52 Shutter 0: Target: 35100, powerstatelocal 0
02:15:52 MQT: sonoff/stat/sonoff-4chpro-8/RESULT = {"POWER1":"OFF"}
02:15:52 MQT: sonoff/stat/sonoff-4chpro-8/POWER1 = OFF
02:15:52 Shutter 0: Real Pos. 35200, Stoppos: 58, relay: 0, direction 1, pulsetimer: 0, rtcshutter: 17.6 [s], operationtime 2
02:15:52 MQT: sonoff/stat/sonoff-4chpro-8/SHUTTER1 = 58
02:15:52 CFG: in Flash gespeichert am F7, zählen 85, Bytes 3824
02:16:10 WIF: Prüfe Verbindung...
02:16:10 WIF: verbunden

titomontes, 2019-02-19:
Very useful german, really.

stefanbode, 2019-02-19:
I can see in the log file that the button was pressed manually or by something else. The target is still 60000 and the position  34800. The software will not stop, but there is a SWITCH command. Something pressed the STOP button at 02:15:52

02:15:50 Shutter 0: Real Pos: 32800, Target 60000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 16.4 [s]
02:15:51 Shutter 0: Real Pos: 34800, Target 60000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 17.4 [s]
02:15:52 APP: Knopf1 level 1-0
02:15:52 SRC: Button
02:15:52 Switched relay: 1 by Button
02:15:52 Shutter 0: Switch OFF motor. Target: 35100, source: Button, powerstate_local 0, switchedRelay 1, manual change 1

closed 2019-03-29 by stefanbode