stefanbode/Sonoff-Tasmota #129: Error compiling

titomontes, 2019-02-19:
Hello

I just tried to flash this fork (in platformio/atom), but for some reason  user_config_override gets triggered:

> Warning! Ignore unknown `build_unflags` option in `[platformio]` section
> Processing sonoff (platform: espressif8266@1.5.0; board: esp01_1m; framework: arduino)
>  
> Warning! `board_flash_mode` option is deprecated and will be removed in the next release! Please use `board_build.flash_mode` inst
> ead.
> Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
> 
> PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
> HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)
> Converting sonoff.ino
> C:/Users/Tito/Downloads/Stefanbode/sonoff/xsns_96_node_bridge.ino:28:0: warning: "NODE_BRIDGE_BUFFER_SIZE" redefined [enabled by d
> efault]
> #define NODE_BRIDGE_BUFFER_SIZE 45
> ^
> C:/Users/Tito/Downloads/Stefanbode/sonoff/xdrv_96_node_bridge.ino:28:0: note: this is the location of the previous definition
> #define NODE_BRIDGE_BUFFER_SIZE 160
> ^
> C:/Users/Tito/Downloads/Stefanbode/sonoff/xsns_96_node_bridge.ino:31:0: warning: "D_UPDATED" redefined [enabled by default]
> #define D_UPDATED "Aktualizacja Gaz"
> ^
> C:/Users/Tito/Downloads/Stefanbode/sonoff/xdrv_96_node_bridge.ino:40:0: note: this is the location of the previous definition
> #define D_UPDATED "Aktualizacja"
> 
> Library Dependency Finder -> http://bit.ly/configure-pio-ldf
> LDF MODES: FINDER(chain) COMPATIBILITY(soft)
> Collected 57 compatible libraries
> Scanning dependencies...
> 
> Dependency Graph
> |-- <PubSubClient> 2.6
> |-- <NeoPixelBus> 2.2.9
> |   |-- <SPI> 1.0
> |-- <IRremoteESP8266> 2.5.2
> |-- <Adafruit ILI9341> 1.2.0
> |   |-- <SPI> 1.0
> |   |-- <Adafruit GFX Library> 1.2.9
> |   |   |-- <SPI> 1.0
> |-- <Adafruit SSD1306> 1.1.2
> |   |-- <SPI> 1.0
> |   |-- <Adafruit GFX Library> 1.2.9
> |   |   |-- <SPI> 1.0
> |   |-- <Wire> 1.0
> |-- <MQTT> 2.4.0
> |-- <SPI> 1.0
> |-- <BME680_driver-bme680_v3.5.9>
> |-- <ESP8266WiFi> 1.0
> |-- <I2Cdevlib-Core>
> |   |-- <Wire> 1.0
> |-- <TasmotaSerial> 2.2.0
> |   |   |-- <Wire> 1.0
> |-- <Ticker> 1.0
> |-- <Adafruit CCS811 Library> 1.0.0
> |   |-- <Wire> 1.0
> |-- <rc-switch> 2.6.2
> |-- <ESP8266httpUpdate> 1.1
> 
> |   |-- <ESP8266WiFi> 1.0
> |   |-- <ESP8266HTTPClient> 1.1
> |   |   |-- <ESP8266WiFi> 1.0
> |-- <TasmotaModbus> 1.1.0
> |   |-- <TasmotaSerial> 2.2.0
> |-- <Mutichannel_Gas_Sensor> 0.0.1
> |   |-- <Wire> 1.0
> Compiling .pioenvs\sonoff\src\core_esp8266_timer.c.o
> Compiling .pioenvs\sonoff\src\core_esp8266_wiring_digital.c.o
> Compiling .pioenvs\sonoff\src\core_esp8266_wiring_pwm.c.o
> Compiling .pioenvs\sonoff\src\sonoff.ino.cpp.o
> Compiling .pioenvs\sonoff\lib001\PubSubClient-EspEasy-2.6.09\PubSubClient.cpp.o
> Compiling .pioenvs\sonoff\libe67\SPI\SPI.cpp.o
> Compiling .pioenvs\sonoff\libd76\NeoPixelBus-2.2.9\internal\HsbColor.cpp.o
> Compiling .pioenvs\sonoff\libd76\NeoPixelBus-2.2.9\internal\HslColor.cpp.o
> Compiling .pioenvs\sonoff\libd76\NeoPixelBus-2.2.9\internal\HtmlColor.cpp.o
> Compiling .pioenvs\sonoff\libd76\NeoPixelBus-2.2.9\internal\HtmlColorNameStrings.cpp.o
> Compiling .pioenvs\sonoff\libd76\NeoPixelBus-2.2.9\internal\HtmlColorNames.cpp.o
> C:/Users/Tito/Downloads/Stefanbode/sonoff/sonoff.ino:34:36: fatal error: user_config_override.h: No such file or directory
> #include "user_config_override.h"         // Configuration overrides for my_user_config.h
> ^
> compilation terminated.
> *** [.pioenvs\sonoff\src\sonoff.ino.cpp.o] Error 1
>  [ERROR] Took 6.29 seconds
>  
>  [SUMMARY]
> Environment wemos               [SKIP]
> Environment nodemcuv2           [SKIP]
> Environment wemeos-ds18x20-PL   [SKIP]
> Environment sonoff              [ERROR]
>  [ERROR] Took 6.29 seconds

titomontes, 2019-02-19:
OK, I just read some posts and found out about commenting language and other stuff. I got the branch flashed but as a NodeMCU!

I don´t know if there will be issues with it, i´ll report later

Thanks again

localhost61, 2019-02-20:
@titomontes don't be afraid, NodeMCU is default device, it will work with ESP8266/8285  and 1MB or more flash. No need to change.

titomontes, 2019-02-25:
Thanks

It seems to be running fine as a NodeMCU.

stefanbode, 2019-03-29:
You can define, if you want your own config. This makes sense in 99% of the use cases. Then you can put in e.g. your credentials of the WIFI and do not need to set it manually. If it compiles with or without your user config is defined in the platform.ini. Use the current version.  Uncomment line 148 to have the overide file included. default is now OFF

closed 2019-04-02 by stefanbode