stefanbode/Sonoff-Tasmota #130: Normal shutter switch. How to connect and properly configure?

titomontes, 2019-02-25:
Hello again: 

I want to use a normal shutter switch, which has three positions: Up, Down and a central Stop (or off). 

[image]

I´m not sure on how the connections should be made (already pasted on another thread, with no responses). Any help will be much appreciated.

calcipher49, 2019-02-25:
Hi,
This shutter switch is indeed 2 bistable switchs, central position set both off.
Use as follows:
- Common entry (in red on switch) to GND
- Up entry to GPIO A, configured as Switch1 in Tasmota
- Down entry to GPIO B configured as Switch2 in Tasmota
Use any available GPIO (except GPIO0) on your device for A and B
You should use Switches, not Buttons, in Tasmota configuration

titomontes, 2019-02-26:
Thanks a lot.

I understand that GPIO0 can be used as long as I disable serial logging, right?

stefanbode, 2019-02-26:
You connect L with the 5v of your esp and up and down with two GPIO. Do not connect the switch to 220v. This will create a hell fire.

titomontes, 2019-02-26:
> You connect L with the 5v of your esp and up and down with two GPIO. Do not connect the switch to 220v. This will create a hell fire.

LOL. Off course, I did know that. I´ve already implemented several sonoff basic around the house using GPIO14 as a normal switch, and already have this switch properly working. 

BUT

For the life of me, I haven´t been able to properly configure MQTT in HA.

This is my current config:

```
  - platform: mqtt
    name: "Persiana Salon"
    availability_topic: "salon/persiana1/tele/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"
    position_topic: stat/salon/persiana1/SHUTTER1
    position_open: 100
    position_closed: 0
    set_position_topic: "salon/persiana1/cmnd/shutterposition"
    command_topic: "salon/persiana1/cmnd/backlog"
    payload_open: "SHUTTEROPEN"
    payload_close: "SHUTTERCLOSE"
    payload_stop: "SHUTTERSTOP"
    retain: false
    optimistic: false
    qos: 1
```

And tried publishing through another client on my broker with `salon/persiana1/cmnd/backlog` and it accepts `SHUTTEROPEN` and `SHUTTERCLOSE`, but I don´t know how to make the shutter available on HA (shows it as unavailable, but the topic `salon/persiana1/LWT` shows it as `Online`

titomontes, 2019-03-03:
FINALLY!

I can confirm the shutters working fully wit this config:

```
  - platform: mqtt
    name: "Persiana Salon"
    availability_topic: "salon/persianaA/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"
    position_topic: salon/persianaA/SHUTTER1
    position_open: 100
    position_closed: 0
    set_position_topic: "salon/persianaA/cmnd/shutterposition"
    command_topic: "salon/persianaA/cmnd/backlog"
    payload_open: "SHUTTEROPEN"
    payload_close: "SHUTTERCLOSE"
    payload_stop: "SHUTTERSTOP"
    retain: false
    optimistic: false
    qos: 1
```
where `salon/persianaA` is the full topic (in reality, if you go to the information page in Tasmota, the full topic will be `salon/persianaA/cmnd`).

I´m gonna modify the Wiki so this sticks.

Cheers

stefanbode, 2019-03-04:
Thanks for updating the wiki!

closed 2019-03-04 by stefanbode

titomontes, 2019-03-07:
Oh crap. Turns out your initial config was completely right. I didn´t realize that I had changed my default MQTT Full topic, so all the MQTT spelling changed. I´ve returned to standard and initial config works perfectly. I´m fixing you wiki (back)

SOO sorry.