stefanbode/Sonoff-Tasmota #131: Shutter blinds position at HA reboot

jokerigno, 2019-03-02:
<<<<<<< HEAD
=======
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

>>>>>>> 8f585b7f5354b19aff4bc6d1f238df4588d5693f
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Hardware used : 
- [ x] If a pre-compiled release or development binary was used, which one? :
- [x ] You have tried latest release or development binaries? :
- [ x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**
I'm using your fork to manage my shutter blinds and everything is ok. 

The only "issue" is that if i Reboot my Home Assistant I can't see shutter position anymore.

There's a way to fix it?

Here's my code for covers in HA:

```
cover:
  - platform: mqtt
    name: "Tapparella"
    availability_topic: "tapparellatele/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"
    position_topic: "tapparella/stat/SHUTTER1"
    position_open: 100
    position_closed: 0
    set_position_topic: "tapparella/cmnd/shutterposition"
    value_template: "{{ value | int }}"
    command_topic: "tapparella/cmnd/backlog"
    payload_open: "SHUTTEROPEN"
    payload_close: "SHUTTERCLOSE"
    payload_stop: "SHUTTERSTOP"
    retain: false
    optimistic: false
    qos: 0
```

titomontes, 2019-03-03:
It seems you need to set `PowerRetain 1` in the console. 

It´s working for me for the time being.

jokerigno, 2019-03-18:
You are right! 

Thank you for your feedback. Issue closed.

closed 2019-03-18 by jokerigno