stefanbode/Sonoff-Tasmota #133: Core 2.4 known bug

titomontes, 2019-03-08:
Hello again

I just wanted to know if it is possible to compile your fork for Core v2.3, since 2.4.2 and 2.5 are buggy and are known to cause MQTT and WiFi reconnects. I already tried to compile changing the platformio.ini file like this:

```
[env:nodemcuv2]
platform = espressif8266 v2.3.0
board = nodemcuv2
framework = arduino
```

But nothings seems to happen (trying just a build)

My sonoff dual is working properly right now, but, as expected, MQTT reconnects every 2 minutes or so. It causes the switch to appear unresponsive sometimes. 

Thanks

stefanbode, 2019-03-08:
Dont know if it is just a typo but the platform definition must have an @ between the number and fist definition. And 2.3 is 1.5. See below
```
[common] ; ************************************************************
; *** Esp8266 core for Arduino version 2.3.0
platform = espressif8266@1.5.0
; *** Esp8266 core for Arduino version 2.4.2
;platform = espressif8266@1.8.0
```

titomontes, 2019-03-08:
Well, the platformio.ini comes as follows

```
[env:sonoff]
platform = espressif8266@1.5.0
;platform = espressif8266
;platform = espressif8266_stage
framework = arduino
board = esp01_1m
board_flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.1m0.ld -DUSE_CONFIG_OVERRIDE
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON, i2cdevlib-ADS1115
upload_speed = 115200

[env:wemos]
;platform = espressif8266@1.5.0
;platform = espressif8266_stage
platform = espressif8266
;platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
framework = arduino
board = d1_mini
; *** Fix espressif8266@1.7.0 induced undesired all warnings
build_unflags = -Wall

; Select one of two board_flash_mode options below
; Sonoff Basic et al. (ESP8266 uses dio or qio)
board_build.flash_mode = dout
; Sonoff Touch and Sonoff 4CH (ESP8285 uses dout)
; board_flash_mode = dout

;build_flags = -DMQTT_MAX_PACKET_SIZE=512
;lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON, i2cdevlib-ADS1115
lib_deps = PubSubClient, NeoPixelBus, ArduinoJSON,IRremoteESP8266,  i2cdevlib-ADS1115
upload_speed = 921600
build_flags =
  -DVTABLES_IN_FLASH
  -DPIO_FRAMEWORK_ARDUINO_LWIP_HIGHER_BANDWIDTH
  -DUSE_CONFIG_OVERRIDE

[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino

; Select one of two board_flash_mode options below
; Sonoff Basic et al. (ESP8266 uses dio or qio)
board_flash_mode = dout
; Sonoff Touch and Sonoff 4CH (ESP8285 uses dout)
; board_flash_mode = dout

build_flags = -DMQTT_MAX_PACKET_SIZE=1000
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON, i2cdevlib-ADS1115
upload_speed = 115200

[env:wemeos-ds18x20-PL]
platform = espressif8266
framework = arduino
board = d1_mini
board_flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.16m.ld -DMQTT_MAX_PACKET_SIZE=707 -DUSE_DS18x20 -DMESSZ=600 -DMY_LANGUAGE=pl-PL
lib_deps = PubSubClient,NeoPixelBus, ArduinoJSON, IRremoteESP8266, i2cdevlib-ADS1115, OneWire
upload_speed = 115200
; Serial Monitor options
monitor_baud = 115200
```

I think it´s a typo, but no errors are registered when compiling. So the @1.5.0 part should *always* be added? (It´s missing in two definitions)

titomontes, 2019-03-13:
OK

I'm just writing to thank Mr @stefanbode for this great fork. 

I just installed the aforementioned  Dual R2 module on a big shutter, and it´s working flawlessly (compiled with core 2.3.0). 

Anyone wanting to install it can ask.

Thanks again.

closed 2019-03-17 by titomontes