stefanbode/Sonoff-Tasmota #134: Update to New Tasmota Template

rt400, 2019-03-10:
Hi ,
First.. your work amazing and i used it alot.
now i have a 3ch gand wifi touch switch with your firmware.
but i want that the 1ch and the 3ch will be used for up and down and the middle will stop any run.
so for that i change the code. and if you will update to new firmware so i can make template .

best regards
yuval

stefanbode, 2019-03-29:
Updated the code to the newest development branch. let me know how to support.

calcipher49, 2019-03-31:
Hi, 
New version 6.5.0.3 is fine, but templates doesn't works.
On web interface - Configuration - Configure Template:
`File Not Found

URI: /tp
Method: GET
Arguments: 0`

stefanbode, 2019-03-31:
Thanks. Let me check. Maybe I messed with the Webserver.ino

closed 2019-04-02 by stefanbode