stefanbode/Sonoff-Tasmota #136: Expose as blinds in Google Home

jokerigno, 2019-03-25:
Hi,

first of all, again, thank you for your work.

I noticed that since few days google assistant has introduced blinds management

(https://developers.google.com/actions/smarthome/guides/blinds)

Do you think is possible to change exposure from light (actual) to blinds, in order to use better voice commands?

Thank you in advance

stefanbode, 2019-03-25:
Hi, yes for sure this is a great idea. I'm already a bit behind the actual version. Too much preparation for next week HMI fare trade in Hannover.

jokerigno, 2019-03-31:
I saw you updated the release. Any news on this?

stefanbode, 2019-04-01:
Not yet. I got the info that the new template page on the webinterface does not work currently. I have to check this. Currently we are now in sync again with the master branch. This also means that the interlock mechanism now works as defined in the master branch. I removed my implementation.

stefanbode, 2019-04-02:
I try to get an idea on how to implement this and gave up. If you have some hints on how to do it please be welcome and help me.

closed 2019-04-02 by stefanbode

jokerigno, 2019-04-07:
Don't ask how or why. But since last google assistant sync cover are exposed as cover also in google home app!

titomontes, 2019-05-16:
> Don't ask how or why. But since last google assistant sync cover are exposed as cover also in google home app!

So how would one go about and set this for google home??

stefanbode, 2019-05-16:
@jokerigno : It would be great if you can enhance the wiki for the shutter and explain how you get it working with google home. This will help all the others too

jokerigno, 2019-05-19:
well: I don't know. I simply exposed shutters in google home and they were exposed as lights (so I was saying "turn on - off" .

Then they simply became covers. And I didn't change anything in my config, only sync my devices.

I'm still with version 6.1.3 and didn't update because I'm scared to loose it !