stefanbode/Sonoff-Tasmota #137: Short Circuit Safe Configuration broken again

tom211, 2019-03-30:
Hi,
first of all thanks for the great work!
I have built an ESP Module based on this: https://luani.de/projekte/esp8266-hvio/
I have made some changes regarding the use as roller shutter device.
The main change is the use of other relais in short circuit fail safe configuration.
The Hardware is tested and works well.

When i use the actual version, the fail safe configuration doesnt work. For up and down only one relay is activated. Its the same behaviour like here: #69 
I have tested the version from  Sep 30, 2018. Everithings works as expected!

I have one additional question:
When performing "shutterclose" both relays are activated and after the runtime only the on/off relay is deactivated. The relay for the direction stays activated (in down position). I think this is not necessary and will lead to unnecessary power consumption...
Would be nice to get a solution for this, so that both relays are deactivated.

Thanks 
Thomas

stefanbode, 2019-03-31:
This a is correct behavior for the circuit safe configuration. Relay one is on/off and the second relay is up/down. Therefore relay one is on always during operation. And relay two depends on up down and stay in this state after reaching end position. Clear?

tom211, 2019-04-01:
Hi, 
thanks for the quick reply. 

First, i think there is a problem with the fail safe mode in the actual software version. 
Shutteropen activates relay1 and not relay2
Shutterclose activates relay2 and **not relay1**

Its clear, that this is the correct behavior. 
But when thinking about energy consumption it is not necessary to hold the second relay in the down position when the first relay is off. 

Example:
Roller shutters are closed at the evening: 
Relay 1: on, relay 2: on;
After closing: Relay1: off, **relay2: on**
The relay is on until the shutter is opened in the morning, then relay2 switches to off and relay1 switches on and after movement to off. After that relay1 and relay2 are off.

I think it would be a nice feature to lower the power consumption if relay2 is off everytime when relay1 is off.

I have tested a little bit and came to the solution that it can be implemented with a rule:
_Rule1 on Power1#State=0 do Power2 0 endon_

I have made some rules for using the switches in short circuit configuration for up and down instead for on and off of the single relays. 
So the problem, you mentioned here (https://github.com/stefanbode/Sonoff-Tasmota/issues/69#issuecomment-433716972) can be solved with another rule:
_Rule2 on switch1#state=1 do shutteropen endon on switch1#state=0 do shutterstop endon on switch2#state=1 do shutterclose endon on switch2#state=0 do shutterstop endon_

For me everithing works with the version from Sep 30, 2018 and the mentioned rules.
But in the actual version i can not get the fail safe mode working.

When im back home at the device, i can test some things or give some logs if you want.

Thanks very much and many greetings
Thomas

stefanbode, 2019-04-01:
Ok fine. I did not make a change; I hope. Just upgrade to the new version. Maybe an issue with the new interlock definition.

tom211, 2019-04-01:
Here is the log with setoption 80 1 and setoption81 0 and no interlock and no pulsetime:

00:00:00 Shutter accuracy digits: 1
00:00:00 Shutter 0 (Relay:1): Init. Pos: 800 [4 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 20000, Opentime 10.0 [s], Closetime 10.0 [s], CoedffCalc: c0: 0, c1 200, c2: 200, c3: 0, c4: 0, binmask 3, is inverted 0, shuttermode 0
00:00:00 Project sonoff Sonoff Version 6.5.0.3 stb-1.1(sonoff)-2_4_2
00:00:00 WIF: Connecting to AP1 123 in mode 11N as sonoff-3377...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff-3377 with IP address 192.168.2.126
21:26:19 MQT: Attempting connection...
21:26:19 MQT: Connected
21:26:19 MQT: tele/sonoff/LWT = Online (retained)
21:26:19 MQT: cmnd/sonoff/POWER = 
21:26:19 MQT: tele/sonoff/INFO1 = {"Module":"Luani HVIO","Version":"6.5.0.3 stb-1.1(sonoff)","FallbackTopic":"cmnd/DVES_4C2D31_fb/","GroupTopic":"sonoffs"}
21:26:19 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-3377","IPAddress":"192.168.2.126"}
21:26:19 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
21:26:19 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
21:26:19 MQT: stat/sonoff/POWER1 = OFF
21:26:19 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
21:26:19 MQT: stat/sonoff/POWER2 = OFF
21:26:27 MQT: tele/sonoff/STATE = {"Time":"2019-04-01T21:26:27","Uptime":"0T00:00:14","Vcc":3.532,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"123","BSSId":"123","Channel":11,"RSSI":40,"LinkCount":1,"Downtime":"0T00:00:04","DeepSleep":0,"Heap":24256"}}
21:26:27 MQT: tele/sonoff/SENSOR = {"Time":"2019-04-01T21:26:27","Switch1":"ON","Switch2":"ON","SHUTTER-1":{"position":4, "direction":0}}
21:27:32 CMD: status0
21:27:32 MQT: stat/sonoff/STATUS = {"Status":{"Module":35,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:27:53 CMD: shutteropen
21:27:53 Start shutter in right direction 1
21:27:54 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
21:27:54 MQT: stat/sonoff/POWER1 = ON
21:27:54 MQT: stat/sonoff/RESULT = {"shutteropen1":100}
21:27:54 Shutter 0: Real Pos: 3400, Target 20000, source: Shutter, start-pos: 4 %, direction: 1, rtcshutter: 1.3  [s]
21:27:55 Shutter 0: Real Pos: 5400, Target 20000, source: Shutter, start-pos: 4 %, direction: 1, rtcshutter: 2.3  [s]
21:27:56 Shutter 0: Real Pos: 7400, Target 20000, source: Shutter, start-pos: 4 %, direction: 1, rtcshutter: 3.3  [s]
21:27:57 Shutter 0: Real Pos: 9400, Target 20000, source: Shutter, start-pos: 4 %, direction: 1, rtcshutter: 4.3  [s]
21:27:58 Shutter 0: Real Pos: 11400, Target 20000, source: Shutter, start-pos: 4 %, direction: 1, rtcshutter: 5.3  [s]
21:27:59 Shutter 0: Real Pos: 13400, Target 20000, source: Shutter, start-pos: 4 %, direction: 1, rtcshutter: 6.3  [s]
21:28:00 Shutter 0: Real Pos: 15400, Target 20000, source: Shutter, start-pos: 4 %, direction: 1, rtcshutter: 7.3  [s]
21:28:01 Shutter 0: Real Pos: 17400, Target 20000, source: Shutter, start-pos: 4 %, direction: 1, rtcshutter: 8.3  [s]
21:28:02 Shutter 0: Real Pos: 19400, Target 20000, source: Shutter, start-pos: 4 %, direction: 1, rtcshutter: 9.3  [s]
21:28:03 MQT: stat/sonoff/SHUTTER1 = 100
21:28:03 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
21:28:03 MQT: stat/sonoff/POWER1 = OFF
21:28:03 MQT: tele/sonoff/RESULT = {"SHUTTER-1":{"position":100, "direction":0}}
**21:28:14 CMD: shutterclose
21:28:14 Start shutter in right direction -1
21:28:14 MQT: stat/sonoff/RESULT = {"POWER2":"ON"}
21:28:14 MQT: stat/sonoff/POWER2 = ON
21:28:14 MQT: stat/sonoff/RESULT = {"shutterclose1":0}**
21:28:15 Shutter 0: Real Pos: 18000, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 1.0  [s]
21:28:16 Shutter 0: Real Pos: 15900, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 2.0  [s]
21:28:17 Shutter 0: Real Pos: 13900, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 3.0  [s]
21:28:18 Shutter 0: Real Pos: 11900, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 4.0  [s]
21:28:19 Shutter 0: Real Pos: 9900, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 5.0  [s]
21:28:20 Shutter 0: Real Pos: 7900, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 6.0  [s]
21:28:21 Shutter 0: Real Pos: 5900, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 7.0  [s]
21:28:22 Shutter 0: Real Pos: 3900, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 8.1  [s]
21:28:23 Shutter 0: Real Pos: 1900, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 9.1  [s]
21:28:24 MQT: stat/sonoff/SHUTTER1 = 0
21:28:24 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
21:28:24 MQT: stat/sonoff/POWER2 = OFF
21:28:24 MQT: tele/sonoff/RESULT = {"SHUTTER-1":{"position":0, "direction":0}}
21:28:32 CMD: setoption80
21:28:32 MQT: stat/sonoff/RESULT = {"SetOption80":"ON"}
21:28:37 CMD: setoption81
21:28:37 MQT: stat/sonoff/RESULT = {"SetOption81":"OFF"}
21:29:05 CMD: interlock
21:29:05 MQT: stat/sonoff/RESULT = {"Interlock":"OFF","Groups":"1,2"}

stefanbode, 2019-04-01:
Ok agree on the error. It looks like it uses the normal operating mode. This might be an issue of the change. I will test this tomorrow morning and upload an update. Should’ve be no big deal.

stefanbode, 2019-04-02:
indeed the selectio of the shuttermode is wrong:
00:00:00 Shutter 0 (Relay:1): Init. Pos: 800 [4 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 20000, Opentime 10.0 [s], Closetime 10.0 [s], CoedffCalc: c0: 0, c1 200, c2: 200, c3: 0, c4: 0, binmask 3, is inverted 0, shuttermode 0
it is 0 and it should be 1. For detection I look into the Settings.interlock[0] if there is a interlock defined. Normally this should be 0

commit 0460a08e (2019-04-02): fixed #137

Due to changes in interlock the right operation mode was not detected correctly. Fixed. As soon as one of the shutterrelay is in interlock configuration is normal or pulse.
closed 2019-04-02 by stefanbode