stefanbode/Sonoff-Tasmota #139: PCF_8574 support

mrmin24, 2019-04-03:
<<<<<<< HEAD
=======
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

>>>>>>> 8f585b7f5354b19aff4bc6d1f238df4588d5693f
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ X] Hardware used : 
Sonoff basic with PCF-8574

- [ ] You have tried latest release or development binaries? :
Latest release with Core_6.5

**(Please, remember to close the issue when the problem has been addressed)**

Hi,

I have not upgraded in a loooong time. I now upgraded to latest release and core and cannot seem to find the #define USE_PCF8574 anywhere. If I do add it, then all sorts of other errors pop up. 

Am I doing something wrong or am I missing something?

Thanks

stefanbode, 2019-04-04:
Oh yes, because it was excluded during compilation I missed all the required changes. Uploaded a new tested version with all the changes required to the new method of web integration

closed 2019-04-04 by stefanbode

mrmin24, 2019-04-04:
Thanks a lot. Seems to be working. 

Just had to add <code>#define USE_PCF8574</code> in my_config.h

commit 5c39211e (2019-04-04): fixed #139

PCF8574 fixed + MQ Sensor fixed, Chirp ixed