stefanbode/Sonoff-Tasmota #144: Blinds stops if while moving forward I move slider back

Snake1980, 2019-05-12:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Hardware used : Sonoff Dual R2 
- [ ] If a pre-compiled release or development binary was used, which one? :
- [x] You have tried latest release or development binaries? : 
- [x] Provide the output of command ``status 0`` :
```
14:15:34 MQT: stat/sonoff-cletto/STATUS = {"Status":{"Module":39,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff-cletto","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
14:15:34 MQT: stat/sonoff-cletto/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T04:28:46","StartupUTC":"2019-05-12T08:46:48","Sleep":50,"CfgHolder":4617,"BootCount":16,"SaveCount":178,"SaveAddress":"F9000"}}
14:15:34 MQT: stat/sonoff-cletto/STATUS2 = {"StatusFWR":{"Version":"6.5.0.3 stb-1.3(sonoff)","BuildDateTime":"2019-04-07T22:54:23","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
14:15:34 MQT: stat/sonoff-cletto/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MYWIFI",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C029","280500000100000000000000000000000000","C0000000"]}}
14:15:34 MQT: stat/sonoff-cletto/STATUS4 = {"StatusMEM":{"ProgramSize":512,"Free":488,"Heap":17968,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000410","0FDAE394","000783A0","23B617CE","00003BC0"]}}
14:15:34 MQT: stat/sonoff-cletto/STATUS5 = {"StatusNET":{"Hostname":"sonoff-cletto-2783","IPAddress":"192.168.1.40","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"1.1.1.1","Mac":"MYMACADDR","Webserver":2,"WifiConfig":4}}
14:15:34 MQT: stat/sonoff-cletto/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_976ADF","MqttUser":"MYMQTTUSER","MqttCount":3,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
14:15:34 MQT: stat/sonoff-cletto/STATUS7 = {"StatusTIM":{"UTC":"Sun May 12 13:15:34 2019","Local":"Sun May 12 14:15:34 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:12","Sunset":"20:20"}}
14:15:34 MQT: stat/sonoff-cletto/STATUS10 = {"StatusSNS":{"Time":"2019-05-12T14:15:34","SHUTTER-1":{"position":0, "direction":0}}}
14:15:34 MQT: stat/sonoff-cletto/STATUS11 = {"StatusSTS":{"Time":"2019-05-12T14:15:34","Uptime":"0T04:28:46","Vcc":3.178,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"MYWIFI","BSSId":"10:13:31:BC:5D:59","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04","DeepSleep":0,"Heap":17904"}}}
```
I think I found a problem about slider to set percentage position.
Here is an example of my problem...
I'm starting with blind completely closed.
Moving slider on 50% position it correctly starts moving...

```
14:27:35 Start shutter in right direction 1
14:27:35 MQT: stat/sonoff-cletto/RESULT = {"POWER1":"ON"}
14:27:35 MQT: stat/sonoff-cletto/POWER1 = ON (salvato)
14:27:35 MQT: stat/sonoff-cletto/RESULT = {"shutterposition1":50}
14:27:36 Shutter 0: Real Pos: 2300, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 1.1  [s]
14:27:37 Shutter 0: Real Pos: 4300, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 2.1  [s]
14:27:38 Shutter 0: Real Pos: 6300, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 3.1  [s]
14:27:39 Shutter 0: Real Pos: 8300, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 4.2  [s]
14:27:40 Shutter 0: Real Pos: 10300, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 5.2  [s]
14:27:41 Shutter 0: Real Pos: 12300, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 6.2  [s]
```

...but...if while opening I move slider back to (for example) 10% it stops. (I hear relay sound two times  and it stops) 

```
14:27:41 Start shutter in right direction -1
14:27:41 MQT: stat/sonoff-cletto/RESULT = {"POWER1":"OFF"}
14:27:41 MQT: stat/sonoff-cletto/POWER1 = OFF (salvato)
14:27:41 MQT: stat/sonoff-cletto/RESULT = {"POWER2":"ON"}
14:27:41 MQT: stat/sonoff-cletto/POWER2 = ON (salvato)
14:27:41 MQT: stat/sonoff-cletto/RESULT = {"shutterposition1":10}
14:27:41 MQT: stat/sonoff-cletto/SHUTTER1 = 28 (salvato)
14:27:41 MQT: stat/sonoff-cletto/RESULT = {"POWER2":"OFF"}
14:27:41 MQT: stat/sonoff-cletto/POWER2 = OFF (salvato)
14:27:41 MQT: tele/sonoff-cletto/RESULT = {"SHUTTER-1":{"position":28, "direction":0}}
14:27:43 MQT: stat/sonoff-cletto/RESULT = {"shutterposition1":28}
```

I think last `stat/sonoff-cletto/RESULT = {"POWER2":"OFF"}` is wrong.
I don't know if I can fix this by some configuration parameter or if it's a bug

Without using slider but just buttons, it works fine...

```
14:35:14 MQT: stat/sonoff-cletto/RESULT = {"POWER1":"ON"}
14:35:14 MQT: stat/sonoff-cletto/POWER1 = ON (salvato)
14:35:14 Shutter 0: Real Pos: 200, Target 40000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 0.1  [s]
14:35:15 Shutter 0: Real Pos: 2200, Target 40000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 1.1  [s]
14:35:16 Shutter 0: Real Pos: 4200, Target 40000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 2.1  [s]
14:35:17 Shutter 0: Real Pos: 6200, Target 40000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 3.1  [s]
14:35:18 Shutter 0: Real Pos: 8200, Target 40000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 4.1  [s]
14:35:19 Shutter 0: Real Pos: 10200, Target 40000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 5.1  [s]
14:35:20 MQT: stat/sonoff-cletto/RESULT = {"POWER1":"OFF"}
14:35:20 MQT: stat/sonoff-cletto/POWER1 = OFF (salvato)
14:35:21 MQT: stat/sonoff-cletto/RESULT = {"POWER2":"ON"}
14:35:21 MQT: stat/sonoff-cletto/POWER2 = ON (salvato)
14:35:21 Shutter 0: Real Pos: 11600, Target 0, source: Shutter, start-pos: 0 %, direction: -1, rtcshutter: 0.2  [s]
14:35:22 Shutter 0: Real Pos: 9600, Target 0, source: Shutter, start-pos: 0 %, direction: -1, rtcshutter: 1.1  [s]
```

I made 2 videos about this

**SLIDER KO**
https://www.youtube.com/watch?v=QinVLlk7qmg

**BUTTONS OK**
https://www.youtube.com/watch?v=VxIdwPuKM2s

stefanbode, 2019-05-12:
Totally agree to your finding and thanks for the log files. I assume you are use the standard mode this is most likely from the logs. I have same configuration here, therefore easy to reproduce. If it behaves like that this is for sure a bug..

stefanbode, 2019-05-12:
The second power2 off is the mistake. The position is correct with 28 but it should not stop until 10. should be easy fixable

stefanbode, 2019-05-13:
There were some changes in the webinterface and the INTERLOCK behavior. Therefore the detection, if someone hits a button or play on the slider in the webinterface was not working anymore. Please feel free o check and give feedback

Snake1980, 2019-05-13:
Thanks! Now it works.

```
18:28:18 MQT: stat/sonoff-cletto/RESULT = {"POWER1":"ON"}
18:28:18 MQT: stat/sonoff-cletto/POWER1 = ON (salvato)
18:28:18 MQT: stat/sonoff-cletto/RESULT = {"shutterposition1":56}
18:28:18 Shutter 0: Real Pos: 1100, Target 22400, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 0.6  [s]
18:28:19 Shutter 0: Real Pos: 3100, Target 22400, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 1.6  [s]
18:28:20 Shutter 0: Real Pos: 5100, Target 22400, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 2.5  [s]
18:28:21 Shutter 0: Real Pos: 7100, Target 22400, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 3.5  [s]
18:28:22 Shutter 0: Real Pos: 9100, Target 22400, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 4.5  [s]
18:28:23 Shutter 0: Real Pos: 11100, Target 22400, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 5.5  [s]
18:28:24 Start shutter in direction -1
18:28:24 MQT: stat/sonoff-cletto/RESULT = {"POWER1":"OFF"}
18:28:24 MQT: stat/sonoff-cletto/POWER1 = OFF (salvato)
18:28:24 MQT: stat/sonoff-cletto/RESULT = {"POWER2":"ON"}
18:28:24 MQT: stat/sonoff-cletto/POWER2 = ON (salvato)
18:28:24 MQT: stat/sonoff-cletto/RESULT = {"shutterposition1":9}
18:28:25 Shutter 0: Real Pos: 10500, Target 3600, source: Shutter, start-pos: 0 %, direction: -1, rtcshutter: 1.0  [s]
18:28:26 Shutter 0: Real Pos: 8500, Target 3600, source: Shutter, start-pos: 0 %, direction: -1, rtcshutter: 2.0  [s]
18:28:27 Shutter 0: Real Pos: 6500, Target 3600, source: Shutter, start-pos: 0 %, direction: -1, rtcshutter: 3.0  [s]
18:28:28 Shutter 0: Real Pos: 4500, Target 3600, source: Shutter, start-pos: 0 %, direction: -1, rtcshutter: 4.0  [s]
18:28:28 MQT: stat/sonoff-cletto/SHUTTER1 = 9 (salvato)
18:28:28 MQT: stat/sonoff-cletto/RESULT = {"POWER2":"OFF"}
18:28:28 MQT: stat/sonoff-cletto/POWER2 = OFF (salvato)
18:28:28 MQT: tele/sonoff-cletto/RESULT = {"SHUTTER-1":{"position":9, "direction":0}}
18:28:28 MQT: stat/sonoff-cletto/RESULT = {"shutterposition1":9}
```

Just a question...
Do you think is correct to add a little delay (maybe about 0.3s) between POWERx OFF  and POWERy ON on switch direction using slider ?
To understand why I made two videos...

**Switch with buttons** (native)
https://www.youtube.com/watch?v=plhQcKpSMXk

**Switch with slider** (after your fix)
https://www.youtube.com/watch?v=npMF2gMbI4Y

stefanbode, 2019-05-13:
This is more tricky than you might expect. There is already a delay implemented with 0.5sec default value. You will find the variable under: MOTOR_STOP_TIME

This will not work if you hit the relay button directly in the webinterface. In this case, the INTERLOCK is switching OFF the first relay and it is very hard to get a SLEEP after this. Maybe I can have a look tomorrow afternoon, but I do not expect an easy change. With the slider, I control the relays. This makes things much easier. The INTERLOCK must be defined but as long as you use the slider or SHUTTERPOSITION the rescue mechanism from INTERLOCK should have no reason to dig in.

Snake1980, 2019-05-13:
Thank you for your time Stefan.
But in the "switch buttons" video I used buttons on web interface...and that delay is ok.
My question about delay was related to the slider

Snake1980, 2019-05-18:
Hi Stefan, any news about that delay ?

stefanbode, 2019-05-18:
Not yet. The slider should work anyway because I’m this situation I control the relay. Will investigate tomorrow afternoon

Snake1980, 2019-05-19:
Than you ;)

stefanbode, 2019-05-21:
New vesion is now aware of a fine motor_stop in any situation. Default time is 0.5 sec. Can be changed in code. Search for: MOTOR_STOP_TIME

closed 2019-05-21 by stefanbode

Snake1980, 2019-05-21:
Thank you Stefan!
I'll try later :)

Snake1980, 2019-05-21:
It works great!

https://youtu.be/glJyaC_wOpg