stefanbode/Sonoff-Tasmota #146: Sonoff 4CH Pro R2 Shutterrelay

tonyado, 2019-06-01:
<<<<<<< HEAD
=======
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

>>>>>>> 8f585b7f5354b19aff4bc6d1f238df4588d5693f
**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : 
<<<<<<< HEAD
=======
- [ ] _Development/Compiler/Upload tools used_ :
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ ] _You have tried latest release or development binaries?_ :
>>>>>>> 8f585b7f5354b19aff4bc6d1f238df4588d5693f
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2019-06-01:
From the log and from the setting it looks ok. Can you fully open and close manually or do you hit the mechanical end stops?

stefanbode, 2019-06-01:
Sorry first comment is wrong. Can you increase the loglevel to get a clear idea why the relay stopped. Looks like something external is triggering the relay. I agree that 37000 is lower than 60000 and the shutter should continue

stefanbode, 2019-06-01:
I see you have shutter1 and shutter2 defined. If so please double check that shutterrelay1 is 1 and shutterrelay2 is 3. If this is wrong maybe a side effect

tonyado, 2019-06-01:
I have tried many time, I have configured 2 relays, and for both the timer stops at 17 seconds, how can I do for help you? I can send a screenshot?

tonyado, 2019-06-01:
12:45:55 MQT: stat/BedroomSonoff/RESULT = {"shutterposition2":100}
12:45:56 Shutter 0: Real Pos: 5500, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 2.7  [s]
12:45:56 Shutter 1: Real Pos: 1400, Target 80000, source: , start-pos: 0 %, direction: 1, rtcshutter: 0.7  [s]
12:45:57 Shutter 0: Real Pos: 7600, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 3.7  [s]
12:45:57 Shutter 1: Real Pos: 3500, Target 80000, source: , start-pos: 0 %, direction: 1, rtcshutter: 1.8  [s]
12:45:58 Shutter 0: Real Pos: 9600, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 4.7  [s]
12:45:58 Shutter 1: Real Pos: 5500, Target 80000, source: , start-pos: 0 %, direction: 1, rtcshutter: 2.7  [s]
12:45:59 Shutter 0: Real Pos: 11600, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 5.7  [s]
12:45:59 Shutter 1: Real Pos: 7500, Target 80000, source: , start-pos: 0 %, direction: 1, rtcshutter: 3.7  [s]
12:46:00 Shutter 0: Real Pos: 13700, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 6.8  [s]
12:46:00 Shutter 1: Real Pos: 9600, Target 80000, source: , start-pos: 0 %, direction: 1, rtcshutter: 4.8  [s]
12:46:01 Shutter 0: Real Pos: 15600, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 7.7  [s]
12:46:01 Shutter 1: Real Pos: 11500, Target 80000, source: , start-pos: 0 %, direction: 1, rtcshutter: 5.7  [s]
12:46:02 Shutter 0: Real Pos: 17600, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 8.8  [s]
12:46:02 Shutter 1: Real Pos: 13500, Target 80000, source: , start-pos: 0 %, direction: 1, rtcshutter: 6.7  [s]
12:46:03 Shutter 0: Real Pos: 19500, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 9.7  [s]
12:46:03 Shutter 1: Real Pos: 15400, Target 80000, source: , start-pos: 0 %, direction: 1, rtcshutter: 7.7  [s]
12:46:04 Shutter 0: Real Pos: 21500, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 10.7  [s]
12:46:04 Shutter 1: Real Pos: 17400, Target 80000, source: , start-pos: 0 %, direction: 1, rtcshutter: 8.7  [s]
12:46:05 Shutter 0: Real Pos: 23600, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 11.8  [s]
12:46:05 Shutter 1: Real Pos: 19500, Target 80000, source: , start-pos: 0 %, direction: 1, rtcshutter: 9.8  [s]
12:46:06 Shutter 0: Real Pos: 25600, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 12.8  [s]
12:46:06 Shutter 1: Real Pos: 21500, Target 80000, source: , start-pos: 0 %, direction: 1, rtcshutter: 10.8  [s]
12:46:07 Shutter 0: Real Pos: 27700, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 13.8  [s]
12:46:07 Shutter 1: Real Pos: 23600, Target 80000, source: , start-pos: 0 %, direction: 1, rtcshutter: 11.8  [s]
12:46:08 Shutter 0: Real Pos: 29600, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 14.7  [s]
12:46:08 Shutter 1: Real Pos: 25500, Target 80000, source: , start-pos: 0 %, direction: 1, rtcshutter: 12.8  [s]
12:46:09 Shutter 0: Real Pos: 31600, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 15.8  [s]
12:46:09 Shutter 1: Real Pos: 27500, Target 80000, source: , start-pos: 0 %, direction: 1, rtcshutter: 13.8  [s]
12:46:10 Shutter 0: Real Pos: 33600, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 16.8  [s]
12:46:10 Shutter 1: Real Pos: 29500, Target 80000, source: , start-pos: 0 %, direction: 1, rtcshutter: 14.7  [s]
12:46:10 MQT: stat/BedroomSonoff/RESULT = {"POWER1":"OFF"}
12:46:10 MQT: stat/BedroomSonoff/POWER1 = OFF
12:46:10 MQT: stat/BedroomSonoff/SHUTTER1 = 43
12:46:10 MQT: tele/BedroomSonoff/RESULT = {"SHUTTER-1":{"position":43, "direction":0}}
12:46:11 Shutter 1: Real Pos: 31500, Target 80000, source: Button, start-pos: 0 %, direction: 1, rtcshutter: 15.8  [s]
12:46:12 Shutter 1: Real Pos: 33500, Target 80000, source: Button, start-pos: 0 %, direction: 1, rtcshutter: 16.8  [s]
12:46:12 MQT: stat/BedroomSonoff/RESULT = {"POWER3":"OFF"}
12:46:12 MQT: stat/BedroomSonoff/POWER3 = OFF
12:46:12 MQT: stat/BedroomSonoff/SHUTTER2 = 42
12:46:12 MQT: tele/BedroomSonoff/RESULT = {"SHUTTER-2":{"position":42, "direction":0}}

tonyado, 2019-06-01:
My configuration is:

SETOPTION80 1
SETOPTION81 1
INTERLOCK1 1,2 3,4
SHUTTERRELAY1 1
SHUTTERRELAY2 3
SHUTTEROPENDURATION XX
SHUTTERCLOSEDURATION XX
SHUTTEROPENDURATION2 XX
SHUTTERCLOSEDURATION2 XX
SHUTTERINVERT1 1

stefanbode, 2019-06-01:
For the loglevel please use weblog 4 or seriallog 5. This might add after the 17 seconds the required information.  Just to make sure it is INTERLOCK 1,2 3,4 and not INTERLOCK1 just in case.

tonyado, 2019-06-01:
Sorry, correct information is INTERLOCK 1 1,2 3,4, INTERLOCK1 it's a mistake.

This is log after weblog 5

16:57:28 CMD: weblog 5
16:57:28 MQT: stat/BedroomSonoff/RESULT = {"WebLog":5}
16:57:28 CFG: Salvato nella flash in F7, Conteggio 202, Bytes 3824
16:57:29 WIF: Controllo connessione...
16:57:29 WIF: Connesso
16:57:39 HTP: Menu Principale
16:57:41 SRC: WebGui from 192.168.1.147
16:57:41 RSL: Topic Ricevuto /shutterposition1, Dimensione Dati 1, Dati 0
16:57:41 RSL: Gruppo 0, Indice 1, Comando SHUTTERPOSITION, Dati 0
16:57:41 shutterposition in: payload 0, index 1, source 7
16:57:41 lastsource 7:, realpos 60000, target 0, payload 0
16:57:41 Start shutter in direction -1
16:57:41 SRC: 
16:57:41 Switched relay: 2 by 
16:57:41 MQT: stat/BedroomSonoff/RESULT = {"POWER2":"ON"}
16:57:41 MQT: stat/BedroomSonoff/POWER2 = ON
16:57:41 MQT: stat/BedroomSonoff/RESULT = {"shutterposition1":100}
16:57:42 Shutter 0: Real Pos: 58900, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 0.6  [s]
16:57:42 CFG: Salvato nella flash in F6, Conteggio 203, Bytes 3824
16:57:43 Shutter 0: Real Pos: 56900, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 1.6  [s]
16:57:43 HTP: Console
16:57:44 Shutter 0: Real Pos: 54900, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 2.5  [s]
16:57:45 Shutter 0: Real Pos: 52900, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 3.5  [s]
16:57:46 Shutter 0: Real Pos: 50900, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 4.5  [s]
16:57:47 Shutter 0: Real Pos: 48900, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 5.5  [s]
16:57:48 Shutter 0: Real Pos: 46600, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 6.7  [s]
16:57:49 Shutter 0: Real Pos: 44900, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 7.5  [s]
16:57:49 WIF: Controllo connessione...
16:57:49 WIF: Connesso
16:57:50 Shutter 0: Real Pos: 42900, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 8.6  [s]
16:57:51 Shutter 0: Real Pos: 40900, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 9.6  [s]
16:57:52 Shutter 0: Real Pos: 38900, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 10.6  [s]
16:57:53 Shutter 0: Real Pos: 36900, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 11.6  [s]
16:57:54 Shutter 0: Real Pos: 34900, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 12.6  [s]
16:57:55 Shutter 0: Real Pos: 32900, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 13.6  [s]
16:57:56 Shutter 0: Real Pos: 30900, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 14.6  [s]
16:57:57 Shutter 0: Real Pos: 28900, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 15.6  [s]
16:57:58 Shutter 0: Real Pos: 26900, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 16.6  [s]
16:57:58 APP: Pulsante2 level 1-0
16:57:58 SRC: Button
16:57:58 Switched relay: 2 by Button
16:57:58 Shutter 0: Switch OFF motor. Target: 26000, source: Button, powerstate_local 0, switchedRelay 2, manual change 1
16:57:58 Shutter 0: Target: 26000, powerstatelocal 0
16:57:58 MQT: stat/BedroomSonoff/RESULT = {"POWER2":"OFF"}
16:57:58 MQT: stat/BedroomSonoff/POWER2 = OFF
16:57:58 Shutter 0: Real Pos. 25800, Stoppos: 43, relay: 1, direction -1, pulsetimer: 0, rtcshutter: 17.1 [s], operationtime 2
16:57:58 MQT: stat/BedroomSonoff/SHUTTER1 = 57
16:57:58 MQT: tele/BedroomSonoff/RESULT = {"SHUTTER-1":{"position":57, "direction":0}}
16:57:59 CFG: Salvato nella flash in F5, Conteggio 204, Bytes 3824
16:58:09 WIF: Controllo connessione...
16:58:09 WIF: Connesso

stefanbode, 2019-06-01:
I think I have it. Please check PULsETIMER1 and so on the should all be 0. It looks like there is a 17 second timer that switches off the relay.

stefanbode, 2019-06-01:
16:57:58 APP: Pulsante2 level 1-0
16:57:58 SRC: Button
16:57:58 Switched relay: 2 by Button

Here you see the root cause for the switching. If I translate correct into English it is the pulse timer messing with your relay

tonyado, 2019-06-01:
This is the log, I think that Pulsetime is disabled because I have the same problem for both shutter and I have another Sonoff 4CH Pro with the same problem. It's a mistery! 

19:58:30 CMD: pulsetime
19:58:30 SRC: WebConsole from 192.168.1.147
19:58:30 RSL: Topic Ricevuto /pulsetime, Dimensione Dati 0, Dati 
19:58:30 RSL: Gruppo 0, Indice 1, Comando PULSETIME, Dati 
19:58:30 MQT: stat/BedroomSonoff/RESULT = {"PulseTime1":"0 (Active 0)"}

stefanbode, 2019-06-01:
I assume you check also pulsetime2. Agree a mystery. Anyhow I can’t fix or change something because the OFF command comes from pulsetime. RESET 2 makes a hard reset and overwrite everything with 0. maybe this can solve it.

tonyado, 2019-06-02:
I maked RESET and I set configuration from begin.
Result it's the same:

06:45:26 Shutter 1 (Relay:3): Init. Pos: 20000 [100 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 20000, Opentime 10.0 [s], Closetime 10.0 [s], CoedffCalc: c0: 0, c1 200, c2: 200, c3: 0, c4: 0, binmask 15, is inverted 0, shuttermode 0
06:45:26 RSL: stat/sonoff4CHpro/RESULT = {"shuttercloseduration1":"30.0"}
06:45:26 CFG: Salvato nella flash in F5, Conteggio 23, Bytes 3824
06:45:39 WIF: Controllo connessione...
06:45:39 WIF: Connesso
06:45:59 WIF: Controllo connessione...
06:45:59 WIF: Connesso
06:46:19 WIF: Controllo connessione...
06:46:19 WIF: Connesso
06:46:26 CMD: SHUTTERCLOSEDURATION2 60
06:46:26 SRC: WebConsole from 192.168.1.147
06:46:26 RSL: Topic Ricevuto /SHUTTERCLOSEDURATION2, Dimensione Dati 2, Dati 60
06:46:26 RSL: Gruppo 0, Indice 2, Comando SHUTTERCLOSEDURATION, Dati 60
06:46:26 Shutter accuracy digits: 1
06:46:26 Shutter 0 (Relay:1): Init. Pos: 60000 [100 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 60000, Opentime 30.0 [s], Closetime 30.0 [s], CoedffCalc: c0: 0, c1 600, c2: 600, c3: 0, c4: 0, binmask 3, is inverted 0, shuttermode 0
06:46:26 Shutter 1 (Relay:3): Init. Pos: 20000 [100 %], Open Vel.: 100 Close Vel.: 16 , Max Way: 20000, Opentime 10.0 [s], Closetime 60.0 [s], CoedffCalc: c0: 0, c1 200, c2: 200, c3: 0, c4: 0, binmask 15, is inverted 0, shuttermode 0
06:46:26 RSL: stat/sonoff4CHpro/RESULT = {"shuttercloseduration2":"60.0"}
06:46:27 CFG: Salvato nella flash in F4, Conteggio 24, Bytes 3824
06:46:39 WIF: Controllo connessione...
06:46:39 WIF: Connesso
06:46:42 CMD: SHUTTEROPENDURATION2 60
06:46:42 SRC: WebConsole from 192.168.1.147
06:46:42 RSL: Topic Ricevuto /SHUTTEROPENDURATION2, Dimensione Dati 2, Dati 60
06:46:42 RSL: Gruppo 0, Indice 2, Comando SHUTTEROPENDURATION, Dati 60
06:46:42 Shutter accuracy digits: 1
06:46:42 Shutter 0 (Relay:1): Init. Pos: 60000 [100 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 60000, Opentime 30.0 [s], Closetime 30.0 [s], CoedffCalc: c0: 0, c1 600, c2: 600, c3: 0, c4: 0, binmask 3, is inverted 0, shuttermode 0
06:46:42 Shutter 1 (Relay:3): Init. Pos: 120000 [100 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 120000, Opentime 60.0 [s], Closetime 60.0 [s], CoedffCalc: c0: 0, c1 1200, c2: 1200, c3: 0, c4: 0, binmask 15, is inverted 0, shuttermode 0
06:46:42 RSL: stat/sonoff4CHpro/RESULT = {"shutteropenduration2":"60.0"}
06:46:43 CFG: Salvato nella flash in FB, Conteggio 25, Bytes 3824
06:46:59 WIF: Controllo connessione...
06:46:59 WIF: Connesso
06:47:14 RSL: tele/sonoff4CHpro/STATE = {"Time":"2019-06-02T06:47:14","Uptime":"0T00:04:59","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"TP-NET","BSSId":"C4:E9:84:33:FB:79","Channel":4,"RSSI":86,"LinkCount":1,"Downtime":"0T00:00:04","DeepSleep":0,"Heap":26720}}
06:47:14 RSL: tele/sonoff4CHpro/SENSOR = {"Time":"2019-06-02T06:47:14","SHUTTER-1":{"position":100, "direction":0},"SHUTTER-2":{"position":100, "direction":0}}
06:47:19 WIF: Controllo connessione...
06:47:19 WIF: Connesso
06:47:26 HTP: Menu Principale
06:47:30 SRC: WebGui from 192.168.1.147
06:47:30 RSL: Topic Ricevuto /shutterposition1, Dimensione Dati 1, Dati 0
06:47:30 RSL: Gruppo 0, Indice 1, Comando SHUTTERPOSITION, Dati 0
06:47:30 shutterposition in: payload 0, index 1, source 7
06:47:30 lastsource 7:, realpos 60000, target 0, payload 0
06:47:30 Start shutter in direction -1
06:47:30 SRC: 
06:47:30 Switched relay: 2 by 
06:47:30 RSL: stat/sonoff4CHpro/RESULT = {"POWER2":"ON"}
06:47:30 RSL: stat/sonoff4CHpro/POWER2 = ON
06:47:30 RSL: stat/sonoff4CHpro/RESULT = {"shutterposition1":0}
06:47:30 Shutter 0: Real Pos: 59700, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 0.2  [s]
06:47:30 CFG: Salvato nella flash in FA, Conteggio 26, Bytes 3824
06:47:31 HTP: Console
06:47:31 Shutter 0: Real Pos: 57400, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 1.3  [s]
06:47:32 Shutter 0: Real Pos: 55700, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 2.1  [s]
06:47:33 Shutter 0: Real Pos: 53700, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 3.1  [s]
06:47:34 Shutter 0: Real Pos: 51800, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 4.1  [s]
06:47:35 Shutter 0: Real Pos: 49800, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 5.1  [s]
06:47:36 Shutter 0: Real Pos: 47700, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 6.2  [s]
06:47:37 Shutter 0: Real Pos: 45700, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 7.2  [s]
06:47:38 Shutter 0: Real Pos: 43700, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 8.2  [s]
06:47:39 Shutter 0: Real Pos: 41700, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 9.2  [s]
06:47:39 WIF: Controllo connessione...
06:47:39 WIF: Connesso
06:47:40 Shutter 0: Real Pos: 39700, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 10.2  [s]
06:47:41 Shutter 0: Real Pos: 37700, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 11.2  [s]
06:47:42 Shutter 0: Real Pos: 35700, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 12.2  [s]
06:47:43 Shutter 0: Real Pos: 33700, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 13.2  [s]
06:47:44 Shutter 0: Real Pos: 31700, Target 0, source: , start-pos: 100 %, direction: -1, rtcshutter: 14.2  [s]
[... cut ...]
06:47:47 Shutter 0: Switch OFF motor. Target: 25900, source: Button, powerstate_local 0, switchedRelay 2, manual change 1
06:47:47 Shutter 0: Target: 25900, powerstatelocal 0
06:47:47 RSL: stat/sonoff4CHpro/RESULT = {"POWER2":"OFF"}
06:47:47 RSL: stat/sonoff4CHpro/POWER2 = OFF
06:47:47 Shutter 0: Real Pos. 25800, Stoppos: 43, relay: 1, direction -1, pulsetimer: 0, rtcshutter: 17.1 [s], operationtime 2
06:47:47 RSL: stat/sonoff4CHpro/SHUTTER1 = 43
06:47:47 RSL: tele/sonoff4CHpro/RESULT = {"SHUTTER-1":{"position":43, "direction":0}}

tonyado, 2019-06-02:
I found it! Now all works perfect!
When I changed switch S6 from Interlock mode to Self-locking/ inching, in order to create through software 2 groups of relays, I don't have made any other changes at switch K5 and K6, then after 16 seconds all relays switched off because original configuration of K5 is 0000, then inching mode is ON for all relays. Sorry for this, it's not a issue, you made a fantastic work and your support is excellent! Thanks for your time Antonio

[image]

tonyado, 2019-06-02:
> # Sonoff 4CH Pro R2
> Sonoff 4CH Pro R2 Shutterrelay does not reach position, after 17 seconds relays stops.
> 
> I'm sorry, this is my first issue opened, if you would to excuse me if I'm wrong.
> I'm using last firmware compiled whit Atom and I have Sonoff 4CH Pro and Sonoff DUAL R2, I would use it for rollershutter.
> For Sonoff DUAL R2 I don't have any problem, but for Sonoff 4CH Pro R2, if I send command for shutter, whit cmnd/Sonoff/Shutterposition XX, or I move the level from web interface, the relay stop works after 17 seconds, and the shutter does not reach set position if the time for reach it is major then 17 seconds. I have set Shutteropenduration and Shuttercloseduration to 40 seconds
> 
> * [x]  _Searched the problem in issues and in the wiki_
> * [x]  _Hardware used_ :
>   Sonoff 4CH Pro R2
>   =======
> * [x]  _Development/Compiler/Upload tools used_ : Atom
> * [ ]  _If a pre-compiled release or development binary was used, which one?_ :
> * [x]  _You have tried latest release or development binaries?_ : yes
> * [x]  _Provide the output of command_`status 0` :
> 
> ```
> 08:03:22 CMD: status 0
> 08:03:22 MQT: stat/BedroomSonoff/STATUS = {"Status":{"Module":23,"FriendlyName":["BedroomSonoff","Sonoff2","Sonoff3","Sonoff4"],"Topic":"BedroomSonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
> 08:03:22 MQT: stat/BedroomSonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:00:50","StartupUTC":"2019-06-01T07:02:32","Sleep":50,"CfgHolder":4617,"BootCount":13,"SaveCount":179,"SaveAddress":"F6000"}}
> 08:03:22 MQT: stat/BedroomSonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0.11 stb-1.5(sonoff)","BuildDateTime":"2019-05-31T19:52:03","Boot":7,"Core":"2_5_1","SDK":"2.2.2-dev(c0eb301)"}}
> 08:03:22 MQT: stat/BedroomSonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TP-NET",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","280500000100000000000000000000000000","C0000000"]}}
> 08:03:22 MQT: stat/BedroomSonoff/STATUS4 = {"StatusMEM":{"ProgramSize":574,"Free":428,"Heap":22880,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000410","0FDAE394","040783A0","23B617CC","01003BC0"]}}
> 08:03:22 MQT: stat/BedroomSonoff/STATUS5 = {"StatusNET":{"Hostname":"BedroomSonoff-2861","IPAddress":"192.168.1.222","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"1.1.1.1","Mac":"DC:4F:22:D9:0B:2D","Webserver":2,"WifiConfig":4}}
> 08:03:22 MQT: stat/BedroomSonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.83","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D90B2D","MqttUser":"openhabian","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
> 08:03:22 MQT: stat/BedroomSonoff/STATUS7 = {"StatusTIM":{"UTC":"Sat Jun 01 07:03:22 2019","Local":"Sat Jun 01 08:03:22 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"04:36","Sunset":"20:03"}}
> 08:03:22 MQT: stat/BedroomSonoff/STATUS10 = {"StatusSNS":{"Time":"2019-06-01T08:03:22","SHUTTER-1":{"position":97, "direction":0},"SHUTTER-2":{"position":0, "direction":0}}}
> 08:03:22 MQT: stat/BedroomSonoff/STATUS11 = {"StatusSTS":{"Time":"2019-06-01T08:03:22","Uptime":"0T00:00:50","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"TP-NET","BSSId":"C4:E9:84:33:FB:79","Channel":4,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04","DeepSleep":0,"Heap":21128}}}
> ```
> 
> **To Reproduce**
> _Steps to reproduce the behavior:_
> 
> Change position from 0 to 100
> 
> 08:08:54 Shutter 0: Real Pos: 1700, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 0.9 [s]
> 08:08:55 Shutter 0: Real Pos: 3900, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 2.0 [s]
> 08:08:56 Shutter 0: Real Pos: 5700, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 2.9 [s]
> 08:08:57 Shutter 0: Real Pos: 7700, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 3.9 [s]
> 08:08:58 Shutter 0: Real Pos: 9700, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 4.8 [s]
> 08:08:59 Shutter 0: Real Pos: 11700, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 5.8 [s]
> 08:09:00 Shutter 0: Real Pos: 13700, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 6.8 [s]
> 08:09:01 Shutter 0: Real Pos: 15700, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 7.8 [s]
> 08:09:02 Shutter 0: Real Pos: 17800, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 8.9 [s]
> 08:09:03 Shutter 0: Real Pos: 19700, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 9.9 [s]
> 08:09:04 Shutter 0: Real Pos: 21700, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 10.9 [s]
> 08:09:05 Shutter 0: Real Pos: 23700, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 11.8 [s]
> 08:09:06 Shutter 0: Real Pos: 25700, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 12.9 [s]
> 08:09:07 Shutter 0: Real Pos: 27700, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 13.9 [s]
[... cut ...]
> 08:09:10 Shutter 0: Real Pos: 33700, Target 60000, source: , start-pos: 0 %, direction: 1, rtcshutter: 16.9 [s]
> 08:09:10 MQT: stat/BedroomSonoff/RESULT = {"POWER1":"OFF"}
> 08:09:10 MQT: stat/BedroomSonoff/POWER1 = OFF
> 08:09:10 MQT: stat/BedroomSonoff/SHUTTER1 = 43
> 08:09:10 MQT: tele/BedroomSonoff/RESULT = {"SHUTTER-1":{"position":43, "direction":0}}
> 
> **Expected behavior**
> _A clear and concise description of what you expected to happen._
> 
> I would that the shutter reach given position
> 
> Thanks for your works
> Antonio

closed 2019-06-02 by tonyado