stefanbode/Sonoff-Tasmota #148: Strange behaviour with latest fw 6.5.0.11 stb-1.5 in Shutter mode

x23piracy, 2019-06-04:
Hi,

first i was using regular tasmota but i was interested in the step stuff. both regular and stefanbode tasmote worked normally without shutter mode, now since i enabled:

- SETOPTION80 1
- SETOPTION81 1
- INTERLOCK 1
- SHUTTEROPENDURATION 27
- SHUTTERCLOSEDURATION 25

when i now try to fully close (webif or console "shutterposition") it starts running and on its way it stops 3 times (relay off/on "click,clack") and fully closes. 

When i now try to open with the shutter (webif) lets say half it only runs short and opens for example 10-15%, after that i cannot open the shutter more with the webif or with the physical switches, the only thing thats working is to use my iphone with home connected via homebridge issuing a mqtt command to open, it will only open in steps it starts opening and after some rolling it turns off, i repeat this 4-5 times until its fully opened again.

I really don't understand this afai can see i understand the options but this is really strange, can someone help me please? I was thinking of disabling shutter mode and just use full open and close with only pulsetime with propper values but i really like the steps thing and maybe i am stupid or something is messed up.

Help please.

Regards X23

x23piracy, 2019-06-05:
@stefanbode is there a way to get in touch with you outside of github?
Wir sind glaube ich beide Muttersprachler :)

Regards X23

stefanbode, 2019-06-05:
Ja, aber ich sitze gleich wieder im Flieger. Ich sehe was von Pulse…shutter. Ggf wiki mal lesen

x23piracy, 2019-06-05:
Hi,

das Wiki habe ich was das Thema angeht raus und runter gelesen, ich habe hier einfach einen Zustand mit dem Shelly2 den ich nicht nachvollziehen kann und dachte ich suche deine direkte Hilfe :)

Das kann gerne warten, guten Flug. Vielleicht hast du ja die Tage mal einen Moment dafür.

Gruß Jens

stefanbode, 2019-06-05:
Hast du denn einen shutter, der nur den Pulse bekommt oder ist das Ein Standard Rollo, welcher Strom bekommt, wenn er sich bewegen soll. Schau dir auch mal das log in der webconsole an. Ggf hier mal senden. Weblog 5 zeigt die auch an wer oder was die relay schaltet. Beim INTERLOCK fehlt noch ein INTERLOCK 1,2
Das ist notwendig bei den normalen relay, dass du nicht rauf und runter gleichzeitig auslösen kannst.  Das Weblog gerne auch hier posten mit ein paar Kommentaren.

stefanbode, 2019-06-05:
Pulsetime muss 0 sein auf beide. Relay, wenn du ein normales Rollo hast. Aktuell halte ich den fehlenden interlock für das Problem

x23piracy, 2019-06-05:
Hi,

ich kann heute Abend wenn ich zuhause bin das ganze nochmal probieren.
Ich hab doch oben beschrieben das ich zuerst die normale Tasmota benutzt hab. Die hat ja keinen Shutter mode da habe ich dann mit Interlock und Pulsetime gearbeitet, das ging perfekt und dabei gehts ja nur um ganz rauf und ganz runter.

Dann habe ich deine FW probiert (selbst compiliert), hier habe ich dann den shutter mode aktiviert (siehe Eingangspost) ich habe alles gemacht was im Wiki steht aber das Ding reagiert leider völlig anders.

Ich weiß nicht was ich nun tun soll, Flash löschen und einmal neu flashen mit default Settings, vielleicht habe ich ja was verstellt.

Mein Rollo Motor braucht dauerhaft Strom das ist kein Impuls ding. Am Shelly sind auch noch die ganz normalen Rolloschalter mit angeschlossen.

Gruß Jens

x23piracy, 2019-06-05:
Hi,

so ist es schon die ganze Zeit bei mir eingestellt:

18:09:01 CMD: pulsetime1
18:09:01 MQT: stat/shelly2-134F5C/RESULT = {"PulseTime1":"0 (Active 0)"}
18:09:04 CMD: pulsetime2
18:09:04 MQT: stat/shelly2-134F5C/RESULT = {"PulseTime2":"0 (Active 0)"}
18:09:06 CMD: interlock
18:09:06 MQT: stat/shelly2-134F5C/RESULT = {"Interlock":"ON","Groups":"1,2"}
18:09:19 CMD: shutteropenduration
18:09:19 MQT: stat/shelly2-134F5C/RESULT = {"shutteropenduration1":"27.0"}
18:09:23 CMD: shuttercloseduration
18:09:23 MQT: stat/shelly2-134F5C/RESULT = {"shuttercloseduration1":"25.0"}
18:09:56 CMD: setoption80
18:09:56 MQT: stat/shelly2-134F5C/RESULT = {"SetOption80":"ON"}
18:10:04 CMD: setoption81
18:10:04 MQT: stat/shelly2-134F5C/RESULT = {"SetOption81":"ON"}

Ich hab ein Rollo das Dauerstrom zum laufen braucht, wie gesagt ich hab physikalische Schalter die am Shelly angeschlossen sind, dass klappt auch. Im Webif oder per Schalter kann ich rauf und runterfahren und wenn die Teleperiod durch ist gibts auch Feedback (300 ist halt echt lang, ich habs Testweise auf 10 gehabt). Bediene ich übrigens den Shelly per MQTT via Homebridge habe ich das Gefühl der Shutter Status (Position) wird nie aktualisiert. Benutze ich den physikalischen Schalter aktualisiert sich nach der teleperiod auch der Schieber im Webif.

Gruß Jens

x23piracy, 2019-06-05:
Hi,

plötzlich funktioniert der Spaß! Und jetzt das scheinbar entscheidende, die teleperiod war auf 10, die habe ich mal wieder auf 300 gestellt und die probleme sind weg.

Kann es sein das ein teleperiod durchlauf dazu führt? aber wie kann ne sw status info zum kurzfristigen ausschalten und direktem wiedereinschalten bei einem relay führen? 

Rechnerisch passen in Anbetracht meiner vollen Laufzeiten (27 Sekunden rauf und 25 Sekunden runter) 3 Unterbrechungen doch ziemlich gut in Bild wenn das Timing passt.

Hmm

Gruß Jens

stefanbode, 2019-06-06:
Sehr unwahrscheinlich. Noch nie gesehen. Im log sieht man relativ genau was er wann und warum macht. Setz teleperiod wieder auf 10 und es wird weiter gehen. Schau dir mal an wie die Position an MQTT gemeldet wird. Nach eine stop wird immer die aktuelle Position gemeldet.

closed 2019-06-26 by stefanbode