stefanbode/Sonoff-Tasmota #149: Tuya-based Switch bricked

Tim3100, 2019-06-09:
<<<<<<< HEAD
=======
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

>>>>>>> 8f585b7f5354b19aff4bc6d1f238df4588d5693f
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Hardware used : Jinvoo Smart Switch 2
- [x] If a pre-compiled release or development binary was used, which one? :https://github.com/stefanbode/Sonoff-Tasmota/tree/master/bin/firmware_sonoff.bin
- [x] You have tried latest release or development binaries? : latest release
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

I'm currently installing electric shutters and therefore tried out a Jinvoo Smart Switch with this cool Tasmota fork. The switch is Tuya based (https://docs.tuya.com/en/hardware/WiFi-module/wifi-e3s-module.html), I could flash standard tasmota with tuya-convert and all seemed to work fine. I could also flash the minimal version of this stefanbode fork, but the full version seem to have bricked the device: when plugged in, the LED flashes quickly, the switch can no longer be accessed in the browser, it does not connect to the wifi.

Is this issue known already, can it somehow be fixed?

(I don't worry about the switch, I am well aware this is experimental)

stefanbode, 2019-06-10:
No, this version is 99% the same like the original one. I try to change as minimal as possible. Therefore the behavior is unusual but it was already reported one time, but not reproduceable. If connecting serial you should be able to flash in any case. Check if there is a WiFi opened by the device.

Tim3100, 2019-06-10:
Thanks for the quick reply, it solved the issue for me! Actually, a sonoff-xxxx WiFi was opened, I could connect, enter the Wifi credentials and configure the device. I'm new to Tasmota and didn't expect the credentials are overwritten -- I'm sorry. Thanks again, I'm now looking forward to awesome shutter control :)

closed 2019-06-10 by Tim3100

stefanbode, 2019-06-10:
The issue is that my config definition is a bit larger than the original one and need some more bytes. The new algorithm for detecting errors reject a new flashed device because of the wrong checksum and bring it back to default. Since now I haven’t found a suitable way for me to overcome this new check.