stefanbode/Sonoff-Tasmota #150: Opening Closing as a percentage of tim

matteo2019, 2019-06-11:
I thank all those who answer me in advance. I have a dual R2 sonoff with stefanbode tasmota firmware. My problem is to open / close the shutter at a certain time with a certain percentage.
Example: I set the 7 o'clock timer and the shutter is raised with the seconds established by the shutteropenduration function, I open it 100%. At 8 o'clock I want to get the blind down by 20% of the total opening, always setting the timer.
I await your answers, please explain in detail.

snakuzzo, 2019-06-26:
Tell me if I understand..
Do you want to open your blind at 80% at 8:00 AM...right ?
Are you using an hub (e.g. homeassistant) ?

matteo2019, 2019-06-27:
quite right!  For now I'm not using home assistant.  It is a device that will work by itself automatically with timers and preset percentages.  Often he will have to work without an internet connection so I need to integrate a battery.  If I do not insert the battery the timers will not work in the absence of internet.  I also need assistance to insert the battery

Ottieni Outlook per Android<https://aka.ms/ghei36>

snakuzzo, 2019-06-27:
You can use timers commands... 

https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#timers

matteo2019, 2019-06-27:
I describe what I have to do: I have to use sonoff dual r2 to raise and lower the blinds with the timer completely or in percentage at set times.  often in my area there is no internet and the sonoff loses the time memory.  I need a battery to be integrated into the sonoff.

Ottieni Outlook per Android<https://aka.ms/ghei36>

https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#timers

stefanbode, 2019-06-27:
@matteo2019  there might be something wrong in the argumentation. Let me explain. You want to operate the shutter. Therefore you must have electricity. The time of the sonoff will be ok, unless it reboots. Maybe there will be some seconds delay, but as long as the internet came back from time to time everything should be ok. The battery is definitely not required unless also the electricity shuts down several times a day. In this case, the shutter might be your smallest problem. Please take a look into TASMOTA wiki how to configure the device that it does NOT reboot, if the internet or MQTT connection is lost.

matteo2019, 2019-06-27:
even the tension go away, right!

I need to insert a buffer battery.  In addition to this problem, I want to activate timers that raise or lower the shutter in percentage!  Not in full!

Ottieni Outlook per Android<https://aka.ms/ghei36>

matteo2019, 2019-06-28:
please tell me how to do it

Ottieni Outlook per Android<https://aka.ms/ghei36>

matteo2019, 2019-07-05:
Let me know if what I wrote is not clear! I await answers, thanks

stefanbode, 2019-07-06:
wificonfig 2
This might help to avoid reboot if connection is lost. See wiki of Tasmota

matteo2019, 2019-07-08:
Your comment is not clear. they told me on github that I can use a wemos d1 mini coupled with an rtc3231 module. I have to route the sonoff to the wemos to get the correct time. I do not have an internet connection where I will connect these devices. When the power grid goes away the sonoff loses the conception of time

matteo2019, 2019-07-08:
please i need help

matteo2019, 2019-07-08:
I have to use the sonoff devices with your 'stefanbode' firmware where I don't have an internet connection. will work with timers. When the power supply disappears the sonoff loses time and all the timers do not work.

matteo2019, 2019-07-17:
stefanbode can I get help from you?

Timer1 {"Arm":1,"Time":"07:30","Window":0,"Days":"-MTWTF-","Repeat":1,"Output":1,"Action":1}

stefanbode, 2019-07-17:
hi @matteo2019 , if the sonoff loses power there is no workaround without internet to get the time correct after reboot. Maybe you just buffer the ESP with a battery. Looks the easiest thing. Otherwise the usecase will not work.

matteo2019, 2019-07-17:
this problem has been solved with rtc 3231. Now I have to open / close the shutters with the timers.  A percentage.  For example at 6.00 am the shutter must open 10%

stefanbode, 2019-07-17:
if this is solved with rtc3231 then where is your problem?

matteo2019, 2019-07-17:
rtc serves to keep track of time on the sonoff.  Now I need to know this:

this problem has been solved with rtc 3231. Now I have to open / close the shutters with the timers. A percentage. For example at 6.00 am the shutter must open 10%

matteo2019, 2019-07-17:
@stefanbode I promise to speak English.  I'm italian.  You managed to understand what I wrote.  I hope for your support

closed 2019-07-20 by matteo2019