stefanbode/Sonoff-Tasmota #154: Can't get pcf_8574A to be recognized

MadDoct, 2019-06-16:
<<<<<<< HEAD
=======
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

>>>>>>> 8f585b7f5354b19aff4bc6d1f238df4588d5693f
**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [x ] _Hardware used_ : nodemcu v3 with pcf8574A
<<<<<<< HEAD
=======
- [ ] _Development/Compiler/Upload tools used_ :
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ ] _You have tried latest release or development binaries?_ :
>>>>>>> 8f585b7f5354b19aff4bc6d1f238df4588d5693f
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
_Steps to reproduce the behavior:_
Trying to use pcf8574A that uses the address 0x27

**Expected behavior**
_A clear and concise description of what you expected to happen._
Recognize the pcf and configure it

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._
I changed the address in the code to 0x20 (which I know means that it will test from 0x20 to 0x27), to no avail. I2CScan recognized an I2C device conected at address 0x27...

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2019-06-17:
When the device boots there are some messages about the pcf search and the results. Please also check if the option is enabled in the compiler. Not sure about the current default.

MadDoct, 2019-06-17:
I don't have it with me right now, but I'm pretty sure there weren't any messages besides the usual tasmota startup messages...
It isn't enabled by default but I added the #define to the config...
Could it be some kind of conflict with display support? LCD displays use, very often, the pcf and those addresses might be reserved...?

stefanbode, 2019-06-17:
If you take a look in the pcf.ino file you can see some messages during bus testing and pcf probing. If correctly enabled you must see at least messages like pcf.... found = 0. if you see nothing the define does not work for whatever reason.

MadDoct, 2019-06-18:
There are no messages...
Do I need anything else besides adding #define USE_PCF8574 in my_user_config.h?
Strange
Could you test, please?

MadDoct, 2019-06-18:
Just found out the problem, from reading the code...

Why did you make enabling mqtt mandatory?
That should be, at least, documented... ;)

stefanbode, 2019-06-19:
Hm, good question. Let me check if there are requirements. If not I will change the code to remove this dependency

stefanbode, 2019-06-19:
I double check, but where do you see the MQTT dependency? Required is:
#ifdef USE_I2C
#ifdef USE_PCF8574
and for sure for configuation, it makes sense that #ifdef USE_WEBSERVER should also be defined.

But that's all.

closed 2019-06-19 by stefanbode

MadDoct, 2019-06-19:
Glad I could help.
It took me a while to figure it out... That was a strange "mistake", but happens to everybody... ;)

Anyway, thanks for the AMAZING work you did, providing support for this expander...

commit ee92ab8b (2019-06-19): fixed #154

Removed mqtt dependency