stefanbode/Sonoff-Tasmota #156: Flashing stefanbode fails

adizag, 2019-06-20:
Trying to flash stefanbode firmware. I understand i must flash the sonoff_minimal.bin before.

If I'm flashing  the sonoff_minimal.bin from stefanbode bin folder - the switch will not recover from restart. Stuck...
If I'm flashing the original sonoff_minimal.bin, the flashing succeeded but trying to then flash stefabode firmware I'm  getting: 
"sonoff firmware upgrade:
Upload Failed

Magic byte is not 0xE9"

The switch is generic tuya curtain switch.

Thanks in advance!!!

adizag, 2019-06-20:
Is there a reason why not flashing firmware directly by ftdi?

udo1toni, 2019-06-20:
You will only need to flash minimal.bin if flashing via OTA.

adizag, 2019-06-20:
The downloaded file was damaged....
Downloaded again - working!

Thanks!!!!

closed 2019-06-20 by adizag