stefanbode/Sonoff-Tasmota #157: Adax Wifi heater with ESP8266 and NXP 8574TS I2C

DenielTorres, 2019-07-11:
**Have you look for this feature in other issues and in the wiki?**
Yes
**Is your feature request related to a problem? Please describe.**
I was looking for information on flashing my board for Adax branded wifi electric heaters with ESP8266 chips, and it turns out it has these kinds of chips as well. 

**Describe the solution you'd like**
I saw that your fork basically supports this but only for relay functioning if I'm not mistaken. Was wondering if it could be repurposed/added as a more generic function. 

**Additional context**
Thanks very much in advance for taking the time and reading this through!!!

**(Please, remember to close the issue when the problem has been addressed)**

DenielTorres, 2019-07-11:
Here is an image of the board if that helps: 
[image]

stefanbode, 2019-07-23:
sorry never seen before. Maybe you ask in the main forum of TASMOTA.

closed 2019-07-23 by stefanbode

trommegutten, 2020-04-26:
> Here is an image of the board if that helps:
> [image]
@DenielTorres Did you try flashing this with Tasmota?

costinchirita, 2023-11-21:
Did you managed to flash it with Tasmota?

trommegutten, 2023-11-21:
No, Sorry. I’ve sold The heaters.MvhJarle H. Sjøthun21. nov.

costinchirita, 2023-11-21:
Ok, thanks!