stefanbode/Sonoff-Tasmota #161: open and close roller shutters with percentage timers.

matteo2019, 2019-07-20:
I use the fork stefanbode on dual sonoff.  I have to open / close the shutters with the timers.  Up to this point everything works.  The problem is to have the timers open at a certain percentage.  For example: at 6.00 o'clock the timer and must close 10% of the shutter.  At 7 o'clock the timer must increase the closure by 10% more.  and at 8 o'clock I want a timer that opens the whole shutter.  (these are just examples, my schedules are different)

stefanbode, 2019-07-20:
Shutterposition is not supported in rules.

matteo2019, 2019-07-20:
can't you associate a position with a timer?

stefanbode, 2019-07-20:
As mentioned before all shuttercomands are not available in automation. The idea is fine but I never reviewed what is required to be part of the rules framework. Without: no chance.

closed 2019-07-23 by stefanbode

matteo2019, 2019-07-23:
I'll explain better to avoid misunderstandings, unfortunately I'm not English.  I have 2 timers now.  A lift lowers the shutter completely at 13, When the sun comes.  The second timer lowers the shutter at 7pm when the sun has completely gone.  If I want to raise the blind at 50% at 18.00, can't I?

matteo2019, 2019-07-23:
I'll explain better to avoid misunderstandings, unfortunately I'm not English.  I have 2 timers now.  One lowers the shutter completely at 13, When the sun comes.  The second timer raises the shutter at 7 pm when the sun has completely gone.  If I want to raise the blind at 50% at 18.00, can't I?

snakuzzo, 2019-07-23:
Now you can't. As explained before, at this time, shuttercommands arn't available in automations.
You can just open or close

matteo2019, 2019-07-23:
ok clear.  If the article is updated in the future.