stefanbode/Sonoff-Tasmota #162: The PCF8574 driver will not load

MaliMrav, 2019-07-22:
Hi all,

I've tried for days on end and can't seem to load the PCF8574 driver.

I'm using PlatformIO
esp12e

in my_user_config, under // -- Low level interface devices -----------------
I've added:

#define USE_I2C
#ifdef USE_I2C
  #define USE_PCF8574
#endif  // USE_I2C

load the firmware, set up a template to use the SDA and SCL correctly

in the console i do a I2CScan and the result is:
09:14:08 MQT: stat/MPRSx8/RESULT = {"I2CScan":"Device(s) found at 0x38"}
so it detects the device but does not seem to attach the driver.

What Am I doing wrong?

I have noticed that from your initial publication of this add-on where there was a PFC8574 library and now there is no library for the driver.
The fork does not contain any reference to an installation/configuration guide.

Ca you please help me (and potentially others) with this?

stefanbode, 2019-07-22:
For me it looks like the USE_PCF8574 did not get executed. Because at the beginning of the log you should see in serial something like this. At least is must say 0 devices found. If you do not see the message. the include does not work.

RSLT: Final max devices: %d, PCF8574 devices %d

MaliMrav, 2019-07-22:
> For me it looks like the USE_PCF8574 did not get executed. Because at the beginning of the log you should see in serial something like this. At least is must say 0 devices found. If you do not see the message. the include does not work.
> 
> RSLT: Final max devices: %d, PCF8574 devices %d

So, how do I force the execution of the USE_PCF8574 ?
I compiled it using the sonoff environment. I've tried compiling the sonoff-sensors but the xsns_11_veml6070.ino is executed as the 0x38 is an address it uses as well.
I'm not sure why it doesn't get loaded.

Schorsch99, 2019-07-22:
I had some issues too and spent a few hours trying to figure it out.

Apart from the "define USE..." in my_user_config I also had to do the following changes:

#define PFC8574_ADDR1        0x20 (instead of 0x38)

If I am not mistaken, address range starting with 0x38 is for PCF8574A (mind the suffix) and range for PCF8574 starts with 0x20. Check the exact type of PCF you have and enter the address accordingly.

I also added the following in Sonoff-Tasmota/sonoff/xdrv_interface.ino:

#ifdef XDRV_90
  &Xdrv90,
#endif

I don't remember if the last step was actually required, but with the above changes I got it working.

Schorsch99, 2019-07-22:
Sorry, forgot to mention:

#define PFC8574_ADDR1 0x20

is in line 30 of Sonoff-Tasmota/sonoff/xdrv_90_pcf8574.ino

MaliMrav, 2019-07-22:
> #ifdef XDRV_90
> &Xdrv90,
> #endif

You, my friend are a genius!

the #ifdef XDRV_90
&Xdrv90,
#endif

did the trick!!!

Thank you!!!

Schorsch99, 2019-07-22:
You are very welcome! Glad I could help.

MaliMrav, 2019-07-22:
Ok so I got everything working just the way I like. 
[image]

One final thing...

in order to integrate with Alexa, I only have 4 of the 8 switches.
[image]

I'm so excited now that  can't wait to get it all going. 
Has the emulation got a hard limit of 4 devices?

stefanbode, 2019-07-22:
Yes, a lot is hard limit to 4. sometimes there are constants that you can change and some stuff is working like timers. But do not expect this for everything.

MaliMrav, 2019-07-23:
Hi all,

Ok, I've worked it out.
In sonoff.h modify the 
const uint8_t MAX_FRIENDLYNAMES = 8;        // Max number of Friendly names

[image]

to get...

[image]

And I hope the above changes in the decode-config.html are enough for the config backup. Stefan, you would know?

stefanbode, 2019-07-23:
Yes this will work. but the config backups are incompatible between the version with 4 and 8 friendly names. You cannot import the one with 4 names into the one with 8 names. Very strange things can happen.....uuhhh. Believe me!

MaliMrav, 2019-07-24:
> Yes this will work. but the config backups are incompatible between the version with 4 and 8 friendly names. You cannot import the one with 4 names into the one with 8 names. Very strange things can happen.....uuhhh. Believe me!

Yes, that’s would be my guess.
That’s why I wasn’t sure whether or not I would have to provission elswere the extra 4 arrays or by adjusting the MAX_FRIENDLYNAMES = 8 was enough to be able to backup/reset config/restore from backup?
Does this already provision for the backup/restore file (Of course, the firmware needs to be the version with 8)?

In any case... nice work, and thank you for this! It is a great piece of code!
Take care,
MM

closed 2019-08-28 by stefanbode

rt400, 2019-09-21:
@stefanbode hi,
sorry to open it again , but i have the same problem i dont see any PCF option in web.
i check everything that other guys said i config all also check in console if device found and it's
```
20:49:49 RSL: stat/sonoff/RESULT = {"I2CScan":"Device(s) found at 0x20"}
```
need help..thanks

rt400, 2019-09-22:
if i remove those lines now its work 
[image]

is that ok ?

stefanbode, 2019-09-23:
As you can see the PCF is defined in the xdrv_90. Therefore is makes no sense to comment this you if you want to use this. This lines must be active. and you have to define the SCL SDA pins.

rt400, 2019-09-23:
@stefanbode i try to explain :
if i add in my_user_config.ino : #define USE_PCF8574 , i got error : 
```
expected primary-expression before ' ' token arduino
```
and now if commet in xdrv_interface.ino in line 791-793:
```
#ifdef XDRV_90
  Xdrv90,
#endif
```
the compile work and i see in webUI the pcf config.
there are to place the those line are show in xdrv_interface.ino , one with & sign and onw with out.

rt400, 2019-09-23:
[image]