stefanbode/Sonoff-Tasmota #164: Domoticz no MQTT Messages

jannnfe, 2019-07-22:
Hello,
I've put sensor idx 10 shutter on the IDX that my virtual device has in Domoticz. An MQTT message is sent to Domoticz when the device is started but the changes to the shutter control are not sent to Domoticz. What am I doing wrong?
Thanks!

roondar, 2019-07-23:
Not really an answer but there is my solution : https://github.com/roondar/shutter-domoticz

stefanbode, 2019-07-23:
@roondar : If you like you can add this to the wiki. Always welcome.

jannnfe, 2019-07-23:
Thanks @roondar 
can you please give a more detailed instruction how to run your docker. I never run a docker and i can't find a good instruction on that.
Thanks! :)

roondar, 2019-07-24:
@stefanbode @jannnfe 
I add [Domoticz-integration](https://github.com/stefanbode/Sonoff-Tasmota/wiki/Domoticz-integration) in the wiki section.

I hope, it'll help you ;)

jannnfe, 2019-07-24:
Hey @roondar 
thanks! This instruction is clear and i could install the docker mod. Nice work!

But how to start the docker on system restart automatically?

Thanks!

roondar, 2019-07-24:
You right, this is the missing part of this guide.
You can follow this
https://docs.docker.com/config/containers/start-containers-automatically/

jannnfe, 2019-07-31:
@roondar And how to add multiple Shutters in shutter.yml?

roondar, 2019-07-31:
Good question, I haven't tested ! :D

roondar, 2019-08-12:
At this time, I'm only have 1 shutter but I updated my code to be compatible with multiple shutters. 
You can test it : 
 - edit  docker-compose.yml and replace "image: roondar/shutter-domoticz" by "image: roondar/shutter-domoticz:dev"
- edit shutter.yml and add multiple shutters like this:
shutters:
  - vr_salon: 1213
  - vr_bureau: 1234

If it's ok for you, I'll merge Dev to Master.

closed 2019-09-17 by stefanbode