stefanbode/Sonoff-Tasmota #165: 2 inching times scenes with 1 sonoff basic

svauz, 2019-07-23:
hi, my question is very particular, I'll try to explain it.. I have 12 shutters in my house, each shutter have installed a nice tt2n unit(is the name of the brand). I can by pushing 1 time the  button open or stop or close the shutter. another important thing is if I press and hold 9 seconds the button I can  force the opening,and if I hold it more than 11seconds the shutter will force to close. my question is.. can I create two scenes with 2 inching times (9sec and the other 11 sec)with a modded dry contact sonoff basic with your firmware?? many many thanks Danilo

stefanbode, 2019-07-23:
I assume this will not work because the shutter does not support rules. But maybe full open and full close will work because you only have to toggle the relay to on. The rest will be managed by the shutter firmware. But don't ask me how to make this 9 or 11 second  stuff. This might be basic TASMOTO. I have no idea, if you can create a trigger on this. good luck.

MadDoct, 2019-08-29:
@svauz , If I were you, I'd change those controll units for something "decent" like shelly2.5... You'd get a lot better options

closed 2019-08-30 by stefanbode