stefanbode/Sonoff-Tasmota #166: err msg "Shutter unknown" wenn einbabe INTERLOCK 1,2

markus-fehling, 2019-07-25:
Hallo Team, ich versuche gerade SONOFF T1 2way zum Laufen zu bringen
SETOPTION 80/81 gesetzt, es hapert bei INTERLOCK 1,2:
CMD: INTERLOCK 1,2
16:40:28 Shutter unknown
16:40:28 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}

Danke, Gruß Markus

stefanbode, 2019-08-18:
interlock is a standard TASMOTA feature. I do not understand your problem. as far as I know setoption80 is not required anymore. only setoption81 to enable the shutter.

stefanbode, 2019-08-26:
regarding to your log above: this is correct fom the point of view, that POWER2 is on. Now you Switch POWER1 on. This causes FIRST POWER2=off, wait 0,5sec and then POWER1 on to move in the opposite direction. with a higher loglevel you can also see WHO is triggering the relay.

stefanbode, 2019-08-30:
With the new TASMOTA firmware the interlock has changed. Please sut in the following two commands to get it workin
```
INTERLOCK 1
INTERLOCK 1,2
```
The first one will ENABLE, the second one will et the group.

closed 2019-08-30 by stefanbode