stefanbode/Sonoff-Tasmota #167: Upgrading from Tasmota 6.5.0.8 failed with WiFi not connecting

Crazor, 2019-07-28:
### BUG DESCRIPTION
I wanted to switch a Sonoff Dual R2 with Tasmota 6.5.0.8 to this fork. I tried to build and upload the normal sonoff/firmware.bin and got an error. Then I remembered that I had to upload sonoff-minimal/firmware.bin on another device before to work around this. So I built and uploaded sonoff-minimal/firmware.bin via the webinterface. Unfortunately, the device did not connect to wifi again and also did not bring up the wifi manager. Pushing the button 4 times (or shorting the other button pin to GND on the header next to the serial port on the Dual R2) did not bring up the WiFi manager either.

I had to then disconnect the device and bring it up on my bench to reflash with esptool.py, which worked fine. I did get a glimpse of the serial output before reflashing, but unfortunately, I managed to copy the logs and not save them anywhere. There were a bunch of messages about the R2 trying to connect to the AP and ultimately failing after several attempts. Both LEDs were on.

Is this a bug or is that behaviour expected when switching from mainline Tasmota to this fork? Is there any way to work around this issue? I've got a couple of Shelly 2.5's that I'd also like to upgrade, but those are a pain to disconnect or even reach to press a button if something goes wrong...

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Device used (i.e. Sonoff Basic) : _____
- [ ] Tasmota binary firmware version number used : ____ / (pre-compiled or self-compiled ?)
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2019-07-29:
Indeed the two config files are not compatible to each other. This is hard to fix and a known limitation. Anyhow the expected behavior is that the newly flashed device starts with default configuration. Is you use a user_config_override you also can preinject the WiFi credentials. This allows you to access the device. That it did not start at all is bad an normally should not happen. The config is a binary and this means everything can happen.

Crazor, 2019-07-29:
I updated two other devices, one is another Sonoff Dual R2 and the other a Shelly 2.5. This time, I used upstream Tasmota's sonoff-minimal.bin, which came up fine. I then uploaded the firmware I compiled and everything worked as expected. I read about it in another issue and decided to try it with devices that would be accessible, if inconvenient, and was lucky enough that they both upgraded flawlessly. I will stick to this way for now.
Maybe the -minimal build should always start WiFi Manager, regardless of what the config says.

stefanbode, 2019-08-18:
This is something you should address in the main forum for TASMOTA

closed 2019-08-18 by stefanbode